espressif/esp-protocols #31: About HW Flow Control

diplfranzhoepfinger, 2022-05-01:
Hello: 

if i want to use ESP_MODEM_FLOW_CONTROL_HW. 

the Scenario would be (in my Opinion, but i might be wrong) 

1. Connect to the Modem with ESP_MODEM_FLOW_CONTROL_NONE
2. issue a esp_modem_set_flow_control(dce, 2, 2)
3. Disconnect 
4. Reconnect to the Modem with ESP_MODEM_FLOW_CONTROL_HW

how do i achieve this ? 

another way would be: 

1. Connect to the Modem with ESP_MODEM_FLOW_CONTROL_NONE
2. issue a esp_modem_set_flow_control(dce, 2, 2)
3. change FlowControl from ESP_MODEM_FLOW_CONTROL_NONE --> ESP_MODEM_FLOW_CONTROL_HW

Literature Reading: 
SIM7070_SIM7080_SIM7090 Series_AT Command Manual_V1.05.pdf
Sections about Flow Control.

franz-ms-muc, 2022-05-01:
In SIM7000 Series UART Application Note_V1.00 there is more info.

One should send AT repeated to synchronize Baudrate. After this one should send AT+IFC=2,2.

franz-ms-muc, 2022-05-01:
Other story would be if one would like to set higher Baudrate. Like 2000000.
Need to connect first with lower Baudrate. The. Send AT+IPR. Then connect by the higher Baudrate.

diplfranzhoepfinger, 2022-05-02:
@david-cermak

david-cermak, 2022-05-04:
Hi,

Most devices support changing the UART parameters both temporarily and permanently. This works with the baudrate at least, here we use a temporary command:

https://github.com/espressif/esp-protocols/blob/38149c8d9bdf469806fe0685558eba74f3366d73/components/esp_modem/src/esp_modem_command_library.cpp#L142-L146

But, we can also issue `AT+IPREX` to set the baudrate permanently (You can play with these settings in the console example)

As for the control flow, I believe that you can set the parameter without reconnection. I also remember using `ESP_MODEM_FLOW_CONTROL_HW` directly from startup and configuring `AT+IFC` immediately after that, but need to double-check that.
I'll test it with some devices and get back to you.

david-cermak, 2022-05-05:
About the control flow: Yes, I was correct. It's just enough to init  the modem with:
```cpp
    esp_modem_dte_config_t dte_config = ESP_MODEM_DTE_DEFAULT_CONFIG();
    dte_config.uart_config.flow_control = ESP_MODEM_FLOW_CONTROL_HW;
    auto uart_dte = create_uart_dte(&dte_config);
```
and just issue this command
```
cmd AT+IFC=2,2
```
to enable flow control on the device side. (this seems to work, as the CTS is asserted in no-flow-control mode, so we could send data and once the device receives it, it activates modem signals and we could receive data, too)

diplfranzhoepfinger, 2022-05-05:
Cool. Will test.

closed 2022-06-27 by diplfranzhoepfinger