espressif/esp-protocols #37: Disable CMUX when it is no longer required

jonathandreyer, 2022-05-15:
Hi everyone,

Does it possible to disable the CMUX after the usage? My idea is to activate the CMUX when the use of PPP is necessary and disable it after the PPP mode is no longer required. I ask that because I have tried with the following code (at the end of the *simple_cmux_example*), but it is not working.
```
if (dce->set_mode(esp_modem::modem_mode::COMMAND_MODE)) {
    std::cout << "Modem has correctly entered command mode" << std::endl;
} else {
    ESP_LOGE(TAG, "Failed to configure desired mode... exiting");
    return;
}
```

Best regards,

Jonathan Dreyer

david-cermak, 2022-05-16:
Yes, it should be possible to exit CMUX mode (haven't really tried with real devices, though), but this is not implemented in the `esp-modem` component. The CMUX mode only creates two virtual terminals so that one could be used for commands and the other one for PPP mode.
When PPP mode is not longer required, you can still exit the PPP mode and have two virtual terminals in command mode.

jonathandreyer, 2022-05-17:
Hi, thanks for your message. I will study both approaches, but I have a preference for closing the CMUX mode. I think that is more elegant than leave unused resources open. Do you have any recommendations about that?
Best regards,
Jonathan

jonathandreyer, 2022-05-23:
Hi, thanks for your reply. I will study the behaviour of the CMUX and see if I can do something for this point (the CMUX protocol seems to me a bit complex). If I have something, I will leave a comment here.
Best regards,
Jonathan

david-cermak, 2022-06-06:
Hi Jonathan,

There's still a problem of the *meaning* of the modes. Currently, when we switch to `CMUX` we actually create two virtual terminals, both in command mode. then any further switching to `PPP`, changes the mode of one virtual terminal only.

So yes, you're correct.
1) set mode to `CMUX`
2) set mode to `PPP`
~ Do work
3) Set mode to `CMD` should exit both virtual terminals and leave only one (physical) in command mode.

I think it would probably make more sense to have the `CMUX` mode automatically create one command terminal and one data terminal (i.e. points `1)` and `2)` in a single step: set mode to `CMUX`).
---

> I have tested but that not work in my case. 

Could you please share the logs?

> implement it in the CMUX example (optionally activated with a define variable)

That would be helpful, thanks!

david-cermak, 2022-06-06:
The modem could stay in the `CMUX` mode and we could just set the `CMUX` mode again when we wake-up. This should normally work, but I'm not sure if it's desirable if you go to sleep for some minutes? perhaps you want to power-off the modem, too?

jonathandreyer, 2022-06-06:
Hi @david-cermak, thanks you for your reply and your confirmation regarding the implementation.

When I have written my previous message, I had the feeling that it is related to the #47 because the indication "NO CARRIER" are received on the UART. With your new message inside this issue, I have tested the PR #42 (with 4 different commits) and now that seems working.

Regarding the example, I will do that as soon as possible, but I have a bit busy this week.

Best regards,

Jonathan Dreyer

david-cermak, 2022-06-07:
@jonathandreyer Thanks for the feedback. Yes, I've updated the PR per the discussion here, to actually simplify the mode switching.

@bbinet I'll have to look into the low power modes for common modems and will comment on that issue. From what I see, these are typically just `AT` commands that configure and setup the PSM parameters (and perhaps some GPIOs as wake-up signals)

> if CMUX on module will still be on after waking up from PSM.

I think the CMUX mode would be still active, but havent' really checked (can test it with `70xx` device family)

jonathandreyer, 2022-06-08:
@david-cermak Thanks for the reply. I have prepared a PR, but I will wait a bit on the PR #52 (which is no longer in WIP).

closed 2022-06-21 by david-cermak