espressif/esp-protocols #46: About the problem of CMUX protocol mismatch

hqs1, 2022-05-30:
Hello, this warning will be reported when running the esp_modem_cmux example，I don't know if it's a cache size issue.
module:ESP32 S2 WROVER
4G module:合宙Air724UG

**W (7798) CMUX: Protocol mismatch: Missed leading SOF, recovering...**

> ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1768
load:0x4004c000,len:0xa48
load:0x40050000,len:0x2e58
entry 0x4004c204
I (21) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (21) boot: compile time 13:03:43
I (21) boot: chip revision: 0
I (24) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (33) boot.esp32s2: SPI Flash Size : 4MB
I (38) boot: Enabling RNG early entropy source...
I (43) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (62) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (69) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  3 factory          factory app      00 00 00010000 00100000
I (84) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (92) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (99) boot: End of partition table
I (103) boot: Defaulting to factory image
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=25924h (153892) map
I (147) esp_image: segment 1: paddr=0003594c vaddr=3ffbe710 size=01e04h (  7684) load
I (149) esp_image: segment 2: paddr=00037758 vaddr=40022000 size=088c0h ( 35008) load
I (162) esp_image: segment 3: paddr=00040020 vaddr=40080020 size=7056ch (460140) map
I (254) esp_image: segment 4: paddr=000b0594 vaddr=4002a8c0 size=03e48h ( 15944) load
I (258) esp_image: segment 5: paddr=000b43e4 vaddr=50000000 size=00010h (    16) load
I (266) boot: Loaded app from partition at offset 0x10000
I (266) boot: Disabling RNG early entropy source...
I (282) cach����~f~�ѥ���cache      : size 8KB, 4Ways, cache line size 32Byte
I (283) cpu_start: Pro cpu up.
I (294) cpu_start: Pro cpu start user code
I (294) cpu_start: cpu freq: 160000000
I (294) cpu_start: Application information:
I (299) cpu_start: Project name:     simple_cmux_client
I (305) cpu_start: App version:      1
I (309) cpu_start: Compile time:     May 30 2022 17:25:46
I (315) cpu_start: ELF file SHA256:  83828e316ca86c55...
I (321) cpu_start: ESP-IDF:          v4.4.1
I (326) heap_init: Initializing. RAM available for dynamic allocation:
I (333) heap_init: At 3FFC3390 len 00038C70 (227 KiB): DRAM
I (340) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (346) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (353) spi_flash: detected chip: generic
I (357) spi_flash: flash io: dio
I (367) cpu_start: Starting scheduler on PRO CPU.
I (368) uart: queue free spaces: 30
Modem has correctly entered multiplexed command/data mode
Operator name:"46000",7
W (7798) CMUX: Protocol mismatch: Missed leading SOF, recovering...
I (7798) CMUX: Protocol recovered
I (13798) esp-netif_lwip-ppp: Connected
I (13798) esp-netif_lwip-ppp: Name Server1: 211.136.17.107
I (13798) esp-netif_lwip-ppp: Name Server2: 0.0.0.0
I (13798) cmux_example: IP          : 10.167.128.10
I (13808) cmux_example: Netmask     : 255.255.255.255
I (13818) cmux_example: Gateway     : 192.168.0.1

david-cermak, 2022-06-01:
@hqs1 Do you see this warning only before the network connects or is it appearing randomly?
This basically reports that some required characters were missing when parsing the CMUX protocol, should recover with no issues (which it has according to your log). Could you please enable verbose logging and share the output?

hqs1, 2022-06-02:
Hello, I modified the log level and found that it seems that the CMUX data has not been received completely, but the data is already being parsed.
```
D (425134) HTTP_CLIENT: Begin connect to: http://json.hqshome.cn:80
D (425134) esp-tls: host:json.hqshome.cn: strlen 15
V (425134) Send: 0x3ffca820   f9 05 ef 85                                       |....|
V (425134) Send: 0x3ffcffe8   7e 21 45 00 00 3d 00 d0  00 00 ff 11 95 ad 0a 68  |~!E..=.........h|
V (425144) Send: 0x3ffcfff8   fc fe 78 c4 a5 07 9b 1f  00 35 00 29 dd d8 64 a2  |..x......5.)..d.|
V (425154) Send: 0x3ffd0008   01 00 00 01 00 00 00 00  00 00 04 6a 73 6f 6e 07  |...........json.|
V (425164) Send: 0x3ffd0018   68 71 73 68 6f 6d 65 02  63 6e 00 00 01 00 01 d1  |hqshome.cn......|
V (425174) Send: 0x3ffd0028   fb 7e                                             |.~|
V (425184) Send: 0x3ffca824   b1 f9                                             |..|
V (426034) Send: 0x3ffca810   f9 05 ef 85                                       |....|
V (426034) Send: 0x3ffcffe8   7e 21 45 00 00 3d 00 d1  00 00 ff 11 95 ac 0a 68  |~!E..=.........h|
V (426034) Send: 0x3ffcfff8   fc fe 78 c4 a5 07 9b 1f  00 35 00 29 dd d8 64 a2  |..x......5.)..d.|
V (426044) Send: 0x3ffd0008   01 00 00 01 00 00 00 00  00 00 04 6a 73 6f 6e 07  |...........json.|
V (426054) Send: 0x3ffd0018   68 71 73 68 6f 6d 65 02  63 6e 00 00 01 00 01 67  |hqshome.cn.....g|
V (426064) Send: 0x3ffd0028   28 7e                                             |(~|
V (426074) Send: 0x3ffca814   b1 f9                                             |..|
V (426214) CMUX Received: 0x3ffcd97c   f9 05 ef 7c 03 7e 21 45  74 01 b8 25 ea 40 00 38  |...|.~!Et..%.@.8|
V (426214) CMUX Received: 0x3ffcd98c   11 f5 a4 78 c4 a5 07 0a  68 fc fe 00 35 9b 1f 01  |...x....h...5...|
V (426214) CMUX Received: 0x3ffcd99c   a4 0b ee 64 a2 81 80 00  01 00 01 00 02 00 12 04  |...d............|
V (426224) CMUX Received: 0x3ffcd9ac   6a 73 6f 6e 07 68 71 73  68 6f 6d 65 02 63 6e 00  |json.hqshome.cn.|
D (426234) CMUX: Payload frame: dlci:01 type:ef payload:62 available:60
V (426244) CMUX Received: 0x3ffcd9bc   00 01 00 01                                       |....|
D (426254) CMUX: Payload frame: dlci:01 type:ef payload:2 available:4
W (426264) CMUX: Protocol mismatch: Missed trailing SOF, recovering...
V (426264) CMUX Received: 0x3ffcd97c   c0 0c 00 01 00 01 00 00  00 c7 00 04 65 2b 59 ca  |............e+Y.|
V (426274) CMUX Received: 0x3ffcd98c   c0 11 00 02 00 01 00 01  4f ef 00 13 05 64 6e 73  |........O....dns|
V (426284) CMUX Received: 0x3ffcd99c   31 35 07 68 69 63 68 69  6e 61 03 63 6f 6d 00 c0  |15.hichina.com..|
V (426294) CMUX Received: 0x3ffcd9ac   11 00 02 00 01 00 01 4f  ef 00 08 05 64 6e 73 31  |.......O....dns1|
V (426304) CMUX Received: 0x3ffcd9bc   36 c0 43 c0 5c 00 01 00  01 00 02 8f 14 00 04 8b  |6.C.\...........|
V (426314) CMUX Received: 0x3ffcd9cc   e0 8e 6f c0 5c 00 01 00  01 00 02 8f 14 00 04 8b  |..o.\...........|
V (426324) CMUX Received: 0x3ffcd9dc   e0 8e 79 c0 5c 00 01 00  01 00 02 8f 14 00 04 27  |..y.\..........'|
V (426334) CMUX Received: 0x3ffcd9ec   60 99 29 c0 5c 00 01 00  01 00 02 8f 14 00 04 27  |`.).\..........'|
V (426344) CMUX Received: 0x3ffcd9fc   60 99 33 c0 5c 00 01 00  01 00 02 8f 14 00 04 2f  |`.3.\........../|
V (426354) CMUX Received: 0x3ffcda0c   76 c7 c9 c0 5c 00 01 00  01 00 02 8f 14 00 04 2f  |v...\........../|
V (426364) CMUX Received: 0x3ffcda1c   76 c7 d3 c0 5c 00 01 00  01 00 02 8f 14 00 04 78  |v...\..........x|
V (426374) CMUX Received: 0x3ffcda2c   4c 6b 29 c0 5c 00 01 00  01 00 02 8f 14 00 04 78  |Lk).\..........x|
V (426384) CMUX Received: 0x3ffcda3c   4c 6b 33 c0 3d 00 01 00  01 00 00 08 a2 00 04 27  |Lk3.=..........'|
V (426394) CMUX Received: 0x3ffcda4c   60 99 34 c0 3d 00 01 00  01 00 00 08 a2 00 04 2f  |`.4.=........../|
V (426404) CMUX Received: 0x3ffcda5c   76 c7 ca c0 3d 00 01 00  01 00 00 08 a2 00 04 2f  |v...=........../|
V (426414) CMUX Received: 0x3ffcda6c   76 c7 d4 c0 3d 00 01 00  01 00 00 08 a2 00 04 78  |v...=..........x|
V (426424) CMUX Received: 0x3ffcda7c   4c 6b 2a c0 3d 00 01 00  01 00 00 08 a2 00 04 78  |Lk*.=..........x|
V (426434) CMUX Received: 0x3ffcda8c   4c 6b 34 c0 3d 00 01 00  01 00 00 08 a2 00 04 8b  |Lk4.=...........|
V (426444) CMUX Received: 0x3ffcda9c   e0 8e 70 c0 3d 00 01 00  01 00 00 08 a2 00 04 8b  |..p.=...........|
V (426454) CMUX Received: 0x3ffcdaac   e0 8e 7a c0 3d 00 01 00  01 00 00 08 a2 00 04 27  |..z.=..........'|
V (426464) CMUX Received: 0x3ffcdabc   60 99 2a c0 5c 00 1c 00  01 00 02 88 7d 5d 00 10  |`.*.\.......}]..|
V (426474) CMUX Received: 0x3ffcdacc   24 08 40 09 05 01 00 00  00 00 00 00 00 00 00 24  |$.@............$|
V (426484) CMUX Received: 0x3ffcdadc   c0 3d 00 1c 00 01 00 00  08 a2 00 10 24 08 40 09  |.=..........$.@.|
V (426494) CMUX Received: 0x3ffcdaec   05 01 00 00 00 00 00 00  00 00 00 23 69 90 7e 53  |...........#i.~S|
V (426504) CMUX Received: 0x3ffcd97c   f9 f9 05 ef 7a 03 7e 21  45 74 01 b8 25 eb 40 00  |....z.~!Et..%.@.|
V (426514) CMUX Received: 0x3ffcd98c   38 11 f5 a3 78 c4 a5 07  0a 68 fc fe 00 35 9b 1f  |8...x....h...5..|
V (426524) CMUX Received: 0x3ffcd99c   01 a4 34 fa 64 a2 81 80  00 01 00 01 00 02 00 12  |..4.d...........|
V (426534) CMUX Received: 0x3ffcd9ac   04 6a 73 6f 6e 07 68 71  73 68 6f 6d 65 02 63 6e  |.json.hqshome.cn|
V (426544) CMUX Received: 0x3ffcd9bc   00 00 01 00 01 c0 0c 00  01 00 01 00 00 01 f7 00  |................|
[... cut ...]
V (433344) CMUX Received: 0x3ffcdabc   27 60 99 29 c0 5c 00 1c  00 01 00 00 09 13 00 10  |'`.).\..........|
V (433354) CMUX Received: 0x3ffcdacc   24 08 40 09 05 01 00 00  00 00 00 00 00 00 00 23  |$.@............#|
V (433364) CMUX Received: 0x3ffcdadc   c0 3d 00 1c 00 01 00 02  7f d4 00 10 24 08 40 09  |.=..........$.@.|
V (433374) CMUX Received: 0x3ffcdaec   05 01 00 00 00 00 00 00  00 00 00 24 08 b2 7e 3f  |...........$..~?|
V (433384) CMUX Received: 0x3ffcd97c   f9                                                |.|

david-cermak, 2022-06-03:
@hqs1 Thanks very much for sharing the logs!

I understand the problem, now. It looks like your device uses two bytes for encoding payload length (which is perfectly okay and according to the protocol), but the cmux implementation takes only the first byte into the account:

https://github.com/espressif/esp-protocols/blob/417591b99ff9abae3265ff66a5e575f3c68f7491/components/esp_modem/src/esp_modem_cmux.cpp#L193

We need to add something like:
```cpp
if ((frame_header[3]&1) == 0) {
  payload_len += frame_header[4]<<7;
  frame.advance(1);
}
```

The actual fix would be a little more complicated, as we have to think of fragmentation (like receiving incomplete header)

Sorry for the inconvenience. Will be fixed.

hqs1, 2022-06-05:
Thanks for the reply, I saw that the CMUX protocol does have one bit in the length byte to indicate whether the length is 1 byte or 2 bytes.

david-cermak, 2022-06-06:
@hqs1 I've fixed the problem in https://github.com/espressif/esp-protocols/pull/53

But I haven't really tested it with an actual device (only added unit test: https://github.com/espressif/esp-protocols/runs/6755900526?check_suite_focus=true#step:5:643). I would be really nice and helpful if you could find some time to check that version and try it with your device.

hqs1, 2022-06-07:
Thank you very much for your help, this version works fine on my device and can receive the json from the server and the signal quality of the 4G modem.
[image]

hqs1, 2022-06-10:
Hello, this happens when the ppp is not received and the query 4G modem is executed. It feels like it needs to be locked.
[image]

david-cermak, 2022-06-10:
@hqs1 I've closed this by merging the fix (the original issue with CMUX protocol mismatch has been resolved).

I'll address the locking issue in https://github.com/espressif/esp-protocols/pull/42

closed 2022-06-10 by david-cermak

DarshanDodal, 2022-08-18:
I am still facing the same issue. Please Help!!