espressif/esp-protocols #47: Bug while changing modem mode from "data" to "command"

bbinet, 2022-05-31:
When my BG95-M3 modem is in data mode, if I try to move to command mode with api `esp_modem_set_mode(dce, ESP_MODEM_MODE_COMMAND)`, then I can the send AT commands, but the esp_modem still thinks it is in data mode, so if I try to move back to data mode, it will fail...

After investigating, it seems `device->set_mode(modem_mode::COMMAND_MODE)` returns `false` at
https://github.com/espressif/esp-protocols/blob/2a4e684848a23efb6d607edd276c3f01b094fb9d/components/esp_modem/src/esp_modem_dce.cpp#L34-L36 which explain why internal `mode` is not changed (see: https://github.com/espressif/esp-protocols/blob/2a4e684848a23efb6d607edd276c3f01b094fb9d/components/esp_modem/src/esp_modem_dce.cpp#L46)

david-cermak, 2022-06-01:
@bbinet Yes, this seems to be a bug, as when the device fails to switch mode, the mode should go to Unknown:

```patch
 if (!device->set_mode(modem_mode::COMMAND_MODE)) { 
     mode = modem_mode::UNDEF:
     return false; 
 } 
```

I can also see that switching modes could be improved, this haven't been thoroughly tested with many devices, but in general it misses some retry mechanism when sending the escape sequence of `+++`.

Moreover, if you're using the C-API, it doesn't seem to return this failure, so even users couldn't implement a retry.

Will address these 3 problems.

bbinet, 2022-06-01:
Thank you @david-cermak.
Also do you know why `device->set_mode(modem_mode::COMMAND_MODE)` always fail for me whereas the mode actually seem to have correctly been changed to command mode as I'm able to send AT commands successfully.

david-cermak, 2022-06-01:
> Also do you know why device->set_mode(modem_mode::COMMAND_MODE) always fail 

Does it *always* fail? 
My guess is that your device perhaps needs more time to respond, or a delay before sending the command or something similar. Switching modes is very tricky and very much depends on the actual device. I haven't tried working with the `BG95-M3` device. (according to the docs on `+++` command: we should retry after 1 second period, until we succeed...)

david-cermak, 2022-06-01:
Yes, it works for `BG96`

The command to exit (escape) PPP mode is implemented here

https://github.com/espressif/esp-protocols/blob/7346ed9765b9ceafc0cbcf5c70a80fee4fd6a21b/components/esp_modem/src/esp_modem_command_library.cpp#L287-L293

bbinet, 2022-06-02:
@david-cermak when I try to log return value at https://github.com/espressif/esp-protocols/blob/7346ed9765b9ceafc0cbcf5c70a80fee4fd6a21b/components/esp_modem/src/esp_modem_command_library.cpp#L292
I get `command_result::TIMEOUT`.

This is weird as if I connect a serial port directly to my bg95-m3 modem, and I manually send command `+++` to switch mode from data to command, I immediately get the response "NO CARRIER", so I should not have a timeout when communicating with the bg95 module through esp_modem.

bbinet, 2022-06-02:
I've tried to add some additional logs to understand why I get a timeout, and in the mean time I've connected the RX line of an additional serial port to the TX line of the bg95 module to act as a sniffer to see its response in realtime.

I've added the following logs:
- "set_read_cb" after line 40: https://github.com/espressif/esp-protocols/blob/2a4e684848a23efb6d607edd276c3f01b094fb9d/components/esp_modem/src/esp_modem_dte.cpp#L40-L41
- "command_term->write" after line 57:
https://github.com/espressif/esp-protocols/blob/2a4e684848a23efb6d607edd276c3f01b094fb9d/components/esp_modem/src/esp_modem_dte.cpp#L57-L58

Then when I switch the mode from data to command, I only see the log "command_term->write", but not "set_read_cb", whereas I can see the "NO CARRIER" response on my additional serial sniffer.

Any idea why the `set_read_cb` callback is not called?

david-cermak, 2022-06-03:
> Any idea why the set_read_cb callback is not called?

Does your serial sniffer show the newline characters (`\r\n`) after the `NO CARRIER` response? That could be the reason why the lambda doesn't get called. I'll try to reproduce it.

bbinet, 2022-06-06:
@david-cermak I confirm that your PR #52 fixes this issue on mode BG95-M3: I can now successfully switch from data to command mode.
Thank you!

closed 2022-06-21 by david-cermak