espressif/esp-protocols #55: About the problem of executing AT command when the buffer is full in CMUX mode

hqs1, 2022-06-11:
Hello, in the CMUX mode, the SOC is receiving network data (the buffer is full, and the network data is still being received), and then sending the AT command will affect the network data. So I'm thinking is it necessary to add mutex locks in data mode and command mode? The command mode can only be executed after the reception is completed in the data mode.

hqs1, 2022-06-11:
My business is to start a task to query the 4G signal quality, and use Alibaba Cloud IoT for cloud connection. I don't know if it is in data mode (the data is still being received, judged according to **on_cmux: state: 2** in the Log), the execution of the AT command caused this exception.

[image]

```
void get_rssi_tim(void *dced)
{
    int rssi, ber;
    for (;;)
    {
        ESP_LOGI(TAG, "get_rssi!!!");
        esp_err_t err = esp_modem_get_signal_quality(dce, &rssi, &ber);
        if (err != ESP_OK)
        {
            ESP_LOGE(TAG, "get_rssi_tim failed with %d", err);
        }
        env_set_device_rssi(rssi);
        ESP_LOGI(TAG, "Signal quality: rssi=%d, ber=%d", rssi, ber);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
    vTaskDelete(NULL);
}

// I added a Log here to print the status
bool CMux::on_cmux(uint8_t *data, size_t actual_len)
{
    if (!data)
    {
#ifdef DEFRAGMENT_CMUX_PAYLOAD
        auto data_to_read = buffer_size - 128; // keep 128 (max CMUX payload) backup buffer)
        if (payload_start)
        {
            data = payload_start + total_payload_size;
            data_to_read = payload_len + 2;
        }
        else
        {
            data = buffer.get();
        }
        actual_len = term->read(data, data_to_read);
#else
        data = buffer.get();
        actual_len = term->read(data, buffer_size);
#endif
    }
    ESP_LOG_BUFFER_HEXDUMP("CMUX Received", data, actual_len, ESP_LOG_VERBOSE);
    CMuxFrame frame = {.ptr = data, .len = actual_len};
    while (frame.len > 0)
    {
        ESP_LOGI("on_cmux", "state:%d", (int)state);
        switch (state)
        {
        case cmux_state::RECOVER:
            if (!on_recovery(frame))
            {
                return false;
            }
            break;
        case cmux_state::INIT:
            if (!on_init(frame))
            {
                return false;
            }
            break;
        case cmux_state::HEADER:
            if (!on_header(frame))
            {
                return false;
            }
            break;
        case cmux_state::PAYLOAD:
            if (!on_payload(frame))
            {
                return false;
            }
            break;
        case cmux_state::FOOTER:
            if (!on_footer(frame))
            {
                return false;
            }
            break;
        }
    }
    return true;
}

```

david-cermak, 2022-06-27:
@hqs1 Was your issue fixed by the recent CMUX changes? (I thought it could, as we added a locking mechanism when switching modes)

hqs1, 2022-06-27:
@david-cermak The problem is still not solved, I try to create a task to read the signal quality constantly. However, this problem occurs when there is network data communication. Not sure if the dce handle is not thread safe
```
V (14061) Send: 0x3ffc9e40   f9 05 ef 5b                                       |...[|
V (14071) Send: 0x3ffd899c   7e 21 45 00 00 28 00 04  00 00 ff 06 a5 dd 0a b9  |~!E..(..........|
V (14081) Send: 0x3ffd89ac   1b e0 2f 67 bf ee c9 d4  01 bb 14 2f 32 15 b6 19  |../g......./2...|
V (14091) Send: 0x3ffd89bc   c4 f7 50 10 11 20 fb e0  00 00 09 f2 7e           |..P.. ......~|
V (14091) Send: 0x3ffc9e44   d7 f9                                             |..|
V (14101) CMUX Received: 0x3ffcee8c   00 01 80 01 dc 7b b6 00  00 04 03 00 48 30 46 02  |.....{......H0F.|
V (14111) CMUX Received: 0x3ffcee9c   21 00 d0 74 b0 e6 61 49  8b 3a 31 01 12 3c 7a 9d  |!..t..aI.:1..<z.|
V (14121) CMUX Received: 0x3ffceeac   b9 5f cc 80 c1 04 26 89  d7 1d ea 17 a9 41 bb 98  |._....&......A..|
V (14131) CMUX Received: 0x3ffceebc   b2 99 02 21 00 b5 df ab  09 1b 11 f4 5c 5e 78 50  |...!........\^xP|
V (14141) CMUX Received: 0x3ffceecc   13 15 d2 3b d4 60 b6 05  c1 d5 31 22 cc 29 1d 08  |...;.`....1".)..|
V (14151) CMUX Received: 0x3ffceedc   bb 33 96 99 22 00 77 00  6f 53 76 ac 31 f0 31 19  |.3..".w.oSv.1.1.|
V (14161) CMUX Received: 0x3ffceeec   d8 99 00 a4 51 15 ff 77  15 1c 11 d9 02 c1 00 29  |....Q..w.......)|
V (14171) CMUX Received: 0x3ffceefc   06 8d b2 08 9a 37 d9 13  00 00 01 80 01 dc 78 65  |.....7........xe|
I (14181) cmux: get_rssi!!!
V (14181) command_lib: get_signal_quality
V (14191) command_lib: generic_get_string
V (14191) Send: 0x3ffd0530   f9 09 ef 0f                                       |....|
V (14201) Send: 0x3ffd0658   41 54 2b 43 53 51 0d                              |AT+CSQ.|
V (14211) Send: 0x3ffd0534   3b f9                                             |;.|

abort() was called at PC 0x40095ff9 on core 0
0x40095ff9: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned int, char) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/include/cxx_include/esp_modem_exception.hpp:63
 (inlined by) esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned 
int, char) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/src/esp_modem_dte.cpp:59

Backtrace:0x4002395a:0x3ffd04e00x40028abd:0x3ffd0500 0x4002ecba:0x3ffd0520 0x40095ff9:0x3ffd0590 0x40096607:0x3ffd0600 0x40098ea2:0x3ffd0630 0x40094ec5:0x3ffd0690 0x400f145d:0x3ffd06b0 0x40088839:0x3ffd06d0 0x4002bd6d:0x3ffd0710
0x4002395a: panic_abort at C:/Users/HQS/esp/esp-idf/components/esp_system/panic.c:402

0x40028abd: esp_system_abort at C:/Users/HQS/esp/esp-idf/components/esp_system/esp_system.c:128

0x4002ecba: abort at C:/Users/HQS/esp/esp-idf/components/newlib/abort.c:46

0x40095ff9: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned int, char) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/include/cxx_include/esp_modem_exception.hpp:63
 (inlined by) esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned 
int, char) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/src/esp_modem_dte.cpp:59

0x40096607: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned int) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/src/esp_modem_dte.cpp:68

0x40098ea2: esp_modem::dce_commands::get_signal_quality(esp_modem::CommandableIf*, int&, int&) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/src/esp_modem_command_library.cpp:64
 (inlined by) esp_modem::dce_commands::get_signal_quality(esp_modem::CommandableIf*, int&, int&) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/src/esp_modem_command_library.cpp:393

0x40094ec5: esp_modem::GenericModule::get_signal_quality(int&, int&) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/src/esp_modem_modules.cpp:46

0x400f145d: esp_modem::command_result esp_modem::DCE::get_signal_quality<int&, int&>(int&, int&) at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/include/cxx_include/esp_modem_dce.hpp:117
 (inlined by) esp_modem_get_signal_quality at C:\Users\HQS\Desktop\4g_modem\build/../components/esp_modem/src/esp_modem_c_api.cpp:206

0x40088839: get_rssi_tim at C:\Users\HQS\Desktop\4g_modem\build/../main/cmux/cmux.c:144

0x4002bd6d: vPortTaskWrapper at C:/Users/HQS/esp/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: d8057b498a1a4f39

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400234e5
0x400234e5: esp_restart_noos_dig at C:/Users/HQS/esp/esp-idf/components/esp_system/esp_system.c:44 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1768
load:0x4004c000,len:0xa48
[... cut ...]
I (293) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (293) cpu_start: Pro cpu up.
D (300) clk: RTC_SLOW_CLK calibration value: 6415770
I (309) cpu_start: Pro cpu start user code
I (309) cpu_start: cpu freq: 240000000
I (309) cpu_start: Application information:
I (314) cpu_start: Project name:     4g_modem
I (319) cpu_start: App version:      1.0.1
I (324) cpu_start: Compile time:     Jun 24 2022 18:46:42
I (330) cpu_start: ELF file SHA256:  d8057b498a1a4f39...
I (336) cpu_start: ESP-IDF:          v4.4.1
```

david-cermak, 2022-06-28:
Ok, let me re-open this and ask you if you can provide steps to reproduce this problem. Thanks!

hqs1, 2022-06-28:
Thanks for the reply, here I download a relatively large picture through http to simulate the network data download, and create a task at the same time, this task is to execute the AT command at intervals to obtain the signal quality of the 4G modem.
code show as below

```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_netif.h"
#include "esp_netif_ppp.h"
#include "mqtt_client.h"
#include "esp_modem_api.h"
#include "esp_log.h"
#include "esp_http_client.h"

static EventGroupHandle_t event_group = NULL;
static const int CONNECT_BIT = BIT0;
esp_modem_dce_t *dce = NULL;
static const char *TAG = "cmux_example_c";

void get_rssi_tim(void *dce_handle);

esp_err_t _http_event_handle(esp_http_client_event_t *evt)
{
    switch (evt->event_id)
    {
    case HTTP_EVENT_ERROR:
        ESP_LOGI(TAG, "HTTP_EVENT_ERROR");
        break;
    case HTTP_EVENT_ON_CONNECTED:
        ESP_LOGI(TAG, "HTTP_EVENT_ON_CONNECTED");
        break;
    case HTTP_EVENT_HEADER_SENT:
        ESP_LOGI(TAG, "HTTP_EVENT_HEADER_SENT");
        break;
    case HTTP_EVENT_ON_HEADER:
        ESP_LOGI(TAG, "HTTP_EVENT_ON_HEADER");
        // printf("%.*s", evt->data_len, (char *)evt->data);
        break;
    case HTTP_EVENT_ON_DATA:
        ESP_LOGI(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
        if (!esp_http_client_is_chunked_response(evt->client))
        {
            // printf("%.*s", evt->data_len, (char *)evt->data);
        }

        break;
    case HTTP_EVENT_ON_FINISH:
        ESP_LOGI(TAG, "HTTP_EVENT_ON_FINISH");
        break;
    case HTTP_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "HTTP_EVENT_DISCONNECTED");
        break;
    }
    return ESP_OK;
}

static void on_ppp_changed(void *arg, esp_event_base_t event_base,
                           int32_t event_id, void *event_data)
{
    ESP_LOGI(TAG, "PPP state changed event %d", event_id);
    if (event_id == NETIF_PPP_ERRORUSER)
    {
        /* User interrupted event from esp-netif */
        esp_netif_t *netif = event_data;
        ESP_LOGI(TAG, "User interrupted event from netif:%p", netif);
    }
}

static void on_ip_event(void *arg, esp_event_base_t event_base,
                        int32_t event_id, void *event_data)
[... 91 lines cut ...]
    }
}

void get_rssi_tim(void *dce_handle)
{
    int rssi, ber;
    for (;;)
    {
        ESP_LOGI(TAG,"about to get rssi");
        esp_err_t err = esp_modem_get_signal_quality(dce_handle, &rssi, &ber);
        if (err != ESP_OK)
        {
            ESP_LOGE(TAG, "get_rssi_tim failed with %d", err);
        }
        ESP_LOGI(TAG, "Signal quality: rssi=%d, ber=%d", rssi, ber);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
    vTaskDelete(NULL);
}
```

run log
[image]
[image]
[image]

hqs1, 2022-07-01:
@david-cermak After deletion the buffer_size variable will be undefined
https://github.com/espressif/esp-protocols/blob/ef1bae5cdd65f10a6739c1b9b0aa3f0283f13ffa/components/esp_modem/src/esp_modem_cmux.cpp#L289

david-cermak, 2022-07-01:
Oh, sorry. Missed this in the last `unique-buffer` refactor; should be `buffer.size` instead of `buffer_size` (*dot*, not *underscore*)

```patch
diff --git a/components/esp_modem/src/esp_modem_cmux.cpp b/components/esp_modem/src/esp_modem_cmux.cpp
index 0102162..eac83c0 100644
--- a/components/esp_modem/src/esp_modem_cmux.cpp
+++ b/components/esp_modem/src/esp_modem_cmux.cpp
@@ -25,7 +25,7 @@ using namespace esp_modem;
  *        This is useful if upper layers expect the entire payload available
  *        for parsing.
  */
-#define DEFRAGMENT_CMUX_PAYLOAD
+//#define DEFRAGMENT_CMUX_PAYLOAD
 
 #define EA 0x01  /* Extension bit      */
 #define CR 0x02  /* Command / Response */
@@ -286,7 +286,7 @@ bool CMux::on_cmux_data(uint8_t *data, size_t actual_len)
         actual_len = term->read(data, data_to_read);
 #else
         data = buffer.get();
-        actual_len = term->read(data, buffer_size);
+        actual_len = term->read(data, buffer.size);
 #endif
```

hqs1, 2022-07-01:
@david-cermak Now after deleting the macro definition, there is no previous problem. Although it is said that because the serial port reads a large amount of data, the watchdog will be triggered, but it does not affect the network data transmission and the signal quality of the AT command to read the 4G modem.
[image]
[image]

hqs1, 2022-07-02:
@david-cermak  The problem seems to be solved, and now I've changed the modem to another model and the problem doesn't appear.

closed 2022-07-08 by hqs1