espressif/esp-protocols #59: Suggestion for custom commands

MisterScience5, 2022-06-17:
I am currently implementing the modem protocol on a SIM7070, and in my use case I would like for it to use the modems built in feature to check received APN on network connection, however this is not available in the library. I have been trying to add a custom command, but I am having some trouble with it. Therefore it would be nice if the library exposed a simple "custom command" that sends a desired string as a command, and simply returns the received data in a string, something like:

char return_string[50];
esp_modem_send_custom_command(dce, "AT+CGNAPN", &return_string);

Which would return for example "+CGNAPN: 1,telenor.iot".

This would make running a few custom commands very easy, at the expense of the user creating his own custom code to deal with the results.

bbinet, 2022-06-22:
I had the same needs, and I've implemented a "raw" command which may be useful to you.
I will submit a PR soon.

bbinet, 2022-06-22:
See PR #63

MisterScience5, 2022-06-22:
Excellent, thank you very much @bbinet :)

david-cermak, 2022-06-23:
https://github.com/espressif/esp-protocols/pull/63 has been merged

closed 2022-06-23 by david-cermak