espressif/esp-protocols #65: Ap_to_pppos example

vil-nakul, 2022-06-23:
AP_to_pppos example doesn't seem to work fully after enabling the debug logs, I get this: 

```
D (1132) memory_layout: Available memory region 0x40095e40 - 0x40096000
V (1138) memory_layout: Examining memory region 0x40096000 - 0x40098000
D (1145) memory_layout: Available memory region 0x40096000 - 0x40098000
V (1152) memory_layout: Examining memory region 0x40098000 - 0x4009a000
D (1159) memory_layout: Available memory region 0x40098000 - 0x4009a000
V (1165) memory_layout: Examining memory region 0x4009a000 - 0x4009c000
D (1172) memory_layout: Available memory region 0x4009a000 - 0x4009c000
V (1179) memory_layout: Examining memory region 0x4009c000 - 0x4009e000
D (1185) memory_layout: Available memory region 0x4009c000 - 0x4009e000
V (1192) memory_layout: Examining memory region 0x4009e000 - 0x400a0000
D (1199) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (1205) heap_init: Initializing. RAM available for dynamic allocation:
D (1213) heap_init: New heap initialised at 0x3ffae6e0
I (1218) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (1224) heap_init: New heap initialised at 0x3ffb7790
I (1229) heap_init: At 3FFB7790 len 00028870 (162 KiB): DRAM
I (1236) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1242) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (1249) heap_init: New heap initialised at 0x40095e40
I (1254) heap_init: At 40095E40 len 0000A1C0 (40 KiB): IRAM
V (1260) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1266) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1275) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (1280) FLASH_HAL: extra_dummy: 1
V (1283) memspi: raw_chip_id: 164020

V (1287) memspi: chip_id: 204016

V (1290) memspi: raw_chip_id: 164020

V (1294) memspi: chip_id: 204016

D (1298) spi_flash: trying chip: issi
D (1302) spi_flash: trying chip: gd
D (1305) spi_flash: trying chip: mxic
D (1309) spi_flash: trying chip: winbond
D (1313) spi_flash: trying chip: generic
I (1317) spi_flash: detected chip: generic
I (1322) spi_flash: flash io: dio
D (1326) cpu_start: calling init function: 0x4013a7cc
0x4013a7cc: _GLOBAL__sub_I__ZN9__gnu_cxx9__freeresEv at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_alloc.cc:348

D (1331) cpu_start: calling init function: 0x4013a57c
0x4013a57c: _GLOBAL__sub_I___cxa_get_globals_fast at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:145

D (1336) cpu_start: calling init function: 0x400e5f94
0x400e5f94: s_set_default_wifi_log_level at C:/Users/prath/esp/esp-idf/components/esp_wifi/src/wifi_init.c:63

D (1341) cpu_start: calling init function: 0x400da3fc
0x400da3fc: esp_ipc_init at C:/Users/prath/esp/esp-idf/components/esp_ipc/src/esp_ipc.c:102

D (1346) cpu_start: calling init function: 0x400d3094
0x400d3094: esp_reset_reason_init at C:/Users/prath/esp/esp-idf/components/esp_system/port/soc/esp32/reset_reason.c:68

D (1351) cpu_start: calling init function: 0x400d237c
0x400d237c: esp_ota_init_app_elf_sha256 at C:/Users/prath/esp/esp-idf/components/app_update/esp_app_desc.c:68

V (1356) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1363) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC02
D (1371) intr_alloc: Connected src 17 to int 3 (cpu 0)
V (1376) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1383) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (1391) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (1396) cpu_start: Starting scheduler on PRO CPU.
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (10) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (10) cpu_start: Starting scheduler on APP CPU.
D (1446) heap_init: New heap initialised at 0x3ffe0440
D (1446) heap_init: New heap initialised at 0x3ffe4350
V (1456) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1456) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1456) intr_alloc: Connected src 16 to int 12 (cpu 0)
I (1476) ap_to_pppos: Enabling PWRKEY
I (8576) ap_to_pppos: Enabling PWRKEY ... done
D (8576) partition: Loading the partition table
V (8576) calculated md5: 0x3ffba2b8   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|
V (8576) stored md5: 0x3f428070   f4 ad 4f 45 38 56 4b 5d  74 35 b6 2c 75 b6 95 24  |..OE8VK]t5.,u..$|
D (8586) partition: Partition table MD5 verified
D (8606) esp_netif_lwip: LwIP stack has been initialized
D (8606) esp_netif_lwip: esp-netif has been successfully initialized
D (8606) event: running task for loop 0x3ffbd0f8
D (8616) event: created task for loop 0x3ffbd0f8
D (8616) event: created event loop 0x3ffbd0f8
D (8626) esp_netif_objects: esp_netif_add_to_list 0x3ffbe04c
D (8626) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (8636) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffbe380
I (8646) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (8656) uart: queue free spaces: 30
V (8656) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (8656) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (8656) intr_alloc: Connected src 35 to int 13 (cpu 0)
V (8676) command_lib: set_pdp_context
V (8686) command_lib: generic_command_common
V (8686) command_lib: generic_command
D (8686) command_lib: generic_command command AT+CGDCONT=1,"IP","internet"
```

Why isn't there a DTE config for this example like the pppos _client example, I can't find where to set my TX and RX pins for the modem. I'm assuming this is the reason the modem isn't responding to any commands.
Pls help me with a fix or how to correctly configure this example.

Thanks,
Nakul

david-cermak, 2022-07-01:
@vil-nakul You can configure the pins directly in the code.

https://github.com/espressif/esp-protocols/blob/ef1bae5cdd65f10a6739c1b9b0aa3f0283f13ffa/components/esp_modem/examples/ap_to_pppos/main/network_dce.c#L21

```cpp
    // setup the DCE
    esp_modem_dte_config_t dte_config = ESP_MODEM_DTE_DEFAULT_CONFIG();
    dte_config.uart_config.tx_io_num = X;
    dte_config.uart_config.rx_io_num = Y;
```

you can also check this PR https://github.com/espressif/esp-protocols/pull/70 which adds more logs/debug info about connectivity issues.

david-cermak, 2022-07-26:
The linked PR has been merged. Please reopen if any other issues.

closed 2022-07-26 by david-cermak