espressif/esp-protocols #66: non-blocking API for AT Commands ?

franz-ms-muc, 2022-06-27:
Hello, 

the AT Command APIs seem to be blocking. 

is there a way to get non-Blocking API ?

Thanks,
Franz

david-cermak, 2022-07-01:
Hi @franz-ms-muc 

Do you have any proposal about the non-blocking API? For example in C++ we could simply write:
```cpp
std::future<bool> NetModule::async_sync()
{
    return std::async(&NetModule::sync, this);
}
```
Would you prefer C++ or plain C API? How should we post the results? callbacks, events, `std:future`, lambda?

VasilNikolovRilabs, 2022-07-06:
We prefer plain C API with callbacks or events(whatever is better for you)

david-cermak, 2022-09-09:
I've updated the default pppos example to demonstrate an async command style that are pure C and use one common callback.
PTAL at https://github.com/david-cermak/esp-protocols/commit/343c2cb23917ad1bb272dd81ccee33ee49db63b7
I don't think we should include this is as our official example, though.

david-cermak, 2023-03-27:
Closing as the command timeouts have been made configurable and asynchronicity could be achieved by user tasks.

closed 2023-03-27 by david-cermak