espressif/esp-protocols #71: esp-modem build problems

AshUK, 2022-07-01:
IDF: 4.4.1
Verison: 0.1.17
Tool Chain: esp-2021r2-patch3-8.4.0
Target: esp32

Hi we are attempting to migrate from the older version of the esp-modem to the latest version and have come across some build issues.

#### Unknown netif type
```
In file included from /Users/ash/dev/project/managed_components/espressif__esp_modem/include/esp_modem_api.h:19,
                 from /Users/ash/dev/project/components/cellular-test/src/cellular_test.c:11:
/Users/ash/dev/project/managed_components/espressif__esp_modem/include/esp_modem_c_api_types.h:66:116: error: unknown type name 'esp_netif_t'; did you mean 'esp_err_t'?
 esp_modem_dce_t *esp_modem_new(const esp_modem_dte_config_t *dte_config, const esp_modem_dce_config_t *dce_config, esp_netif_t *netif);
                                                                                                                    ^~~~~~~~~~~
                                                                                                                    esp_err_t
```

#### Invalid cast

```
/Users/ash/dev/project/components/cellular-test/src/cellular_test.c:180:10: warning: assignment to 'esp_modem_dce_t *' {aka 'struct esp_modem_dce_wrap *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
    p_dce = esp_modem_new_dev(ESP_MODEM_DCE_BG96, &dte_config, &dce_config, p_esp_netif);
```
Where p_dce is  'esp_modem_dce_t*'

### Missing functions
```
/Users/ash/dev/project/components/cellular-test/src/cellular_test.c: In function 'cellular_connect_task':
/Users/ash/dev/project/components/cellular-test/src/cellular_test.c:170:10: error: implicit declaration of function 'esp_modem_new_dev'; did you mean 'esp_modem_reset'? [-Werror=implicit-function-declaration]
  p_dce = esp_modem_new_dev(ESP_MODEM_DCE_BG96, &dte_config, &dce_config, p_esp_netif);
          ^~~~~~~~~~~~~~~~~
          esp_modem_reset
```

My cellular-test component has `esp_modem` in the PRIV_REQUIRES list, and I can see the component has been correctly configured in the managed_components directory.

david-cermak, 2022-07-01:
Hi @AshUK 

> migrate from the older version of the esp-modem to the latest version

Could you please tell us which was the *older* version? (just enough to post a commit id from this repo)

It seems like some implicit inclusion issue. You can check the current examples (assuming you're using C-API)

https://github.com/espressif/esp-protocols/blob/ef1bae5cdd65f10a6739c1b9b0aa3f0283f13ffa/components/esp_modem/examples/pppos_client/main/pppos_client_main.c#L10-L15

They are build in the CI against many IDF releases (including `v4.4`)

Will the compilation issue go away if you add
```
#include "esp_netif.h"
#include "esp_netif_ppp.h"
```

before including `esp_modem_api.h` ?

AshUK, 2022-07-01:
We originally used the esp-modem that was part of esp-idf.

`git show --quiet 5a41e6afa98ee0710ca75009d4d5360a1766544a^1`

```
commit 15fec25dfff04bc7328973d26d5e198942db68bc
Author: David Cermak <cermak@espressif.com>
Date:   Thu Nov 25 11:17:18 2021 +0100

    Docs: Reference esp_modem in libs/framweorks and migration guide
```

Here is my include order.

```
#include <stdbool.h>
#include <stdint.h>
#include <stdio.h>
#include <string.h>

// #define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE

#include "esp_err.h"
#include "esp_log.h"

#include "esp_modem_api.h"
#include "esp_modem_c_api_types.h"
#include "esp_modem_config.h"
#include "esp_modem_dce_config.h"

#include "esp_netif.h"
#include "esp_netif_defaults.h"
#include "esp_netif_ppp.h"
#include "esp_netif_types.h"

#include "freertos/portmacro.h"
#include "freertos/projdefs.h"
#include "freertos/task.h"

#include "soc/soc.h"
```

AshUK, 2022-07-01:
@david-cermak thank you, correcting the include order fixed the build issue.

AshUK, 2022-07-05:
@david-cermak Do you known which includes are required to access `esp_modem_set_pdp_context` using the C API or are you able to provide an example ?

Current include order is:

```
#include "esp_netif.h"
#include "esp_netif_defaults.h"
#include "esp_netif_types.h"

#include "esp_modem_api.h"
#include "esp_modem_c_api_types.h"
#include "esp_modem_config.h"
#include "esp_modem_dce_config.h"
```

david-cermak, 2022-07-05:
@AshUK I'm sorry, this API was missed in the C-API, added in #72

david-cermak, 2022-07-12:
Let me reopen this. Will close once we merge the linked PR.

The initial report (about the order of include headers) should be addressed too, btw (will add as a separate commit to the same PR)

PR #72 (2022-07-12): fix(esp-modem): Add missing set_pdp_context() to C-API
Closes #71