espressif/esp-protocols #94: PPP Events

AshUK, 2022-07-21:
We are testing reconnection handling on PPP and having issues detecting connection losses.

When the device disconnects we can observe `esp-netif_lwip-ppp: Connection lost` info log message however no status event is fired. It looks like internally the PPP disconnect is detected but not propagated.

We are able to trigger connection losses by using a faraday cage and  we are fortunate enough that our MVNO offer a sim reconnection facility via a portal.

We are subscribing to the PPP status events using the following.
```
esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, ppp_event_handler, NULL);
````

We do not wish to use MQTT or other app state to determine disconnect logic.

david-cermak, 2022-07-25:
Hi @AshUK 

Which version of IDF are you using? This should work correctly on latest `v5.0-dev` and `v4.4`, but I recall we had some issues with event propagation on some older branches.

AshUK, 2022-07-25:
We are targeting v4.4.1.

david-cermak, 2022-07-26:
Strange, this must work. Could you please share the initial part of setting up the `esp-netif` and modem's DTE?

Note, that there's also a run-time configuration in `esp_netif_ppp_config_t` that could enable/disable the phase/state events separately (but the error event should be enabled by default). Could you please try to call `esp_netif_ppp_config_t()` API after creating the interface:
```cpp
    esp_netif_config_t ppp_netif_config = ESP_NETIF_DEFAULT_PPP();
    esp_netif_t *ppp_netif = esp_netif_new(&ppp_netif_config);
    assert(ppp_netif);
    esp_netif_ppp_config_t ppp_config = { true, true };
    esp_netif_ppp_config_t(ppp_netif, &ppp_config);
```

AshUK, 2022-07-26:
Ahh this makes sense, thank you very much, now using `esp_netif_ppp_set_params` I am now seeing all the events come through as expected. The misnomer was were seeing `NETIF_PPP_ERRORUSER` errors on init as we always exit PPP mode regardless of config.

closed 2022-07-26 by AshUK

AshUK, 2022-07-27:
The flow of events for anyone reading this is the future. 

--- connection reset via MVNO portal ---
NETIF_PPP_PHASE_INITIALIZE
NETIF_PPP_PHASE_ESTABLISH
NETIF_PPP_PHASE_AUTHENTICATE
NETIF_PPP_PHASE_NETWORK
NETIF_PPP_PHASE_RUNNING
-- force a modem disconnect ---
NETIF_PPP_PHASE_NETWORK
NETIF_PPP_PHASE_ESTABLISH
NETIF_PPP_PHASE_DISCONNECT
NETIF_PPP_PHASE_DEAD