espressif/esp-protocols #97: Question about get_network_attachment_state

diplfranzhoepfinger, 2022-07-22:
Hello, 

if I am in CMUX Mode, and sending Data over the Data Channel. 
and i issue say a "get_network_attachment_state" ? 

will it issue the AT Command, wait for the Response, and then Continue with the Data Transmission ? 

OR does it issue the AT Command, continue the Data Transmission, and then get the Response ?

david-cermak, 2022-07-25:
In CMUX mode, we'd just send an AT command into the already running data stream. Both *data* and *command* transmission use their own specific virtual terminals, i.e. are tagged differently so the modem could tell them apart. The same applies for responses, so we could determine if the incoming data belong to PPP/Network and pass it to the networking stack, or if it's a reply from the issued AT command.

VasilNikolovRilabs, 2022-07-25:
Hello @david-cermak,

thank you for the clarification. If I correctly understand you - DATA and COM share the same physical layer(UART) but they are tagged differently and split respectively by the modem. 

Would that mean that a pending AT response(e.g taking 10 seconds or more like **CGATT?**) shall not block DATA transmission belonging to the PPP/Network and vice versa? Is that behavior the same in the SIMCOM - the 2 virtual channels work completely decoupled?

Best regards,
Vasil Nikolov

diplfranzhoepfinger, 2022-07-25:
Yes.

But as he already said:

We should better read Signal Strength instead of
get_network_attachment_state.

We could ask Simcom if get_network_attachment_state is blocking modem side
something.

But better read Signal Strength.

VasilNikolovRilabs, 2022-07-25:
Hello Franz and David,

reading the attachment state is just an example more less for an AT command, other commands will be issued as well in normal operation. What I would like to know is if both virtual channels work completely independent.