espressif/esp-protocols #102: NULL pointer at netif->ip_info

OckertM, 2022-07-29:
Hi all
I'm having huge trouble getting my SIM7600 working again after I updated ESP-IDF.
I'm getting a run time error because it's trying to load a null pointer (LoadProhibited). I pasted a snippet of the debug log output.

```
D (72461) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffc32c4
D (72468) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (72474) Modem: Modem has correctly entered DATA_MODE
D (78484) esp-netif_lwip-ppp: Phase Authenticate
D (78485) event: running post NETIF_PPP_STATUS:263 with handler 0x400da864 and context 0x3ffc4930 on loop 0x3ffbded4
0x400da864: esp_modem::Netif::on_ppp_changed(void*, char const*, long, void*) at C:/Users/Ockert/esp/Projects/esp_water_dispenser/components/espressif__esp_modem/src/esp_modem_netif.cpp:28

D (78486) esp-netif_lwip-ppp: Phase Network
D (78493) event: running post NETIF_PPP_STATUS:265 with handler 0x400da864 and context 0x3ffc4930 on loop 0x3ffbded4
0x400da864: esp_modem::Netif::on_ppp_changed(void*, char const*, long, void*) at C:/Users/Ockert/esp/Projects/esp_water_dispenser/components/espressif__esp_modem/src/esp_modem_netif.cpp:28

D (78555) esp_netif_lwip: esp_netif_internal_dhcpc_cb lwip-netif:0x3ffc3348
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400f9854  PS      : 0x00060130  A0      : 0x800f9979  A1      : 0x3ffc0070  
A2      : 0x3ffc3348  A3      : 0x00000000  A4      : 0x3ffc35d0  A5      : 0x0a060a06  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0xbd6f7764  A9      : 0x00000000  
A10     : 0x3ffc35d0  A11     : 0x3f411554  A12     : 0x3f411754  A13     : 0x000132db  
0x400f9854: esp_netif_internal_dhcpc_cb at C:/Users/Ockert/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1120

A14     : 0x3f411554  A15     : 0x3f412168  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff8  

Backtrace:0x400f9851:0x3ffc00700x400f9976:0x3ffc00c0 0x400fe84b:0x3ffc00e0 0x400fe8f6:0x3ffc0100 0x4010d899:0x3ffc0180 0x40115305:0x3ffc01b0 0x40117ea5:0x3ffc01d0 0x401180da:0x3ffc01f0 0x401149bf:0x3ffc0210 0x4010d669:0x3ffc0230 0x4010df64:0x3ffc0260 0x4010e0c1:0x3ffc0280 0x400fca75:0x3ffc02a0 0x400fcaf0:0x3ffc02c0 0x40089aa9:0x3ffc02f0 
0x400f9851: esp_netif_internal_dhcpc_cb at C:/Users/Ockert/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1117 (discriminator 9)

0x400f9976: netif_callback_fn at C:/Users/Ockert/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:117

0x400fe84b: netif_invoke_ext_callback at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/core/netif.c:1819

0x400fe8f6: netif_set_addr at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/core/netif.c:733

0x4010d899: sifaddr at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/netif/ppp/ppp.c:1065

0x40115305: ipcp_up at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/netif/ppp/ipcp.c:2036

0x40117ea5: fsm_rconfack at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/netif/ppp/fsm.c:496

0x401180da: fsm_input at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/netif/ppp/fsm.c:360

0x401149bf: ipcp_input at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/netif/ppp/ipcp.c:693

0x4010d669: ppp_input at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/netif/ppp/ppp.c:934

0x4010df64: pppos_input at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/netif/ppp/pppos.c:577

0x4010e0c1: pppos_input_sys at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/netif/ppp/pppos.c:448 (discriminator 3)

0x400fca75: tcpip_thread_handle_msg at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:174

0x400fcaf0: tcpip_thread at C:/Users/Ockert/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:148

0x40089aa9: vPortTaskWrapper at C:/Users/Ockert/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

```

It seems like the value in NETIF is null (or very close to null for esp_netif->ip_info_old) if I log the pointer address (outside range  0x3fxxxxxx - 0x6xxxxxxx like [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited) suggest.
```
    esp_netif_ip_info_t *ip_info = esp_netif->ip_info;
    esp_netif_ip_info_t *ip_info_old = esp_netif->ip_info_old;
```

I was on version 0.1.12 of the esp_modem component and switched between 0.1.18 and 0.1.12 to see if it might be fixed in the newer component with no luck.

I also tried checking out previous versions of the IDF master branch that I know was working flawlessly for me ~2 weeks before with 
```
cd $IDF_PATH
git fetch
git checkout [commit_id]
git pull
git submodule update --init --recursive
```
and I still got the same issue. I also tried checking out v5.0 as the last merge was 5th of July (at this point) which should have worked for me but it didn't.

I also needed to add `esp_event esp_netif` to the esp_modem's dependencies for it to compile.

What did I mess up? 
Did I not checkout correctly?
Is there a bug?
What can I do to help?

Please let me know, I will gladly supply more information!
Thanks for making epic products and a great framework!

david-cermak, 2022-07-29:
Hi @OckertM 

I'm sorry, there have been some incompatibilities with both IDF and this repo, trying the latest versions, as of today, should help with both issues:
1) IDF: incorrect esp-netif ptr for PPP interfaces after updating to newer lwip (reported in https://github.com/espressif/esp-idf/issues/9345, fixed in https://github.com/espressif/esp-idf/commit/64f4f0ac1e58bd59b715caef27ee9157e39d2aaf)
2) esp-protocols: Dependency update per IDFv5.0, fixed in https://github.com/espressif/esp-protocols/commit/010f98ca80512a17824483fda10aa31de1de4d49

OckertM, 2022-07-29:
Hi @david-cermak 
Thanks for the speedy reply!

I updated everything and it is working great again!
Thank you and the rest of the team for doing epic work and listening and replying to our issues.
Have a great day!

closed 2022-07-29 by OckertM