espressif/esp-protocols #103: esp_modem - cant exit CMUX on BG96

chegewara, 2022-08-01:
Hi,
esp_modem example works fine to enter CMUX on BG96, but i am having problem to exit CMUX and switch to command mode.
For some reason my BG96 is not responding on close DLCI0 command.
```
V (543) command_lib: set_cmux
V (543) command_lib: generic_command_common
V (543) command_lib: generic_command
D (543) command_lib: generic_command command AT+CMUX=0

D (553) command_lib: Response: 
OK

V (653) CMUX Received: 0x3ffc0d10   f9 03 73 01 d7 f9                                 |..s...|
D (653) CMUX: Payload frame: dlci:00 type:73 payload:0 available:2
V (663) CMUX Received: 0x3ffc0d10   f9 07 73 01 15 f9                                 |..s...|
D (663) CMUX: Payload frame: dlci:01 type:73 payload:0 available:2
V (673) CMUX Received: 0x3ffc0d10   f9 0b 73 01 92 f9                                 |..s...|
D (673) CMUX: Payload frame: dlci:02 type:73 payload:0 available:2
V (683) command_lib: set_echo
V (683) command_lib: generic_command_common
V (693) command_lib: generic_command
D (693) command_lib: generic_command command ATE0

V (703) CMUX Received: 0x3ffc0d10   f9 05 ef 0d 0d 0a 4f 4b  0d 0a 5f f9              |......OK.._.|
D (713) CMUX: Payload frame: dlci:01 type:ef payload:6 available:8
D (713) command_lib: Response: 
OK

V (723) command_lib: set_pdp_context
V (723) command_lib: generic_command_common
V (723) command_lib: generic_command
D (733) command_lib: generic_command command AT+CGDCONT=1,"IP","plus"

V (763) CMUX Received: 0x3ffc0d10   f9 05 ef 0d 0d 0a 4f 4b  0d 0a 5f f9              |......OK.._.|
D (763) CMUX: Payload frame: dlci:01 type:ef payload:6 available:8
D (773) command_lib: Response: 
OK

V (773) command_lib: set_data_mode
V (773) command_lib: generic_command
D (783) command_lib: generic_command command ATD*99##

V (793) CMUX Received: 0x3ffc0d10   f9 05 ef 2b 0d 0a 43 4f  4e 4e 45 43 54 20 31 35  |...+..CONNECT 15|
V (793) CMUX Received: 0x3ffc0d20   30 30 30 30 30 30 30 0d  0a 83 f9 f9 01 ff 09 e3  |0000000.........|
V (803) CMUX Received: 0x3ffc0d30   05 07 89 8f f9                                    |.....|
D (813) CMUX: Payload frame: dlci:01 type:ef payload:21 available:33
D (823) command_lib: Response: 
CONNECT 150000000

D (823) CMUX: Payload frame: dlci:00 type:ff payload:4 available:6
Modem has correctly entered multiplexed command/data mode
V (833) command_lib: get_operator_name
V (843) command_lib: generic_get_string
V (853) CMUX Received: 0x3ffc0d10   f9 09 ef 3b 0d 0a 2b 43  4f 50 53 3a 20 30 2c 30  |...;..+COPS: 0,0|
V (853) CMUX Received: 0x3ffc0d20   2c 22 50 6c 75 73 22 2c  30 0d 0a 0d 0a 4f 4b 0d  |,"Plus",0....OK.|
V (863) CMUX Received: 0x3ffc0d30   0a 18 f9                                          |...|
D (873) CMUX: Payload frame: dlci:02 type:ef payload:29 available:31
} (873) command_lib: Token: {

V (883) command_lib: Token: {+COPS: 0,0,"Plus",0}

} (883) command_lib: Token: {

V (893) command_lib: Token: {OK}

Operator name:"Plus",0
V (893) command_lib: get_imsi
V (893) command_lib: generic_get_string
V (903) command_lib: generic_get_string
V (913) CMUX Received: 0x3ffc0d10   f9 09 ef 33 0d 0a 32 36  30 30 31 31 35 30 34 36  |...3..2600115046|
[... 99 lines cut ...]
V (17773) CMUX Received: 0x3ffc0d10   f9 01 ff 09 e3 05 07 09  8f f9                    |..........|
D (17773) CMUX: Payload frame: dlci:00 type:ff payload:4 available:6
V (18073) CMUX Received: 0x3ffc0d10   f9 01 ff 09 e3 05 07 09  8f f9                    |..........|
D (18073) CMUX: Payload frame: dlci:00 type:ff payload:4 available:6
V (18413) CMUX Received: 0x3ffc0d10   f9 01 ff 09 e3 05 07 09  8f f9                    |..........|
D (18413) CMUX: Payload frame: dlci:00 type:ff payload:4 available:6
I (24583) esp-netif_lwip-ppp: User interrupt
I (24583) esp_modem_netif: PPP state changed event 5
V (24583) CMUX Received: 0x3ffc0d10   f9 07 73 01 15 f9                                 |..s...|
D (24583) CMUX: Payload frame: dlci:01 type:73 payload:0 available:2
V (24603) CMUX Received: 0x3ffc0d10   f9 0b 73 01 92 f9                                 |..s...|
D (24603) CMUX: Payload frame: dlci:02 type:73 payload:0 available:2
E (25623) cmux_example: Failed to configure COMMAND_MODE command mode... exiting
```
Maybe i am missing something. This is code i am using to switch CMUX->COMMAND mode:
```
    if (dce->set_mode(esp_modem::modem_mode::COMMAND_MODE)) {
        std::cout << "Modem has correctly entered COMMAND_MODE mode" << std::endl;
    } else {
        ESP_LOGE(TAG, "Failed to configure COMMAND_MODE command mode... exiting");
        return;
    }
```

Thanks

chegewara, 2022-08-01:
Im not sure, but this may be related to still maintained DATA/PPP connection on DLCI1.

david-cermak, 2022-08-07:
Thanks for the report!
I don't have the device with me to test it at this moment; and unfortunately cannot tell from the log what's wrong.

I think that your initial thought about the DLCI0's failure to exit seems plausible, and I recall that I have tested this code with BG96, so the only idea at this point would be timeouts. There's a small bug in the CMUX exit sequence:

https://github.com/espressif/esp-protocols/blob/1ffc20c8e30f19a06b32969a75539810ddc6cb14/components/esp_modem/src/esp_modem_cmux.cpp#L326-L360

Could you please try to zero out the timeout counter, adding this line:
```cpp
timeout = 0;
```
just after these two lines, here:

https://github.com/espressif/esp-protocols/blob/master/components/esp_modem/src/esp_modem_cmux.cpp#L332

and (more importantly) here:

https://github.com/espressif/esp-protocols/blob/master/components/esp_modem/src/esp_modem_cmux.cpp#L347

as it looks like it was the ` send_disconnect(0);` that failed.

>  this may be related to still maintained DATA/PPP connection on DLCI1.

This should work as (per spec) the DATA mode should gracefully exit for all muxed terminals when terminating CMUX (and seemed to work per the log:
```
I (24583) esp_modem_netif: PPP state changed event 5
```

chegewara, 2022-08-08:
Thanks @david-cermak , your answer helped me to clean my mind and to find few bugs, which are preventing to switch CMUX to COMMAND mode on BG96.
Here is summary. 1 low level bug(timeout counter) and 2 important bugs:
- adding missing `timeout = 0;` in this line
https://github.com/espressif/esp-protocols/blob/1ffc20c8e30f19a06b32969a75539810ddc6cb14/components/esp_modem/src/esp_modem_cmux.cpp#L348
- wrong command to close DLCI0 is sent, so no answer from BG96:
https://github.com/espressif/esp-protocols/blob/1ffc20c8e30f19a06b32969a75539810ddc6cb14/components/esp_modem/src/esp_modem_cmux.cpp#L85
here is BG96 docs page 30
https://www.quectel.com/wp-content/uploads/2021/03/Quectel_BG96_MUX_Application_Note_V1.0.pdf
[image]

- and last issue, probably this line should also unlock semaphore on type == 0xEF
https://github.com/espressif/esp-protocols/blob/1ffc20c8e30f19a06b32969a75539810ddc6cb14/components/esp_modem/src/esp_modem_cmux.cpp#L135

Now i can easy switch from CMUX to COMMAND.

Again, thanks for help

chegewara, 2022-08-08:
There is one more thing i am struggling with. The answer from BG96 is not always sent by modem or not received by library, because after reset my code, when UART is init, i am getting partially response packet:
```
W (1540) command_lib: generic_command command AT+CMUX=0

W (1540) command_lib: Response: 
OK

I (530) uart: queue free spaces: 30
data len: 2, type: 0xef, dlci: 0
packet: c1 01 
```
As we can see modem is responding on `AT+CMUX=0` so it is no longer in CMUX mode, but it cant enter new CMUX mode until i reset BG96.

Any help or advice will be appreciated 

Thanks

chegewara, 2022-08-08:
I am suspecting it is something with BG96.
When i am trying to switch CMUX => COMMAND pretty much after few seconds, it fail (no response from modem), but when i leave it in CMUX for some time (few dozens seconds) it works fine. At least it is my current status, but it also may be a random response from modem.

```
I (704840) esp-netif_lwip-ppp: User interrupt
I (704840) esp_modem_netif: PPP state changed event 5
I (704960) CLOSE packet: f9 03 ef 05 c3 01 f2 f9 
I (704960) CLOSE response data: c1 01 
```

david-cermak, 2022-08-15:
Thanks @chegewara for taking a deep dive into the CMUX protocol and proposing fixes! I'll try to reproduce the problems and test it with other devices. (please feel free to post a PR btw)
Interestingly, CMUX mode switching works okay with my BG96 on current master.

>  may i have one more suggestion; please dont return false in case when COMMAND mode is requested and it is already set up, or in similar situations:

This can be done, but would break the current API. No problem with that, just wanted to bring this up.

> When i am trying to switch CMUX => COMMAND pretty much after few seconds,

This is probably because the CMUX mode implementation (here in `esp-modem`) automatically enters the DATA mode via the second virtual terminal and therefore it has to exit this mode too (which generally means to wait for a user interrupt from PPP protocol and that may take up to several seconds).
(Again, tried to test it with my BG96 and seems to work okay even if I tried to exit the CMUX before if fully established the PPP connection)

chegewara, 2022-08-15:
Hi,
thanks for update.
Unfortunately i have to stick to arduino (idf 4.4).
Maybe i didnt explain it properly:
- i am enabling CMUX on modem
- connecting with PPP, confirmed with IP address from network
- now, when i try to switch after few seconds then it stuck with timeout, but when i wait few dozens then im not getting timeout

About that breaking change, its not a problem, i can change it on my side. I am also willing to admit its problem with my BG96, if its working for you, i know BG96 may be problematic depending on firmware on modem.

Thanks for help

david-cermak, 2022-08-19:
@chegewara Have created https://github.com/espressif/esp-protocols/pull/115 to fix the clear issues you reported -- Thanks again for providing hints and suggestions! Verified on my devices (BG96 and SIM7600).

> when i try to switch after few seconds then it stuck with timeout, but when i wait few dozens then im not getting timeout

Still wasn't able to reproduce that.

> About that breaking change...

I think that your proposal makes sense and will eventually update the state machine to return successfully when we're already in that state. I'd just like to fix the immediate problems in the linked PR. I would appreciate if you could take a look at the PR and help with the review as well.

closed 2022-08-29 by david-cermak