espressif/esp-protocols #104: Error while building project with dependency `esp_websocket_client==0.0.2`

abobija, 2022-08-04:
Hi. I'm using `ESP-IDF v5.0-dev-4547-gfde4afc67a` and I have installed esp_websocket_client (latest version 0.0.2) via dependency manager. When I try to build project I'm getting error:

```
.../managed_components/espressif__esp_websocket_client/include/esp_websocket_client.h:16:10: fatal error: esp_event.h: No such file or directory
   16 | #include "esp_event.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
HINT: Please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'.
```
I have check that in [6e4e4fa](https://github.com/espressif/esp-protocols/commit/6e4e4fab1d2c6c63c796a81b5dce33d4d33892e4#diff-683fe540d9e42e2d8a0c607edecc803e1c3eaed7986dbd27261b47633c77ac12R12) esp_event has been added to REQUIRES in CMakeLists.txt but new package version with the latest changes is not published to IDF Component Registry yet.

Please make the new version of esp_websocket_client and push it to the Component Registry so we can use esp_websocket_client in our projects.

@suren-gabrielyan-espressif sorry for tag but maybe you can publish the new version of package?

Thanks in advance.
Cheers.

gabsuren, 2022-08-05:
@abobija thank you for the suggestion good point! 
I submitted [PR](https://github.com/espressif/esp-protocols/pull/105) for new version  and it will be merged soon.

abobija, 2022-08-05:
@gabsuren thank you so much for the fast response! 👍🏼

abobija, 2022-08-05:
I see, version 0.0.3 is already live on Component Registry. I have try it and now I can build my project!
Thanks again for the fast response and support!

closed 2022-08-05 by abobija