espressif/esp-protocols #108: Example Request: PPP and c++ API

AshUK, 2022-08-09:
Are you able to create an example of using the C++ API with PPP as netif ? I have tried myself many times but just end up with build errors or crashes.

david-cermak, 2022-08-15:
Just to better understand your request, are you asking for a general PPP example, for example like this one:

https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/pppos_client

but using C++ API of `esp-modem` ?

AshUK, 2022-08-15:
Yeah that is correct.

david-cermak, 2022-09-09:
I have created a new example `pppos_client_cxx` in https://github.com/david-cermak/esp-protocols/tree/feat/modem_simple_cxx_demo/components/esp_modem/examples/pppos_client_cxx
but not sure if we want to keep it here as an official example. Will perhaps add it later, when demonstrating a new feature or custom settings.

This branch is just FYI. I've migrated the C-API to C++API in a separate commit to show the differences. Please check https://github.com/david-cermak/esp-protocols/commit/f933950914cb899821d9ed7d04b153b61b0e2ff6

david-cermak, 2022-10-21:
I'm closing this as we don't want to maintain two official examples that do the same thing. Hope the posted project helped, but will not merge it. We're adding two more examples soon, so will make sure to use C++ API, at least for one.

closed 2022-10-21 by david-cermak