espressif/esp-protocols #118: CMUX: Protocol mismatch: Missed leading SOF, recovering..." warning in esp_modem

DarshanDodal, 2022-08-21:
Hello,
I am facing an issue with the "esp_modem" component. I am getting a warning " CMUX: Protocol mismatch: Missed leading SOF, recovering..." and the success rate of connection after this warning is 60%, 40% of times it does not connect. I have tried increasing the rx tx buffer, but still no luck. And, even if connection happens, I get "Protocol Recovered" log very very rarely.

P.S. I am on the latest commit.

@david-cermak

david-cermak, 2022-08-22:
@DarshanDodal Could you please share more details:
* Which device/modem are you using?
* Debug (or better verbose) logs of one occurrence of the `Protocol mismatch` error (For example the log shared here https://github.com/espressif/esp-protocols/issues/46#issuecomment-1144798695 helped me to fix the issue without a real device)

Could you please also try to remove or comment this line?

https://github.com/espressif/esp-protocols/blob/8f00fc182a139155261599c740149e655b809263/components/esp_modem/src/esp_modem_cmux.cpp#L28

to see if anything (e.g. the success ration) changes?

DarshanDodal, 2022-08-22:
Hello,
I am using Simcom A7672S. 

After commenting the said line, I am getting an error. Looks like the "buffer_size" variable used in esp_modem_cpp at line 289, is nowhere declared.  

I shall be sharing the logs in some time.

david-cermak, 2022-10-25:
Hi @DarshanDodal 

I was able to test this on the real device, A7670E and it seems to me that it's an issue with their CMUX mode implementation. I've added 28de345 to fix this (WIP, not sure if we're going to include in the official library). Note that the modem replies with
```
f9 ef c3 f2 f9
```
apparently missing the address field of the CMUX protocol.

(Also the `DEFRAGMENT_CMUX_PAYLOAD` option has been added to the component configuration menu and is available for users to choose if needed)

DarshanDodal, 2022-10-26:
@david-cermak In my case, the issue was that I was not waiting for the modem to boot completely, it was still in the process of sending the SIM READY,A SMS READY, PB READY msgs. I was able to resolve the issue by adding a small delay at startup and then calling the CMUX function.

david-cermak, 2022-10-26:
@DarshanDodal Oh, thanks for the update! I think it could be closed then.

closed 2022-10-26 by david-cermak

Gurupawaskar, 2023-09-17:
Anyone have  schematic diagram and pcb of "A7672S SIMCOM"