espressif/esp-protocols #124: Modem Console Sample crashing

franz-ms-muc, 2022-08-24:


franz-ms-muc, 2022-08-24:
Latest master (by 24.08.2022) is crashing the Modem Console Example. 

ESP-IDF v4.4.2

Log: 

```
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 14:39:50
I (29) boot: chip revision: 0
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00177000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=1befch (114428) map
I (117) esp_image: segment 1: paddr=0002bf24 vaddr=3fc92a30 size=02810h ( 10256) load
I (119) esp_image: segment 2: paddr=0002e73c vaddr=40374000 size=018dch (  6364) load
I (124) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=7a8dch (501980) map
I (221) esp_image: segment 4: paddr=000aa904 vaddr=403758dc size=0d150h ( 53584) load
I (233) esp_image: segment 5: paddr=000b7a5c vaddr=50000000 size=00010h (    16) load
I (239) boot: Loaded app from partition at offset 0x10000
I (239) boot: Disabling RNG early entropy source...
I (253) cpu_start: Pro cpu up.
I (253) cpu_start: Starting app cpu, entry point is 0x40375374
0x40375374: call_start_cpu1 at /home/franz/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (232) cpu_start: App cpu up.
I (268) cpu_start: Pro cpu start user code
I (268) cpu_start: cpu freq: 160000000
I (268) cpu_start: Application information:
I (270) cpu_start: Project name:     modem_console
I (276) cpu_start: App version:      v0.5-25-gfb63164
I (282) cpu_start: Compile time:     Aug 24 2022 14:40:03
I (288) cpu_start: ELF file SHA256:  05eb8f74fdc40952...
I (294) cpu_start: ESP-IDF:          v4.4.2
I (299) heap_init: Initializing. RAM available for dynamic allocation:
I (306) heap_init: At 3FC96DA0 len 00049260 (292 KiB): D/IRAM
I (312) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (319) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (325) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (332) spi_flash: detected chip: gd
I (336) spi_flash: flash io: dio
W (340) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (353) sleep: Configure to isolate all GPIO pins in sleep state
I (360) sleep: Enable automatic switching of GPIO sleep configuration
I (367) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (387) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (387) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2397) modem_pwkey: status = 1
I (2407) uart: queue free spaces: 30
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42006e38  PS      : 0x00060c33  A0      : 0x82006f85  A1      : 0x3fcf34b0  
0x42006e38: find_desc_for_source at /home/franz/esp-idf-v4.4.2/components/esp_hw_support/intr_alloc.c:161

A2      : 0x0000001c  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00060c23  
A6      : 0x00000002  A7      : 0xff000000  A8      : 0x82062dbc  A9      : 0x00000000  
A10     : 0x3fce5498  A11     : 0x3fce5498  A12     : 0x3fcf3434  A13     : 0x00000000  
A14     : 0x3fcf09b0  A15     : 0x00000005  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x82062dbc  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe  

Backtrace:0x42006e35:0x3fcf34b00x42006f82:0x3fcf34d0 0x420071bf:0x3fcf3510 0x4200739d:0x3fcf3550 0x42017801:0x3fcf3580 0x42018759:0x3fcf35c0 0x4200b656:0x3fcf3610 0x4200aaae:0x3fcf3ad0 0x4200a68c:0x3fcf3bc0 0x4200864d:0x3fcf3c00 0x42007d2d:0x3fcf3f70 0x4207a5d5:0x3fcf3f90 0x4037da91:0x3fcf3fb0 
[... 20 lines cut ...]
0x4200864d: modem_console_main at /home/franz/git/krone/pppos_client_sample/modem_console/components/modem_console/modem_console_main.cpp:79

0x42007d2d: app_main at /home/franz/git/krone/pppos_client_sample/modem_console/main/main.c:22

0x4207a5d5: main_task at /home/franz/esp-idf-v4.4.2/components/freertos/port/port_common.c:141 (discriminator 2)

0x4037da91: vPortTaskWrapper at /home/franz/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 05eb8f74fdc40952

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037592c
0x4037592c: esp_restart_noos at /home/franz/esp-idf-v4.4.2/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1650
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
entry 0x403c9954
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
```

franz-ms-muc, 2022-08-24:
the Strange thing is that the cmux\_sample does run.

david-cermak, 2022-08-25:
@franz-ms-muc Are you running just the clean console example without a modification? Do you use any specific configuration (IDF or project)?

> Latest master (by 24.08.2022) is crashing the Modem Console Example. 
> ESP-IDF v4.4.2

Have just tested the latest master 8f00fc182a139155261599c740149e655b809263 with IDF `v4.4.2` https://github.com/espressif/esp-idf/commit/1b16ef6cfc2479a08136782f9dc57effefa86f66 and works as expected.

---
As for the reported crash, the exception comes from this line

https://github.com/espressif/esp-idf/blob/300a4a16a5d63d937f9db31a7668bcd7e8d89d74/components/esp_hw_support/intr_alloc.c#L111

trying to dereference a linked list of interrupt vector from

https://github.com/espressif/esp-idf/blob/300a4a16a5d63d937f9db31a7668bcd7e8d89d74/components/esp_hw_support/intr_alloc.c#L93

which at seems to point to an address similar to this number:
```
EXCVADDR: 0x82062dbc
```
and that's not a valid pointer. It looks like a memory corruption of the interrupt table.

franz-ms-muc, 2022-08-29:
clean console example without a modification:

brings: 
CMake Error at /home/franz/esp-idf-v4.4.2/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cdc_acm_host'.

franz-ms-muc, 2022-08-29:
try same with removing USB-Stack. 
does compile. 
but same Crash.

franz-ms-muc, 2022-08-29:
Done some Tests. 
the Crash comes from the new Exception Handling ... 
i will keep researching.

franz-ms-muc, 2022-08-29:
If i add this to skdconfig.
https://github.com/espressif/esp-idf/blob/master/examples/cxx/exceptions/sdkconfig.defaults
it does not crash anymore !

franz-ms-muc, 2022-08-29:
changed the Title to reflect the Problem more precise.

david-cermak, 2022-09-08:
> Done some Tests. the Crash comes from the new Exception Handling ... i will keep researching.

Is it the same crash as you initially posted? Coming from  the uart terminal constructor when registering a uart interrupt ?
```
uart_driver_install at /home/franz/esp-idf-v4.4.2/components/driver/uart.c:1622
```

franz-ms-muc, 2022-09-08:
Thanks so much

Holen Sie sich Outlook für Android<https://aka.ms/AAb9ysg>

[https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]

Meisterschulen am Ostbahnhof
Mühldorfstr. 6
81671 München

Tel.: 089 416002 - 0
Fax: 089 416002 - 111

Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

david-cermak, 2022-09-09:
Hi @franz-ms-muc 

I would still like to better understand this issue.

**So the steps to reproduce this is to**
1) Remove the `components/esp_modem/examples/modem_console/components`
2) Compile with the default config (exceptions are disabled by default) and esp32s3
3) Run the app
...and It crashes with the backtrace posted [above](https://github.com/espressif/esp-protocols/issues/124#issuecomment-1240772502)

**Compilation issue: `Failed to resolve component 'cdc_acm_host'`**

The example should compile clean for all targets both with and without the USB host component (on `v4.4.2`) Could you please share more details how you build it? (The usual trouble is that sometimes people switch between different IDF versions and keep multiple repos of different versions so the build system might reference one for generation step and another for compilation...?)

diplfranzhoepfinger, 2022-10-11:
> Hi @franz-ms-muc
>
> I would still like to better understand this issue.
>
> **So the steps to reproduce this is to**
>
> 1. Remove the `components/esp_modem/examples/modem_console/components`
> 2. Compile [...]

do not understand your Question.

david-cermak, 2022-10-13:
Sorry that wasn't very clear. I was just asking for more details to be able to reproduce (*and fix*) this issue.

So far, we've been discussing these two problems:
1) The crash if we just run the default example (provided exceptions are off)
2) Compilation problem -- component dependency

Let's start with `2)` as it's very simple and we should be able to recreate it very quickly. Could you please provide the steps to reproduce? As I wasn't able -- and maybe it's been fixed in the meantime (per changes in the USB component)

As for the `1)` I've tried the steps described above, but no luck so far. Is there any specific config you're using? (except for exceptions:-) Or any kind of update/modification of that example? Or just anything that could help me to locate the problem?

franz-ms-muc, 2022-10-24:
OK, i will make an Update to latest ESP-Modem and try out.

diplfranzhoepfinger, 2022-11-22:
so, with #178 this issue seems also fixed.

closed 2022-11-22 by franz-ms-muc