espressif/esp-protocols #133: Support for 4G Modem with USB interface

rmshub, 2022-09-04:
Is esp-modem supports usb interface to connect with 4G modem using PPP for ESP32-S3 device

abhi-ratnman, 2022-09-04:
I've the same question. Please let me know if possible @team

rmshub, 2022-09-07:
Hi Team,

Any update on this request ?

david-cermak, 2022-09-08:
The console example supports USB interface, please check

https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console#usb-dte-support

It's just a sub-component to this example at this moment. We're working on making this an independent component, deployable separately in the component manager.

abhi-ratnman, 2022-09-10:
@david-cermak thanks for the update. How long should it take for the independent component? As we are moving with production code. So we should wait or move forward with the sub-component?

tore-espressif, 2022-09-12:
@abhi-ratnman @rmshub Could you please take a look at https://github.com/espressif/esp-protocols/pull/89 ? And let me know whether this changes fit your needs?

Very early version of USB_DTE plugin is uploaded [here](https://components.espressif.com/component/espressif/esp_modem_usb_dte), and source code is under review https://github.com/espressif/idf-extra-components/pull/92. I'd appreciate any ideas!

abhi-ratnman, 2022-10-03:
I checked the component and example. Is there support for other modules? The code is for BG96. How to add support for SIM7672? @david-cermak @tore-espressif

abhi-ratnman, 2022-10-04:
`     struct esp_modem_usb_term_config usb_config = ESP_MODEM_DEFAULT_USB_CONFIG(0x1e0e, 0x9011);
    usb_config.interface_idx = 3;
    const esp_modem_dte_config_t dte_config = ESP_MODEM_DTE_DEFAULT_USB_CONFIG(usb_config);  
    ESP_LOGI(TAG, "Waiting for USB device connection...");
    auto dte = create_usb_dte(&dte_config);
    std::unique_ptr<DCE> dce = create_SIM7600_dce(&dce_config, dte, esp_netif);
`

Used the above config. Still the code is stuck in waiting for USB Device connection. The interface number in the sheet is 1 but tried from 0-3 still it is not getting connected.

The hardware connection is good as i have tested the same connection with [https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module](url) this code. And it gets connection (in menuconfig selected sim7600 as the board.)

tore-espressif, 2022-10-05:
@abhi-ratnman could you share full device and configuration descriptors? I will take a look

tore-espressif, 2022-10-05:
@abhi-ratnman In case you missed it:
[image]

abhi-ratnman, 2022-10-05:
@tore-espressif which device and configuration descriptors are required?

Also the reset part is taken care of and i try restarting it manually in case not done automatically. I've tested the usb CDC code  like that only and it works fine.

abhi-ratnman, 2022-10-06:
[USB DESC.txt](https://github.com/espressif/esp-protocols/files/9726233/USB.DESC.txt)
 

Please find the above attached file

abhi-ratnman, 2022-10-08:
Hi @tore-espressif any update? Do u want any more information?

tore-espressif, 2022-10-10:
> Please find the above attached file @tore-espressif

please use `usb_config.interface_idx = 5;`, the rest of your configuration seems correct.

abhi-ratnman, 2022-10-10:
@tore-espressif checked this as well the code is still stuck at the same place. Is there any way to change the USB in and out address as per mentioned in this issue https://github.com/espressif/esp-iot-solution/issues/121#issuecomment-1021137502

abhi-ratnman, 2022-10-11:
`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1664
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f74
entry 0x403b6248
I (24) boot: ESP-IDF v4.4.1-dirty 2nd stage bootloader
I (25) boot: compile time 16:16:10
I (25) boot: chip revision: 0
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=23b8ch (146316) map
I (119) esp_image: segment 1: paddr=00033bb4 vaddr=3fc932a0 size=029e4h ( 10724) load
I (121) esp_image: segment 2: paddr=000365a0 vaddr=40374000 size=09a78h ( 39544) load
I (133) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=861f8h (549368) map
I (232) esp_image: segment 4: paddr=000c6220 vaddr=4037da78 size=05820h ( 22560) load
I (237) esp_image: segment 5: paddr=000cba48 vaddr=50000000 size=00010h (    16) load
I (244) boot: Loaded app from partition at offset 0x10000
I (244) boot: Disabling RNG early entropy source...
I (261) cpu_start: Pro cpu up.
I (261) cpu_start: Starting app cpu, entry point is 0x403753cc
0x403753cc: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_system/port/cpu_start.c:160

I (0) cpu_stDrt: A1pp cusu:up.
_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (268) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 13 bit
D (283) clk: RTC_SLOW_CLK calibration value: 3626547
I (292) cpu_start: Pro cpu start user code
I (292) cpu_start: cpu freq: 160000000
I (292) cpu_start: Application information:
I (295) cpu_start: Project name:     modem-console
I (300) cpu_start: App version:      afafcb7-dirty
I (306) cpu_start: Compile time:     Oct 11 2022 16:16:02
I (312) cpu_start: ELF file SHA256:  41dca3480fadb3cf...
I (318) cpu_start: ESP-IDF:          v4.4.1-dirty
V (323) memory_layout: reserved range is 0x3c0b3b8c - 0x3c0b3bac
D (329) memory_layout: Checking 5 reserved memory ranges:
D (335) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc93298
D (341) memory_layout: Reserved memory range 0x3fc932a0 - 0x3fc97868
D (347) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
D (354) memory_layout: Reserved memory range 0x40374000 - 0x40383298
0x40374000: _WindowOverflow4 at C:/Espressif/frameworks/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_vectors.S:1736

D (360) memory_layout: Reserved memory range 0x600fe000 - 0x600fe000
D (367) memory_layout: Building list of available memory regions:
V (373) memory_layout: Examining memory region 0x40374000 - 0x40378000
0x40374000: _WindowOverflow4 at C:/Espressif/frameworks/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_vectors.S:1736

0x40378000: esp_flash_erase_region at C:/Espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/esp_flash_api.c:540 (discriminator 2)

V (379) memory_layout: Region 0x40374000 - 0x40378000 inside of reserved 0x40374000 - 0x40383298
0x40374000: _WindowOverflow4 at C:/Espressif/frameworks/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_vectors.S:1736

0x40378000: esp_flash_erase_region at C:/Espressif/frameworks/esp-idf-v4.4.1/components/spi_flash/esp_flash_api.c:540 (discriminator 2)

0x40374000: _WindowOverflow4 at C:/Espressif/frameworks/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_vectors.S:1736

V (388) memory_layout: Examining memory region 0x3fc88000 - 0x3fc90000
V (395) memory_layout: Region 0x3fc88000 - 0x3fc90000 inside of reserved 0x3fc84000 - 0x3fc93298
[... 270 lines cut ...]
D (6095) cdc_acm: Checking list of connected USB devices
D (6145) cdc_acm: Checking list of connected USB devices
D (6195) cdc_acm: Checking list of connected USB devices
D (6245) cdc_acm: Checking list of connected USB devices
D (6295) cdc_acm: Checking list of connected USB devices
D (6345) cdc_acm: Checking list of connected USB devices
D (6395) cdc_acm: Checking list of connected USB devices
D (6445) cdc_acm: Checking list of connected USB devices
D (6495) cdc_acm: Checking list of connected USB devices
D (6545) cdc_acm: Checking list of connected USB devices
D (6595) cdc_acm: Checking list of connected USB devices
D (6645) cdc_acm: Checking list of connected USB devices
D (6695) cdc_acm: Checking list of connected USB devices
D (6745) cdc_acm: Checking list of connected USB devices
D (6795) cdc_acm: Checking list of connected USB devices
D (6845) cdc_acm: Checking list of connected USB devices
D (6895) cdc_acm: Checking list of connected USB devices
D (6945) cdc_acm: Checking list of connected USB devices
D (6995) cdc_acm: Checking list of connected USB devices
D (7045) cdc_acm: Checking list of connected USB devices
D (7095) cdc_acm: Checking list of connected USB devices
D (7145) cdc_acm: Checking list of connected USB devices
D (7195) cdc_acm: Checking list of connected USB devices
D (7245) cdc_acm: Checking list of connected USB devices
D (7295) cdc_acm: Checking list of connected USB devices`

abhi-ratnman, 2022-10-13:
@tore-espressif @david-cermak  any clue from the logs? Actually we are running late for the integration in our product. Hence I request if you can escalate the issue. Thanks :)

tore-espressif, 2022-10-13:
Hi @abhi-ratnman , I just got mine SIMCom SIM7600E-H.
It is a little different from the module that you have. My PID, VID is: `0x1E0E:0x9001`. I'm using endpoints 0x86 (IN) and 0x04 (OUT), which belong to interface no 3.

Please note that we updated the USB-DTE a little in past weeks, but the changes should not affect your usecase.
I'm working with:
- `esp_modem` version 0.1.23
- `esp_modem_usb_dte` version 1.0.0
- `usb_host_cdc_acm` version 1.0.3
- `esp-idf` master branch

Everything is working as expected with this initialization code:
``` cxx
struct esp_modem_usb_term_config usb_config = ESP_MODEM_DEFAULT_USB_CONFIG(0x1e0e, 0x09001, 3); // VID, PID and interface num of SIM7600E-H modem
const esp_modem_dte_config_t dte_config = ESP_MODEM_DTE_DEFAULT_USB_CONFIG(usb_config);  
ESP_LOGI(TAG, "Waiting for USB device connection...");
auto dte = create_usb_dte(&dte_config);
std::unique_ptr<DCE> dce = create_SIM7600_dce(&dce_config, dte, esp_netif);
```
Example output:
[image]

> any clue from the logs?

From the logs I can see two things:
1. You aren't using the latest `modem_console` example. Try updating your local branch. Then configure the project for esp32s3, USB DTE and change only the configuration of USB-DTE
2. The USB library detects connection of a new device. But fails to open it. Updating to the latest code should fix your problem

abhi-ratnman, 2022-10-13:
I updated to the latest esp-protocols something strange is happening even if i set target to esp32s3 the target is set to esp32c3.

Do not know the reason for that. Earlier when i would set target to esp32s3 it would set but in the latest update 

`C:\Espressif\frameworks\esp-idf-v4.4.1\projects\esp-protocols\components\esp_modem\examples\modem_console>idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'c:\espressif\frameworks\esp-idf-v4.4.1\projects\esp-protocols\components\esp_modem\examples\modem_console\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory c:\espressif\frameworks\esp-idf-v4.4.1\projects\esp-protocols\components\esp_modem\examples\modem_console\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=1 c:\espressif\frameworks\esp-idf-v4.4.1\projects\esp-protocols\components\esp_modem\examples\modem_console"...
-- Found Git: C:/Espressif/frameworks/tools/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/frameworks/tools/Espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/frameworks/tools/Espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/frameworks/tools/Espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
Skipping optional dependency: espressif/esp_modem_usb_dte
Solving dependencies requirements
Updating lock file at C:/Espressif/frameworks/esp-idf-v4.4.1/projects/esp-protocols/components/esp_modem/examples/modem_console\dependencies.lock
Processing 2 dependencies:
[1/2] espressif/esp_modem (0.1.23)
[2/2] idf (4.4.1)
Skipping optional dependency: espressif/esp_modem_usb_dte
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-v4.4.1/projects/esp-protocols/components/esp_modem/examples/modem_console/sdkconfig
Loading defaults file C:/Espressif/frameworks/esp-idf-v4.4.1/projects/esp-protocols/components/esp_modem/examples/modem_console/sdkconfig.defaults...
-- Found Python3: C:/Espressif/frameworks/tools/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "modem-console" version: 511ed54
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/projects/esp-protocols/components/esp_modem/examples/modem_console/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib-time.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.4.1/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32c3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_modem esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: C:/Espressif/frameworks/esp-idf-v4.4.1/components/app_trace C:/Espressif/frameworks/esp-idf-v4.4.1/components/app_update C:/Espressif/frameworks/esp-idf-v4.4.1/components/asio C:/Espressif/frameworks/esp- [... cut ...] nity C:/Espressif/frameworks/esp-idf-v4.4.1/components/usb C:/Espressif/frameworks/esp-idf-v4.4.1/components/vfs C:/Espressif/frameworks/esp-idf-v4.4.1/components/wear_levelling C:/Espressif/frameworks/esp-idf-v4.4.1/components/wifi_provisioning C:/Espressif/frameworks/esp-idf-v4.4.1/components/wpa_supplicant
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Espressif/frameworks/esp-idf-v4.4.1/projects/esp-protocols/components/esp_modem/examples/modem_console/build
`

tore-espressif, 2022-10-13:
Ah, I'm sorry about that, that;s a regression from https://github.com/espressif/esp-protocols/pull/155. Please remove this line https://github.com/espressif/esp-protocols/blob/master/components/esp_modem/examples/modem_console/sdkconfig.defaults#L4

abhi-ratnman, 2022-10-13:
@tore-espressif  thanks for a prompt reply

Yes now it is solved but the earlier issue still remains in the latest code. As soon as USB is connected the SIM module gets powered off and it gets restart again and again. 

While the same connection works fine with https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/host/usb_cdc_4g_module this code. But the issue in this code is that I want to access CMUX but that is not implemented in the above example.

tore-espressif, 2022-10-13:
@abhi-ratnman the unexpected resets might come from incorrect/insufficient power supply. Please check that your power supply has enough power for powering the SIM7672.

From the setup in the photo, I can see the USB VBUS signal is invalid too. ESP32-S3 devkit does NOT output +5V on the USB cable, so your VBUS line is not powered. Try connecting +5v on the VBUS line on SIM7672 USB connector.

abhi-ratnman, 2022-10-13:
@tore-espressif 

For power supply a separate 5v 2A SMPS is used to power the SIM module which provides sufficient power to the module. 

> @abhi-ratnman the unexpected resets might come from incorrect/insufficient power supply. Please check that your power supply has enough power for powering the SIM7672.

> From the setup in the photo, I can see the USB VBUS signal is invalid too. ESP32-S3 devkit does NOT output +5V on the USB cable, so your VBUS line is not powered. Try connecting +5v on the VBUS line on SIM7672 USB connector.

The same hardware connection work with the esp-iot-solution code as per the link provided. So if that would have been the issue it should occur there too?

tore-espressif, 2022-10-18:
Hi @abhi-ratnman , our support options are running short :( I have a little different modem than you have (although, the USB interface should be the same), with different firmware version. Unfortunately, I can't reproduce the issue.

Now, all we can do is (1)suggest a fix and (2)_you_ have to try it and repeat until the issue is solved.

Have you tried connecting +5V on the modem's USB connector? 

> The same hardware connection work with the esp-iot-solution code as per the link provided. So if that would have been the issue it should occur there too?

I _guess_ that the modem is not properly enumerated by the ESP32-S3 because it fails to reconnect after _USB 2nd reset_.
_USB 2nd reset_ is a standard thing in enumeration process; every USB Host (Windows, Linux PCs) does it. But, the USB host implementation in the code that you're linking does _not_ do it, simply because it is a pre-release version of Espressif's USB host.

abhi-ratnman, 2022-10-23:
Hi @tore-espressif have tried with 5v usb supply still issue is the same

guianmon99, 2023-01-10:
HI @tore-espressif , I  have a similar issue I am using the usb cdc_acm_host library to connect to a hl8548 from sierra wl. I am trying to use the cdc_acm_host_open() function to connect to it and send AT commands. The function works and i am able to connect to the endpoints but when i send some AT commands the module doesn't respond back. I am using the esp_idfV4.4.2. 
 we have been testing with the BG96 module and all works perfectly. could there be an issue with the fact that this module uses interface association ? I think the function I am suing supports it but I am not sure.

guianmon99, 2023-01-10:
@tore-espressif this are the deescriptors from the module I am using. I have tried to connect to all interfaces 
Bus 001 Device 032: ID 1519:0020 Comneon HSIC Device
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x1519 Comneon
  idProduct          0x0020 HSIC Device
  bcdDevice           17.29
  iManufacturer           1 Comneon
  iProduct                2 7 CDC-ACM
  iSerial                 3 014869005829421
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x01e7
    bNumInterfaces         15
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              5
      CDC Header:
        bcdCDC               1.20
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x07
          sends break
          line coding and serial state
          get/set/clear comm features
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
[... 458 lines cut ...]
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber       14
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1
      bInterfaceProtocol      0
      iInterface             25
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x8f  EP 15 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval               1

tore-espressif, 2023-03-16:
Hello,
We have recently fixed several enumeration issues in `esp-idf` master. There are also few bugfixes in [esp_modem_usb_dte v1.1.0](https://components.espressif.com/components/espressif/esp_modem_usb_dte). All the fixes are already implemented in esp_modem examples.
Could you please update your esp-idf and esp_modem and try again?

closed 2023-05-04 by tore-espressif

gidlstar, 2025-02-01:
Hi, I have the same issue that the modem won't connect. It hangs on "Waiting for USB device connection...".
I am using the latest IDF and Component versions. The Modem is a sim7600ce-t and the following is the lsusb output of it. I used the pppos example and basically just changed the menuconfig to usb and the modem to sim7600. I checked if the vID pID are correct and they seem to be alright (vid 0x1e0e, pid 0x9001). The default setting uses the interface 3 but I've testet with all the 6 possible interfaces. 
Wiring and power supply are alright, and if I use the UART config everything is working as expected. But I would need the higher througput so USB is the was to go. 

Anyone have an idea what could be the problem?

Bus 001 Device 013: ID 1e0e:9001 Qualcomm / Option SimTech, Incorporated
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e0e Qualcomm / Option
  idProduct          0x9001 
  bcdDevice            3.18
  iManufacturer           1 SimTech, Incorporated
  iProduct                2 SimTech, Incorporated
  iSerial                 3 0123456789ABCDEF
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0102
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
[... 195 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)