espressif/esp-protocols #143: Example Request: C API - waiting of incoming call

RaD, 2022-09-16:
How to set read_cb from C code in order to check +CLIP output?

david-cermak, 2022-09-20:
It's not possible to set `on_read_cb` from C API yet.

As a workaround you can check what's done in the `mdoem-console` example. It implements a *generic command* that takes a pattern matching string as an argument, so you can wait for a ready signal, or some specific string (such as `+CLIP:` in your case), e.g.
```
esp> cmd -n "" -t 50000 -p "SIM"
I (442984) modem_console: Sending command  with timeout 50000
I (459154) modem_console: 
RDY

+CME ERROR: SIM not inserted

esp>
```

https://github.com/espressif/esp-protocols/blob/b8d1e58778d89c5db5dea80dcbc5d5eaa63aafe9/components/esp_modem/examples/modem_console/main/modem_console_main.cpp#L228-L238

Is it better for you to be able to set the actual callback and check the results manually? Or alternatively extend the existing `at_command()` to allow for the pattern matching in replies?
The first option is more universal, but wouldn't support sending any other `AT` commands while it's active (Unless you use one more virtual terminal)

RaD, 2022-09-22:
Thanks, will try to implement a solution...

RaD, 2022-09-25:
I have tried to send empty command:

```
W (115445) INET: Waiting Caller ID failed, retry...
W (125445) INET: Waiting Caller ID failed, retry...
W (135445) INET: Waiting Caller ID failed, retry...
W (145445) INET: Waiting Caller ID failed, retry...
W (148515) command_lib: RING found
E (148515) ESP_MODEM_THROW: esp_modem_dte.cpp:60 ERROR

abort() was called at PC 0x400e295e on core 0

Backtrace:0x40081c5a:0x3ffb12e00x40086901:0x3ffb1300 0x4008e31e:0x3ffb1320 0x400e295e:0x3ffb1390 0x400e2643:0x3ffb14c0 0x400e8aec:0x3ffb14f0 0x400ded76:0x3ffb1550 0x400dc9f9:0x3ffb1570 0x400d875d:0x3ffb15c0 0x400d75cb:0x3ffb1620 0x4013ec37:0x3ffb1650 0x4008987d:0x3ffb1670

```

david-cermak, 2022-09-30:
> E (148515) ESP_MODEM_THROW: esp_modem_dte.cpp:60 ERROR

We had a similar issue once. It's been fixed in a8714730fb3edd2616bb9d9cb60a03a67c1dc368

Could you please check that your version includes this commit?

RaD, 2022-10-01:
this bug was gone accidently, sorry

closed 2022-11-23 by david-cermak