espressif/esp-protocols #154: Unable to add PPP support for other modems

Tater-Ctrl, 2022-10-06:
So I've got this FiPy I've wanted to try and hook up using the pppos examples, I've tried some older versions of the pppos-client example and gotten the IP connected where it reports back the IP address, but then I get stuck with DNS lookup failed err=202 res=0x0 and couldn't find a way past it.

david-cermak, 2022-10-07:
>  I get stuck with DNS lookup failed err=202 res=0x0 and couldn't find a way past it.

Do you use just one network interface (the PPP only) or are you also connecting via WiFi or Ethernet at the same time?

> So I wanted to try this newest version, and I've not really found how to add another DCE configuration here, and simply changing the pins with the BG96 gives me signal failed with 263, switching to the generic one gives me a stack-overflow error.

* You can increase the stack size too, it's also a menuconfig option.
*  As for the "signal failed with 263" -- have you initialized both `esp_event` (by creating the default event loop) and the `esp_netif` components? (in other words, are you running the original examples or have you made some updates?)

david-cermak, 2023-10-31:
Closing as it's not an issue of this repository, but should be fixed in IDF.

Note that we doesn't have a solution for the DNS issue yet, but the workaround is demonstrated in this example:

https://github.com/espressif/esp-protocols/tree/master/examples/esp_netif/multiple_netifs

closed 2023-10-31 by david-cermak