espressif/esp-protocols #159: Strange Echo on SIM7070

diplfranzhoepfinger, 2022-10-11:


diplfranzhoepfinger, 2022-10-11:
Normal: 

```

esp> cmd AT
I (14062) modem_console: Sending command AT with timeout 1000
I (14062) modem_console: AT
OK

esp> cmd AT
I (15232) modem_console: Sending command AT with timeout 1000
I (15242) modem_console: AT
OK

esp> cmd AT
I (16172) modem_console: Sending command AT with timeout 1000
I (16182) modem_console: AT
OK

esp> cmd AT
I (16982) modem_console: Sending command AT with timeout 1000
I (16992) modem_console: AT
OK

esp> cmd AT
I (17622) modem_console: Sending command AT with timeout 1000
I (17622) modem_console: AT
OK

esp> cmd AT
I (18332) modem_console: Sending command AT with timeout 1000
I (18342) modem_console: AT
OK

I (18992) modem_console: Sending command AT with timeout 1000
I (19002) modem_console: AT
OK

esp> cmd AT
I (19652) modem_console: Sending command AT with timeout 1000
I (19652) modem_console: AT
OK

esp> 
```

diplfranzhoepfinger, 2022-10-11:
[image]

diplfranzhoepfinger, 2022-10-11:
it seems like the Buffer is not processed one time ?

diplfranzhoepfinger, 2022-10-12:
throw away the Data is wrong for sure. 

maybe the solution would be that "arbitrary Data" is collected by a Callback. 

OR: 

we have a option to say "we expect Number of \\r" like 5 or so for multiline respones ... 

i will think more about it.

franz-ms-muc, 2022-10-12:
see here: 

[image]

even with 30000000 timeout it immediately return after the OK, and then we see a Echo of the Sent command. ... 
shouldnt the RX and TX be seperate ?

david-cermak, 2022-10-13:
> even with 30000000 timeout it immediately return after the OK, and then we see a Echo of the Sent command. ...

This console command looks for the `OK` by default, but you can supply your own pattern to check for:
```bash
esp>cmd AT+NETOPEN -p +NETOPEN:
```
(example for SIM7600 to open the network in command mode, this returns `OK` immediately and, after a couple of second the actual network status)

closed 2023-03-24 by david-cermak