espressif/esp-protocols #160: Using the asio component in esp-idf v5.0-beta1-720-gda9a78ebfc is not possible (IDFGH-8515)

chrta, 2022-10-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-720-gda9a78ebfc

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

I expect that the asio component is available to be used by the build system, including the header and source code from asio itself.

### What is the actual behavior?

The folder `managed_components/espressif__asio/asio/` is empty and `idf.py build` reports that the registered include directory is not a directory, see build logs below.

### Steps to reproduce.

1. Create a simple c++ project
2. Add asio dependency via `idf.py add-dependency espressif/asio^1.0.2` as described on https://components.espressif.com/components/espressif/asio
3. `idf.py build`

### Build or installation Logs.

```plain
$ cd prj
$ idf.py build
...
Processing 2 dependencies:
[1/2] espressif/asio (1.0.2)
[2/2] idf (5.0.0)
-- Project sdkconfig file /home/user/projects/esp/sip_call/sdkconfig
Loading defaults file /home/user/projects/esp/sip_call/sdkconfig.defaults...
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/user/.espressif/python_env/idf5.0_py3.10_env/bin/python (found version "3.10.6") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "sip-client" version: 8cea32b-dirty
-- Adding linker script /home/user/projects/esp/sip_call/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/user/projects/esp/esp-idf/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script /home/user/projects/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script /home/user/projects/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script /home/user/projects/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script /home/user/projects/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script /home/user/projects/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script /home/user/projects/esp/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script /home/user/projects/esp/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld
CMake Error at /home/user/projects/esp/esp-idf/tools/cmake/component.cmake:312 (message):
  Include directory
  '/home/user/projects/esp/sip_call/managed_components/espressif__asio/asio/asio/include'
  is not a directory.
Call Stack (most recent call first):
  /home/user/projects/esp/esp-idf/tools/cmake/component.cmake:486 (__component_add_include_dirs)
  managed_components/espressif__asio/CMakeLists.txt:26 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/home/user/projects/esp/sip_call/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/user/projects/esp/sip_call/build/log/idf_py_stderr_output_98387 and /home/user/projects/esp/sip_call/build/log/idf_py_stdout_output_98387
```

### More Information.

The dowload archive at https://components.espressif.com/components/espressif/asio https://components-file.espressif.com/0806398a-355d-4b0e-b4e5-22da7e2fc383.tgz also has an empty asio folder.
There is probably a issue with git submodules and the new idf component management/downloading/packaging.

I am trying to upgrade https://github.com/chrta/sip_call to newest commit of esp-idf in  release/v5.0 branch, but this issue blocks me.

gabsuren, 2022-10-17:
The issue resolved, in order to use new package please use this command:

idf.py add-dependency espressif/asio^1.14.1~3

closed 2022-10-17 by gabsuren

chrta, 2022-10-17:
Thx for the fast response. It works as expected, now.