espressif/esp-protocols #167: Enter CMUX, Leave CMUX, Enter CMUX again

diplfranzhoepfinger, 2022-10-18:
Hi, 

how about enter CMUX mode, leave it, and enter it again ? 

should work ? 
i tested it on a SIM7070 and it seems not working. 

[log1.txt](https://github.com/espressif/esp-protocols/files/9810310/log1.txt)
[log2.txt](https://github.com/espressif/esp-protocols/files/9810311/log2.txt)

diplfranzhoepfinger, 2022-10-18:
@david-cermak my main Question is: 
is this working with other Modems and ESP-Modem ? 
so must we check more on ESP-Modem side, or more on Simcom side ?

david-cermak, 2022-10-18:
I've tested that on SIM7600 and BG96 and it seems to be working.

Also from what I see in the logs, it's also working for you if you enable verbose logging (`log2.txt`, there's an `ERROR` reply from the modem in one place, but that's expected as we've already entered data mode in the previous step and that step is ignored)
Maybe it's failing due to some timing condition for your device. Does it work if you add say 200ms delay between switching modes?

franz-ms-muc, 2022-10-18:
```
I (00:00:29.485) esp-netif_lwip-ppp: User interrupt
I (00:00:29.486) esp_modem_netif: PPP state changed event 5
Modem has correctly entered command mode
E (00:00:40.655) cmux_example: Failed to configure multiplexed command mode... exiting

```

franz-ms-muc, 2022-10-18:
so in INFO Log Level not working. added 1 Second Sleep ...

diplfranzhoepfinger, 2022-10-18:
[log3.txt](https://github.com/espressif/esp-protocols/files/9811371/log3.txt)
in Verbose mode same, 1 s delay not helping anything !

diplfranzhoepfinger, 2022-10-18:
[image]

diplfranzhoepfinger, 2022-10-19:
Question: 

do you need to stop the netif and start the netif again ? 

@david-cermak

david-cermak, 2022-10-21:
> Question:
> 
> do you need to stop the netif and start the netif again ? 
> 

No need to do that, that should work automatically. I think I'll order `SIM7070` and test it.

diplfranzhoepfinger, 2022-10-21:
> > Question:  
> > do you need to stop the netif and start the netif again ?
> 
> No need to do that, that should work automatically. I think I'll order `SIM7070` and test it.

yes, that would be most great. 

then we are on the same Page.

david-cermak, 2022-10-25:
Note that if reconnecting to PPPoS fails with `NO CARRIER` or similar, it usually helps to *hang-up* the current connection by sending `ATH\r` (will try to incorporate this to some retry mechanism in this library)

david-cermak, 2023-01-04:
Closing as this has been fixed in https://github.com/espressif/esp-protocols/pull/169