espressif/esp-protocols #168: more Control over CMUX virtual Channels

diplfranzhoepfinger, 2022-10-19:
the Issue 
https://github.com/espressif/esp-protocols/issues/167
was about enter CMUX and leave CMUX.

but this was not the right Question. Or at least not getting to the point too much.

if we enter CMUX Mode we can have up to 5 Virtual Channels according to CMUX Spec. 
as i understood, ESP-Modem does establish 2 channels, 1 AT Channel, and one Data Channel. 

but now `+PPP disconnect` comes in as a URC, then this cannel is in command mode. 
so we now have 2 Channels in Command mode. 

so 

1. ... how to get more info about each channel ? 
2. ... how to switch this channel again into Data Mode ?

diplfranzhoepfinger, 2022-10-19:
@david-cermak

diplfranzhoepfinger, 2022-10-19:
other Question: 

is there a way to pro-actively shut down the DataMode (PPP Session) on one Channel without leaving CMUX, and again enter Data mode later ?

RilabsAutomotive, 2022-10-19:
This issue is also observed on our side. We do the following steps:
Enter Command mode
Enter CMux mode and and connect to internet successfully
Simulate disconnection event
Re-Enter command mode 
Enter Data Mode and and connect to internet successfully
send AT command => does not work any more. Seems like the port is used for Data only instead of AT commands

So after we enter CMUX mode, we need to have control over the 2 ports in order to be able to switch between Command and Data mode

david-cermak, 2022-10-21:
That was the original design, but was simplified in f3ff98bb828cb7c45b9749e15ff66df3ce245db4 to automatically maintain DATA mode in one virtual terminal when in CMUX. I would like to provide more control of the individual terminals in CMUX, but that would complicate the API. I think the main reason for using this auto CMUX mode was, that for the devices I tested seamlessly exited (all possible) data mode(s) when closing CMUX mode.

> This issue is also observed on our side. We do the following steps:
> 1) Enter Command mode
> 2) Enter CMux mode and and connect to internet successfully
> 3) Simulate disconnection event
> 4) Re-Enter command mode
> 5) Enter Data Mode and and connect to internet successfully

@RilabsAutomotive about the step `5)`: Why entering Data mode here, if you wanted to use AT commands? Would it work if you enter `CMUX` mode?

RilabsAutomotive, 2022-10-21:
Right, so the re-entering in Data mode in step 5, we tried in order to retrigger the PPP to establish and authenticate.
SIMCOM support explained that CMUX is entered only once after start-up to create the virtual ports. Then we shall only switch between Command and Data mode in order to re-establish connection via PPP.

david-cermak, 2022-10-21:
> SIMCOM support explained that CMUX is entered only once after start-up 

Does that mean that your device doesn't support runtime exit of CMUX?

I'll try to think about how we can extend the interface and provide better control over these virtual terminals.

diplfranzhoepfinger, 2022-10-21:
> > SIMCOM support explained that CMUX is entered only once after start-up
> 
> Does that mean that your device doesn't support runtime exit of CMUX?

exit was possible, but re-Enter not. 

> I'll try to think about how we can extend the interface and provide better control over these virtual terminals.

diplfranzhoepfinger, 2022-10-21:
also the Topic is: if a PPPos Session on Data terminates, the Channel switch to command. (see initial Post above)

david-cermak, 2022-10-21:
> also the Topic is: if a PPPos Session on Data terminates, the Channel switch to command. (see initial Post above)

Good point, thanks! Will have to do something about re-entering the DATA mode if it spontaneously exits. On my devices it's usually enough to transition `CMUX` -> `CMD` and back `CMD` -> `CMUX` (the first transition assures that the PPPoS graciously shutdowns on these devices)

closed 2022-11-23 by david-cermak