espressif/esp-protocols #170: Timeout and RTS/CTS vs DTR config on LILYGO® TTGO T-SIM7000G

nicklasb, 2022-10-28:
Dear community/Espressif!

I am trying to get esp-modem to work with my board, and I am a bit confused by the configuration. 
When I tried TinyGSM (which worked), I did not have to set these pins, but here i have to set RTC/CTS.

As far as I can see, no such pins are designated on the board, is that because it doesn't have HW flow control?
https://www.aliexpress.com/item/4000571249574.html?spm=a2g0o.9042311.0.0.31624c4dHiqssj

Either way it seems like the SIM7000G chip should be supported by this library, so I am not sure why it doesn't work.
I think I have set all pins correctly and feel that I have almost tried everything. 
Yes, I have yelled at the card, as required, but to no avail. :-)

Now I'd like to be frank and admit that I am by no means  an UART-expert, so it is quite possible that this is a very dumb question, but perhaps someone can point me in the right direction?

Kind regards, Nicklas

espressif-abhikroy, 2022-10-31:
Hi,
Make sure you have the following pins connected correctly.
* Tx
* Rx
* 5V
* GND
That’s all that is needed. You can comment out RTS/CTS pin configuration.
Optionally you can also connect the PWRKEY pin.

Before trying your own example, please try esp-protocols/components/esp_modem/examples/modem_console.
In case the example crashes try increasing EXAMPLE_MODEM_UART_EVENT_TASK_STACK_SIZE and/or ESP_MAIN_TASK_STACK_SIZE via ```idf.py menuconfig```.

Once you’re successful with the modem_console example, try running ```esp_modem_sync(dce)``` a few times before ```esp_modem_get_signal_quality(dce, &rssi, &ber)```.

Also, make sure the baud rate is configured correctly.

nicklasb, 2022-10-31:
Hi and thanks @espressif-abhikroy !

I didn't think remember that it took so long starting up, running esp_modem_sync() until it worked was the solution, and I realized it by using the console and sending a couple of ATs. Great tip!

(BTW, this card has a power pin that has to be set, perhaps that could be handled by the library and examples as well?)

Thanks again, 
Nicklas

closed 2022-10-31 by nicklasb