espressif/esp-protocols #175: usb 4G modem usb_host_install occurred panic_abort

wishinlife, 2022-11-15:
I use example pppos_client and open CONFIG_EXAMPLE_SERIAL_CONFIG_USB on esp-idf release/v5.0 panic error.
The error occurred in the statement `_esp_modem_new_dev_usb_`

dependencies:
  espressif/esp_modem: "^0.1.25"
  espressif/esp_modem_usb_dte:
    version: "^1.0.0"

Logs:
```
I (1227) BLE: BLE Host Task Started
I (1232) PPP_CLIENT: Initializing esp_modem for the BG96 module...
I (1237) PPP_CLIENT: Waiting for USB device connection...
I (1242) NimBLE: GAP procedure initiated: stop advertising.

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42033dfd  PS      : 0x00060530  A0      : 0x8202b066  A1      : 0x3fcf39c0  
0x42033dfd: usb_host_install at /home/zhuxy/esp/esp-idf/components/usb/usb_host.c:441 (discriminator 2)

A2      : 0x00000000  A3      : 0x3c0e71cc  A4      : 0x3fcb4f5c  A5      : 0x3fcb4fb4  
A6      : 0x00000105  A7      : 0xff000000  A8      : 0x82033d63  A9      : 0x3fcf39a0  
A10     : 0x00000105  A11     : 0x3c0e71f0  A12     : 0x3fcb5040  A13     : 0x00000105  
A14     : 0x00000000  A15     : 0x3fcb5034  SAR     : 0x00000001  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x42033dfa:0x3fcf39c0 0x4202b063:0x3fcf3a20 0x4202b423:0x3fcf3c00 0x4202aa90:0x3fcf3c30 0x4202a792:0x3fcf3c70 0x4200b69b:0x3fcf3cb0 0x4200a417:0x3fcf3e20 0x420adac7:0x3fcf3e50 0x40385921:0x3fcf3e80
0x42033dfa: usb_host_install at /home/zhuxy/esp/esp-idf/components/usb/usb_host.c:441 (discriminator 2)
0x4202b063: esp_modem::UsbTerminal::UsbTerminal(esp_modem_dte_config const*) at /home/zhuxy/esp-projects/IoT-Station/managed_components/espressif__esp_modem_usb_dte/esp_modem_usb.cpp:59
0x4202b423: std::_MakeUniq<esp_modem::UsbTerminal>::__single_object std::make_unique<esp_modem::UsbTerminal, esp_modem_dte_config const*&>(esp_modem_dte_config const*&) at /home/zhuxy/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/11.2.0/bits/unique_ptr.h:962
 (inlined by) esp_modem::create_usb_terminal(esp_modem_dte_config const*) at /home/zhuxy/esp-projects/IoT-Station/managed_components/espressif__esp_modem_usb_dte/esp_modem_usb.cpp:179
0x4202aa90: esp_modem::create_usb_dte(esp_modem_dte_config const*) at /home/zhuxy/esp-projects/IoT-Station/managed_components/espressif__esp_modem_usb_dte/esp_modem_usb_api_target.cpp:17
0x4202a792: esp_modem_new_dev_usb at /home/zhuxy/esp-projects/IoT-Station/managed_components/espressif__esp_modem_usb_dte/esp_modem_usb_c_api.cpp:19
0x4200b69b: ppp_client_init at /home/zhuxy/esp-projects/IoT-Station/main/src/ppp_client.c:205 (discriminator 13)
0x4200a417: app_main at /home/zhuxy/esp-projects/IoT-Station/main/main.c:254 (discriminator 2)
0x420adac7: main_task at /home/zhuxy/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)
0x40385921: vPortTaskWrapper at /home/zhuxy/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```

david-cermak, 2022-11-22:
@wishinlife which modem are you using? Is it really BG96 as this line suggests?

```
I (1232) PPP_CLIENT: Initializing esp_modem for the BG96 module...
```

Could you please check that the config of VID and PID are correct for your device?

https://github.com/espressif/esp-protocols/blob/3f0262dfe5aa3861f56185f931ff872c33f8c03e/components/esp_modem/examples/pppos_client/main/pppos_client_main.c#L213-L214

The actual issue with the load prohibited exception is in error handling of usb host component, but the initial problem comes from initializing `usbh` and one potential reason could be incorrect configuration.

wishinlife, 2022-11-23:
I use EC200S， VID and PID refer to the linux driver of EC200S, and i tested USB interface number from 0-4, and it was the same fault. 
This code is modified from the official example program, I forgot to change the print statement content.

`void ppp_client_init(void)
{
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed, NULL));

    /* Configure the PPP netif */
    esp_modem_dce_config_t dce_config = ESP_MODEM_DCE_DEFAULT_CONFIG(CONFIG_EXAMPLE_MODEM_PPP_APN);
    esp_netif_config_t netif_ppp_config = ESP_NETIF_DEFAULT_PPP();
    esp_netif_t *esp_netif = esp_netif_new(&netif_ppp_config);
    assert(esp_netif);

    event_group = xEventGroupCreate();
    while (1) {
        ESP_LOGI(TAG, "Initializing esp_modem for the EC200S module...");
        struct esp_modem_usb_term_config usb_config = ESP_MODEM_DEFAULT_USB_CONFIG(0x2C7C, 0x6026, 3); // VID, PID and interface num of EC200 modem
        const esp_modem_dte_config_t dte_usb_config = ESP_MODEM_DTE_DEFAULT_USB_CONFIG(usb_config);
        ESP_LOGI(TAG, "Waiting for USB device connection...");
        esp_modem_dce_t *dce = esp_modem_new_dev_usb(ESP_MODEM_DCE_GENETIC, &dte_usb_config, &dce_config, esp_netif);
        assert(dce);
        esp_modem_set_error_cb(dce, usb_terminal_error_handler);
        vTaskDelay(pdMS_TO_TICKS(1000)); // Although the DTE should be ready after USB enumeration, sometimes it fails to respond without this delay
    xEventGroupClearBits(event_group, CONNECT_BIT | GOT_DATA_BIT | USB_DISCONNECTED_BIT);

    int rssi, ber;
    esp_err_t err = esp_modem_get_signal_quality(dce, &rssi, &ber);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_get_signal_quality failed with %d %s", err, esp_err_to_name(err));
        return;
    }
    ESP_LOGI(TAG, "Signal quality: rssi=%d, ber=%d", rssi, ber);

    err = esp_modem_set_mode(dce, ESP_MODEM_MODE_DATA);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_set_mode(ESP_MODEM_MODE_DATA) failed with %d", err);
        return;
    }
    /* Wait for IP address */
    ESP_LOGI(TAG, "Waiting for IP address");
    xEventGroupWaitBits(event_group, CONNECT_BIT | USB_DISCONNECTED_BIT, pdFALSE, pdFALSE, portMAX_DELAY);
    CHECK_USB_DISCONNECTION(event_group);

    esp_mqtt_client_config_t mqtt_config = {
        .broker.address.uri = BROKER_URL,
    };

    esp_mqtt_client_handle_t mqtt_client = esp_mqtt_client_init(&mqtt_config);
    esp_mqtt_client_register_event(mqtt_client, ESP_EVENT_ANY_ID, mqtt_event_handler, NULL);
    esp_mqtt_client_start(mqtt_client);
    ESP_LOGI(TAG, "Waiting for MQTT data");
    xEventGroupWaitBits(event_group, GOT_DATA_BIT | USB_DISCONNECTED_BIT, pdFALSE, pdFALSE, portMAX_DELAY);
    CHECK_USB_DISCONNECTION(event_group);

    esp_mqtt_client_destroy(mqtt_client);
    err = esp_modem_set_mode(dce, ESP_MODEM_MODE_COMMAND);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_set_mode(ESP_MODEM_MODE_COMMAND) failed with %d", err);
        return;
    }
    char imsi[32];
    err = esp_modem_get_imsi(dce, imsi);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_get_imsi failed with %d", err);
        return;
    }
    ESP_LOGI(TAG, "IMSI=%s", imsi);

    // USB example runs in a loop to demonstrate hot-plugging and sudden disconnection features.
    ESP_LOGI(TAG, "USB demo finished. Disconnect and connect the modem to run it again");
    xEventGroupWaitBits(event_group, USB_DISCONNECTED_BIT, pdFALSE, pdFALSE, portMAX_DELAY);
    CHECK_USB_DISCONNECTION(event_group); // dce will be destroyed here
} // while (1)
}
`

wishinlife, 2022-11-29:
@tore-espressif Following your prompt, I checked my code and did not execute usb_install, but did execute gpio_install_isr_service before executing esp_modem_new_dev_usb. After I commented gpio_install_isr_service, panic_abort is not displayed any more. Without comments, usbh_install returns 261 (ret=261).

Although not panic_abort after commented gpio_install_isr_service, but it's also not connected to the Internet.
I tested interface from 0 to 4, vid tested 0x6026 and 0x6000, log output end at "I (1141) PPP_CLIENT: Waiting for USB device connection...", Not after that. If I pull out the usb, it will output :
```
E (759991) HUB: Bad transfer status: CHECK_SHORT_CONFIG_DESC
E (759991) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC
```

tore-espressif, 2022-12-06:
@wishinlife I can see that you modified the example heavily. Is it possible to share your full application?

tore-espressif, 2022-12-12:
@wishinlife thank you the update.

> After I commented gpio_install_isr_service, panic_abort is not displayed any more. Without comments, usbh_install returns 261 (ret=261).
Why is that?

In the code you shared, there is no `gpio_install_isr_service`, so I can only guess, that there is a problem with interrupt allocation during `usbh_install`.

Now at least, the program doesn't crash, but it fails to connect to your EC200S modem.

I also checked [EC200S USB User Guide](https://forums.quectel.com/uploads/short-url/5dI1KXd7qZVkI7fqgZc6EoYktj6.pdf) where the VID/PID is defined as 0x2C7C:0x6002 (compared to your 0x2C7C:0x60**26**). Could you please share your full USB device descriptors so we can double-check? (`lsusb -v` on Linux or use USB Device Tree Viewer on Windows)

wishinlife, 2022-12-20:
@tore-espressif Sorry, the previously uploaded file lacks content, it has been added.
[4G-modem-test.zip](https://github.com/espressif/esp-protocols/files/10268275/4G-modem-test.zip)

I use VID/PID as 0x2C7C:0x6002 same issue.

wishinlife, 2022-12-20:
USB device descriptors:
`Bus 003 Device 003: ID 2c7c:6002 Quectel Wireless Solutions Co., Ltd. Android
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x2c7c Quectel Wireless Solutions Co., Ltd.
  idProduct          0x6002 
  bcdDevice            3.18
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00c4
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass        224 Wireless
      bFunctionSubClass       1 Radio Frequency
      bFunctionProtocol       3 RNDIS
      iFunction               5 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      3 RNDIS
      iInterface              5 
      ** UNRECOGNIZED:  05 24 00 10 01
      ** UNRECOGNIZED:  05 24 01 00 01
      ** UNRECOGNIZED:  04 24 02 00
      ** UNRECOGNIZED:  05 24 06 00 01
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              5 
[... 114 lines cut ...]
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              16
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
`

tore-espressif, 2023-04-02:
In case the problem is still the same, then you can check following comment about ESP interrupt allocation:
https://github.com/espressif/esp-idf/issues/10903#issuecomment-1456176716

Or you can read related section in our [IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/intr_alloc.html)