espressif/esp-protocols #181: Analyze this Trace

franz-ms-muc, 2022-11-26:
Analyze this Trace

franz-ms-muc, 2022-11-26:
```
D (3446) command_lib: generic_command command AT

D (3506) command_lib: Response: AT
OK

I (3506) cmux_example: sync Success after 1 try
V (3506) command_lib: set_flow_control
V (3506) command_lib: generic_command_common
V (3506) command_lib: generic_command
D (3516) command_lib: generic_command command AT+IFC=2, 2

D (3606) command_lib: Response: AT+IFC=2, 2
OK

I (3606) cmux_example: set_flow_control OK
V (3606) command_lib: sync
V (3606) command_lib: generic_command_common
V (3606) command_lib: generic_command
D (3616) command_lib: generic_command command AT

D (3706) command_lib: Response: AT
OK

I (3706) cmux_example: sync Success after 0 try
V (3706) command_lib: set_cmux
V (3706) command_lib: generic_command_common
V (3706) command_lib: generic_command
D (3716) command_lib: generic_command command AT+CMUX=0

D (3806) command_lib: Response: AT+CMUX=0
OK

V (3906) CMUX Received: 0x3fce596c   f9 03 73 01 d7 f9                                 |..s...|
D (3906) CMUX: Payload frame: dlci:00 type:73 payload:0 available:2
V (3916) CMUX Received: 0x3fce596c   f9 07 73 01 15 f9                                 |..s...|
D (3916) CMUX: Payload frame: dlci:01 type:73 payload:0 available:2
V (3926) CMUX Received: 0x3fce596c   f9 0b 73 01 92 f9                                 |..s...|
D (3936) CMUX: Payload frame: dlci:02 type:73 payload:0 available:2
Modem has correctly entered mode CMUX_MANUAL_MODE
V (3946) command_lib: set_echo
V (3946) command_lib: generic_command_common
V (3946) command_lib: generic_command
D (3956) command_lib: generic_command command ATE0

V (3956) Send: 0x3fcf50b0   f9 05 ef 0b                                       |....|
V (3966) Send: 0x3fcf5388   41 54 45 30 0d                                    |ATE0.|
V (3966) CMUX Received: 0x3fce596c   f9 05 ff 0b 41 54 45 30  0d ae f9 f9 05 ff 0d 0d  |....ATE0........|
V (3986) CMUX Received: 0x3fce597c   0a 4f 4b 0d 0a 4a f9                              |.OK..J.|
D (3996) CMUX: Payload frame: dlci:01 type:ff payload:5 available:19
D (3996) CMUX: Payload frame: dlci:01 type:ff payload:6 available:8
D (4006) command_lib: Response: 
OK

V (3986) Send: 0x3fcf50b4   bb f9                                             |..|
V (4016) command_lib: set_pdp_context
V (4016) command_lib: generic_command_common
V (4026) command_lib: generic_command
D (4026) command_lib: generic_command command AT+CGDCONT=1,"IP","nbiot.global-m2m.net"

V (4036) Send: 0x3fcf5030   f9 05 ef 53                                       |...S|
V (4046) Send: 0x3fce7490   41 54 2b 43 47 44 43 4f  4e 54 3d 31 2c 22 49 50  |AT+CGDCONT=1,"IP|
V (4046) CMUX Received: 0x3fce596c   f9 05 ff 0d 0d 0a 4f 4b  0d 0a 4a f9              |......OK..J.|
D (4056) CMUX: Payload frame: dlci:01 type:ff payload:6 available:8
V (4066) Send: 0x3fce74a0   22 2c 22 6e 62 69 6f 74  2e 67 6c 6f 62 61 6c 2d  |","nbiot.global-|
D (4076) command_lib: Response: 
OK

V (4076) Send: 0x3fce74b0   6d 32 6d 2e 6e 65 74 22  0d                       |m2m.net".|
V (4086) Send: 0x3fcf5034   d9 f9                                             |..|
[... 903 lines cut ...]
V (84906) Send: 0x3fce2734   ef f9                                             |..|
V (86646) CMUX Received: 0x3fce596c   f9 05 ff ed 7e 21 45 00  00 9d 96 d7 40 00 39 11  |....~!E.....@.9.|
V (86646) CMUX Received: 0x3fce597c   79 98 d9 0e a4 23 64 52  4f 5c 00 35 cd 7c 00 89  |y....#dRO\.5.|..|
V (86656) CMUX Received: 0x3fce598c   f3 59 fb ed 81 80 00 01  00 04 00 00 00 00 07 67  |.Y.............g|
V (86666) CMUX Received: 0x3fce599c   74 70 69 6f 74 31 09 69  7a 61 74 63 6c 6f 75 64  |tpiot1.izatcloud|
D (86676) CMUX: Payload frame: dlci:01 type:ff payload:118 available:60
V (86686) CMUX Received: 0x3fce59ac   03 6e 65 74 00 00 01 00  01 c0 0c 00 05 00 01 00  |.net............|
V (86696) CMUX Received: 0x3fce59bc   00 09 ad 00 14 07 67 74  70 69 6f 74 31 06 67 74  |......gtpiot1.gt|
V (86706) CMUX Received: 0x3fce59cc   70 67 65 6f 03 63 6f 6d  00 c0 33 00 05 00 01 00  |pgeo.com..3.....|
V (86716) CMUX Received: 0x3fce59dc   01 4d 1d 00 0e 07 67 74  70 69 e2 f9              |.M....gtpi..|
D (86716) CMUX: Payload frame: dlci:01 type:ff payload:58 available:60
V (86726) CMUX Received: 0x3fce596c   f9 05 ff 09 6f 74 31 03  4d f9 f9 05 ff 51 67 74  |....ot1.M....Qgt|
V (86736) CMUX Received: 0x3fce597c   70 c0 14 c0 53 00 01 00  01 00 00 00 32 00 04 36  |p...S.......2..6|
V (86746) CMUX Received: 0x3fce598c   47 b7 ab c0 53 00 01 00  01 00 00 00 32 00 04 23  |G...S.......2..#|
V (86756) CMUX Received: 0x3fce599c   a0 a9 7f ce 9b 7e 2f f9                           |.....~/.|
D (86766) CMUX: Payload frame: dlci:01 type:ff payload:4 available:52
D (86776) CMUX: Payload frame: dlci:01 type:ff payload:40 available:42
V (86776) Send: 0x3fce2730   f9 05 ef 7b                                       |...{|
V (86786) Send: 0x3fce87a4   7e 21 45 00 00 38 00 1d  00 00 ff 01 8a c7 64 52  |~!E..8........dR|
V (86796) Send: 0x3fce87b4   4f 5c d9 0e a4 23 03 03  3b 68 00 00 00 00 45 00  |O\...#..;h....E.|
V (86806) Send: 0x3fce87c4   00 9d 96 d7 40 00 39 11  79 98 d9 0e a4 23 64 52  |....@.9.y....#dR|
V (86816) Send: 0x3fce87d4   4f 5c 00 35 cd 7c 00 89  f3 59 a2 7b 7e           |O\.5.|...Y.{~|
V (86816) Send: 0x3fce2734   ef f9                                             |..|

```

franz-ms-muc, 2022-11-26:
code:

[https://gist.github.com/franz-ms-muc/febec6a7f8836b140d06758c74d25582](https://gist.github.com/franz-ms-muc/febec6a7f8836b140d06758c74d25582)

@david-cermak the Question is: is it constantly receiving Data ? the Code just enters into IP Network, and no Action then.

franz-ms-muc, 2022-11-26:
[http://izatcloud.net/](http://izatcloud.net/) seems to be the GPS Data Provider of Qualcom.

franz-ms-muc, 2022-11-26:
```
................
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2d (SPI_FAST_FLASH_BOOT)
Saved PC:0x4206dd76
0x4206dd76: esp_pm_impl_waiti at /home/franz/esp-idf-v4.4.2/components/esp_pm/pm_impl.c:839

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1654
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
entry 0x403c9954
I (24) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (25) boot: compile time 13:57:41
I (25) boot: chip revision: 0
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 factory          factory app      00 00 00010000 00177000
I (87) boot:  4 ota_0            OTA app          00 10 00190000 00177000
I (95) boot:  5 ota_1            OTA app          00 11 00310000 00177000
I (102) boot: End of partition table
I (106) boot: Defaulting to factory image
I (111) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=275a8h (161192) map
I (148) esp_image: segment 1: paddr=000375d0 vaddr=3fc92220 size=02928h ( 10536) load
I (151) esp_image: segment 2: paddr=00039f00 vaddr=40374000 size=06118h ( 24856) load
I (160) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=71f00h (466688) map
I (246) esp_image: segment 4: paddr=000b1f28 vaddr=4037a118 size=08100h ( 33024) load
I (254) esp_image: segment 5: paddr=000ba030 vaddr=50000000 size=00010h (    16) load
I (260) boot: Loaded app from partition at offset 0x10000
I (260) boot: Disabling RNG early entropy source...
I (275) cpu_start: Pro cpu up.
I (275) cpu_start: Starting app cpu, entry point is 0x4037535c
0x4037535c: call_start_cpu1 at /home/franz/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
D (284) clk: RTC_SLOW_CLK calibration value: 3848602
I (294) cpu_start: Pro cpu start user code
I (294) cpu_start: cpu freq: 160000000
I (294) cpu_start: Application information:
I (297) cpu_start: Project name:     simple_cmux_client
I (303) cpu_start: App version:      v0.6-78-gdf396ca-dirty
I (309) cpu_start: Compile time:     Nov 26 2022 13:57:50
I (315) cpu_start: ELF file SHA256:  33498ce217941ee5...
I (321) cpu_start: ESP-IDF:          v4.4.2-dirty
V (327) memory_layout: reserved range is 0x3c0a75a8 - 0x3c0a75c8
D (333) memory_layout: Checking 5 reserved memory ranges:
D (338) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc92218
D (345) memory_layout: Reserved memory range 0x3fc92220 - 0x3fc97b20
D (351) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
D (357) memory_layout: Reserved memory range 0x40374000 - 0x40382218
0x40374000: _WindowOverflow4 at /home/franz/esp-idf-v4.4.2/components/freertos/port/xtensa/xtensa_vectors.S:1736

D (364) memory_layout: Reserved memory range 0x600fe000 - 0x600fe000
D (370) memory_layout: Building list of available memory regions:
V (376) memory_layout: Examining memory region 0x40374000 - 0x40378000
0x40374000: _WindowOverflow4 at /home/franz/esp-idf-v4.4.2/components/freertos/port/xtensa/xtensa_vectors.S:1736

0x40378000: spi1_flash_os_yield at /home/franz/esp-idf-v4.4.2/components/spi_flash/spi_flash_os_func_app.c:135

V (383) memory_layout: Region 0x40374000 - 0x40378000 inside of reserved 0x40374000 - 0x40382218
0x40374000: _WindowOverflow4 at /home/franz/esp-idf-v4.4.2/components/freertos/port/xtensa/xtensa_vectors.S:1736
[... 466 lines cut ...]
V (10946) command_lib: generic_command
D (10956) command_lib: generic_command command AT

V (10956) Send: 0x3fcf5180   f9 09 ef 07                                       |....|
V (10966) Send: 0x3fcf5458   41 54 0d                                          |AT.|
V (10956) CMUX Received: 0x3fce596c   f9 09 ff 0d 0d 0a 4f 4b  0d 0a cd f9              |......OK....|
D (10986) CMUX: Payload frame: dlci:02 type:ff payload:6 available:8
D (10986) command_lib: Response: 
OK

V (10986) Send: 0x3fcf5184   35 f9                                             |5.|
V (41496) command_lib: sync
V (41496) command_lib: generic_command_common
V (41496) command_lib: generic_command
D (41496) command_lib: generic_command command AT

V (41496) Send: 0x3fcf5180   f9 09 ef 07                                       |....|
V (41506) Send: 0x3fcf5458   41 54 0d                                          |AT.|
V (41496) CMUX Received: 0x3fce596c   f9 09 ff 0d 0d 0a 4f 4b  0d 0a cd f9              |......OK....|
D (41516) CMUX: Payload frame: dlci:02 type:ff payload:6 available:8
V (41526) Send: 0x3fcf5184   35 f9                                             |5.|
D (41536) command_lib: Response: 
OK

```

david-cermak, 2022-12-01:
>  is it constantly receiving Data ? the Code just enters into IP Network, and no Action then.

I can only see that the PPP session is up and running. The actual network connection is up the device itself.

david-cermak, 2023-03-24:
@franz-ms-muc Could you please share an update if any? Could we close it?

diplfranzhoepfinger, 2023-03-31:
the Problems here come from the SIM7070 Module. 

the Connections to 

gtpiot1.izatcloud.net

gtpiot1.gtpgeo.com

come from the Modem Firmware and are not issued by Application nor by the esp-modem. 

nothing Espressif can do here. 
it is a SIMCOM/Qualcomm issue. 

(SIMCOM Solved this after a while but the Topic with SIM7070 is still the GPS and Data is not parallel but only one after the other)

david-cermak, 2023-11-28:
> nothing Espressif can do here.

okay, closing

closed 2023-11-28 by david-cermak