espressif/esp-protocols #183: esp_modem_at throwing timeout error

StarOfSlytherin, 2022-11-30:
`esp_modem_at()` is throwing `ESP_ERR_TIMEOUT` when used with CMUX.

Here's my code:
```c
/*
 * SPDX-FileCopyrightText: 2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */
/* PPPoS Client Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_netif.h"
#include "esp_netif_ppp.h"
#include "mqtt_client.h"
#include "esp_modem_api.h"
#include "esp_log.h"
#include "sdkconfig.h"
#include "driver/gpio.h"

#define BUF_SIZE 1024
char at_response[BUF_SIZE];

esp_modem_dce_t *dce;

static const char *TAG = "pppos_example";
static EventGroupHandle_t event_group = NULL;
static const int CONNECT_BIT = BIT0;
static const int GOT_DATA_BIT = BIT2;

esp_mqtt_client_handle_t client;

static void mqtt_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    ESP_LOGD(TAG, "Event dispatched from event loop base=%s, event_id=%d", base, event_id);
    esp_mqtt_event_handle_t event = event_data;
    client = event->client;
    int msg_id;
    switch ((esp_mqtt_event_id_t)event_id)
    {
    case MQTT_EVENT_CONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
        msg_id = esp_mqtt_client_subscribe(client, "/topic/esp-pppos/#", 0);
        ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);
        break;
    case MQTT_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
        esp_mqtt_client_reconnect(client);
        break;
    case MQTT_EVENT_SUBSCRIBED:
        ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
        msg_id = esp_mqtt_client_publish(client, "/topic/esp-pppos", "{\"sentFrom\":\"ESP32 via PPPoS\"}", 0, 0, 0);
        ESP_LOGI(TAG, "sent publish successful, msg_id=%d", msg_id);
        break;
    case MQTT_EVENT_UNSUBSCRIBED:
        ESP_LOGI(TAG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_PUBLISHED:
        ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_DATA:
[... 181 lines cut ...]
    }
}
```
Output:
```
I (00:01:27.471) pppos_example: MQTT_EVENT_DATA
TOPIC=/topic/esp-pppos/cmux
DATA={"sentFrom":"ESP32 via PPPoS CMUX"}
E (00:01:32.282) pppos_example: Failed to run esp_modem_at(dce, AT+CPSI?) with error: ESP_ERR_TIMEOUT
I (00:01:32.591) pppos_example: MQTT_EVENT_DATA
TOPIC=/topic/esp-pppos/cmux
DATA={"sentFrom":"ESP32 via PPPoS CMUX"}
E (00:01:37.382) pppos_example: Failed to run esp_modem_at(dce, AT+CPSI?) with error: ESP_ERR_TIMEOUT
I (00:01:37.631) pppos_example: MQTT_EVENT_DATA
TOPIC=/topic/esp-pppos/cmux
DATA={"sentFrom":"ESP32 via PPPoS CMUX"}
```
When I check the SIM7600's Tx pin I can see the response to the AT Command:
```
~!Ed{@6m[y]^d]=[zr9iP0:/topic/esp-pppos/cmux{"sentFrom":"ESP32 via PPPoS CMUX"}]?~	
+CPSI: LTE,Online,xxx-xx,0x2xxx,xxxxxxxx,xx,EUTRAN-BAND3,xxxx,5,5,-100,-907,-608,16
	

OK
```

david-cermak, 2022-12-01:
@StarOfSlytherin Have you tried to extend the timeout a bit? the device might reply in 100ms, but the real terminal is shared with network virtual terminal, so might be locked for receving/sending network packets. I think a little longer response in CMUX compared to the plain DATA mode is somewhat expected.

StarOfSlytherin, 2022-12-02:
I've tried it with 9000ms too as per the datasheet (Max response time) and it has not worked. It's a bit weird as functions like `esp_modem_get_signal_quality` or `esp_modem_at(dce, "AT+CPMUTEMP", at_response, 100);` work fine. `AT+CPSI?` does not work most of the time. By any chance is it because of the response length?

closed 2023-01-09 by StarOfSlytherin