espressif/esp-protocols #184: How to increase size of AT commands response?

MRCBoss, 2022-12-01:
Hello. I'm using ESP32 + BG96, 1 serial channel, CMUX, esp-modem version 0.1.24. Data mode sends sensor data beyond MQTT (PPP) and I need command mode to get GNSS data. The AT+QGPSLOC command response is received truncated, because of ESP_MODEM_C_API_STR_MAX limit of 64. How could I change this limit without modifying the ESP_MODEM_C_API_STR_MAX definition inside esp_modem_c_api.cpp code?

david-cermak, 2022-12-08:
Hi @MRCBoss 

You can define it directly in the project makefile by adding these two lines after project definitions
```cmake
idf_component_get_property(esp_modem esp_modem COMPONENT_LIB)
target_compile_definitions(${esp_modem} PRIVATE  "ESP_MODEM_C_API_STR_MAX=128")
```

(e.g. for the modem-console example, place the above after this line:

https://github.com/espressif/esp-protocols/blob/457f8335bb876b8af80a65b096e5381f49f29878/components/esp_modem/examples/modem_console/CMakeLists.txt#L6

)

david-cermak, 2022-12-09:
Have you placed into the project makefile (not the `CMakeLists.txt` in the main folder) and just **after** the `project()`directive? This should work for correctly, at least on IDF v4.4 and v5.+.

Which IDF version are you using?

MRCBoss, 2022-12-09:
Yes, I've placed into the project CMakeLists.txt, but I had put it **before** the project() directive. Now I've put it **after** that directive and the same error ocurred. I'm using IDF 4.4.2.
The esp-modem managed component is used inside a custom component (not main), could this be related?.

david-cermak, 2022-12-09:
Oh, I see! In that case, we have to specify the **full name** of the component (including namespace *"espressif"*), which is `espressif__esp_modem`

The cmake commands should then look like:
```cmake
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(your-project)

idf_component_get_property(esp_modem espressif__esp_modem COMPONENT_LIB)
target_compile_definitions(${esp_modem} PRIVATE  "ESP_MODEM_C_API_STR_MAX=128")
```

MRCBoss, 2022-12-10:
Now it works fine. Thank you @david-cermak !

closed 2022-12-10 by MRCBoss