espressif/esp-protocols #185: Wrong MDNS name mangled

tannewt, 2022-12-01:
In CircuitPython we have two mdns hostnames. One is unique (cpy-MAC.local) and one that is shared (circuitpython.local). The unique name is the internal hostname and the shared one is a delegate. IDF version 4.4 of the mdns code will detect a collision on circuitpython.local and then mangle cpy-MAC. I've got code that limits the hostname mangling but wonder what the right thing to do is. Its weird that the internal mdns code can mangle the host and instance names but there isn't an api to read them back.

Should the delegate names be mangled too? Should any be mangled at all?

david-cermak, 2023-02-16:
Hi @tannewt 

Could you please share a simple example to reproduce this problem?
As mentioned above, there's a bug in conflict resolution, or better said an incomplete implementation of [`Simultaneous Probe Tiebreaking for Multiple Records`](https://www.rfc-editor.org/rfc/rfc6762.html#section-8.2.1), which may result in devices competing for a unique name, but only self-hostnames or services (for example if two devices probe for one name with two records:  `A` and `AAAA`, then it might be possible that one device *wins* for `A`-record and loses for `AAAA`, while the other one *loses* and *wins* respectively, so both devices would mangle their names to continue with tiebreaking)

After thoroughly testing these races, I wasn't able to find any relation to the delegated names and services, so the issue you're reporting might not be related. Could you please share more details about your setup? Are you also advertizing services for the delegated names (or for the self-hostname?), are you using multiple instances, setting the instance name? Any info would be helpful, as I wasn't able to reproduce any problem when setting **unique self-names**.

tannewt, 2023-02-16:
> Hi @tannewt
> 
> Could you please share a simple example to reproduce this problem?

Unfortunately, I don't have a simple example already. I saw this within CircuitPython's use (workflow is true):

Starting it up and adding the delegate name:

https://github.com/adafruit/circuitpython/blob/bbadc0059930519badba98cc340531cb3efb28a8/ports/espressif/common-hal/mdns/Server.c#L50-L70

Adding the service:

https://github.com/adafruit/circuitpython/blob/bbadc0059930519badba98cc340531cb3efb28a8/supervisor/shared/web_workflow/web_workflow.c#L321-L340

> As mentioned above, there's a bug in conflict resolution, or better said an incomplete implementation of [`Simultaneous Probe Tiebreaking for Multiple [...]

I don't think this will hit CircuitPython because the IDF may not consider it a self-hostname.

> After thoroughly testing these races, I wasn't able to find any relation to the delegated names and services, so the issue you're reporting might not be related. Could you please share more details [...]

I believe we do advertise a service but it should be under the unique name. We do also set the instance name.

> Any info would be helpful, as I wasn't able to reproduce any problem when setting **unique self-names**.

Here are the changes I made so that it'd work like I expected: https://github.com/adafruit/esp-idf/commit/9eb4f5855dfa222e2ebfad42b074f49fe340a361

david-cermak, 2023-02-17:
Thanks for the pointers! I was able to reproduce the issue, but only if one of the devices uses the delegated name as its own self-hostname.

Thank you also for sharing your fix, I think it could be used if we move it just one condition above (i.e. skip the transition to probing state). I'll try to address the problem more generally, also for services and TXT's.

tannewt, 2023-02-17:
Feel free to use it! Thanks for looking into this.

closed 2023-03-20 by david-cermak