espressif/esp-protocols #186: ap_to_pppos example is crashing

tohyf, 2022-12-02:
I am trying the ap_to_pppos example as-is with only modification of the tx/rx pin for the gsm module and it is crashing with the following:
 <0x1b>[0;32mI (31731) ap_to_pppos: wifi_init_softap finished. SSID:myssid password:mypassword channel:1<0x1b>[0m␍␊
[17:25:16:101] ␍␊
[17:25:16:102] assert failed: vTaskGenericNotifyGiveFromISR tasks.c:5831 (( ( &( pxTCB->xEventListItem ) )->pvContainer ) == ((void *)0))␍␊

esp IDF versions that i am using:
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.23.1 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20220411 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)
  -

Operating System: linux
Java Runtime Version: 11.0.17+8-post-Ubuntu-1ubuntu220.04
Eclipse Version: 4.20.0.v20210611-1600
Eclipse CDT Version: 10.3.0.202104042017
IDF Eclipse Plugin Version: 2.1.2.202106231127
ESP-IDF v4.4.2
Python set for IDF_PYTHON_ENV: Python 3.8.10
Python declared in PATH variable: Python 2.7.18

tohyf, 2022-12-02:
raising main task stack size to 8192 solves the issue

david-cermak, 2022-12-08:
This is weird, so it didn't report stack overflow of the main task? just the assertion you mentioned above?

Will update stack sizes for all examples, soon. Already running target tests in the CI so we should be able to detect these issues quickly in future.

Alpha-zh, 2023-01-09:
I got the same problem.  use ap_to_ppp example

1、自定义了modem硬件，用的是ML302芯片，拨号成功，日志如下：
D (30800) command_lib: ML302 Sync Ok,retry(0)
D (30920) command_lib: generic_command command ATE0

D (31040) command_lib: Response:
OK

D (31040) command_lib: generic_command command AT+CGDCONT=1,"IP","cmnet"

D (31160) command_lib: Response:
OK

D (31160) command_lib: generic_command command ATD*99***1#

D (31600) command_lib: Response:
CONNECT

I (37830) esp-netif_lwip-ppp: Connected
I (37830) esp-netif_lwip-ppp: Name Server1: 120.196.165.7
I (37830) esp-netif_lwip-ppp: Name Server2: 0.0.0.0
D (37840) ap_to_pppos: IP event! 6
I (37840) ap_to_pppos: Modem Connect to PPP Server
I (37850) ap_to_pppos: ~~~~~~~~~~~~~~
I (37850) ap_to_pppos: IP          : 10.130.216.177
I (37860) ap_to_pppos: Netmask     : 255.255.255.255
I (37860) ap_to_pppos: Gateway     : 192.168.0.1
I (37870) ap_to_pppos: Name Server1: 120.196.165.7
I (37870) ap_to_pppos: Name Server2: 0.0.0.0
I (37880) ap_to_pppos: ~~~~~~~~~~~~~~
I (37880) ap_to_pppos: GOT ip event!!!

2、SOFT AP启动也成功
I (38930) wifi:mode : softAP (34:ab:95:9a:57:05)
I (38930) wifi:Total power save buffer number: 8
I (38930) wifi:Init max length of beacon: 752/752
I (38930) wifi:Init max length of beacon: 752/752
D (38940) ap_to_pppos: UnCare wifi event id:12
I (38940) ap_to_pppos: wifi_init_softap finished. SSID:WIFI-ML302 password:1231231230 channel:7
I (38950) ap_to_pppos: wifi init soft ap success...
I (38950) ap_to_pppos: ip napt enable ok

3、好像是在这个位置报的错误，也可能是其它任务的报错：
代码：ap_to_pppos.c的最后部分，在app_main（）

void app_main(void)
{
……

    // Provide a recovery if disconnection of some kind registered
    while (true) {
        ESP_LOGD(TAG,"Waiting for disconnect event,if disconneted will restart network.");
        EventBits_t bits = xEventGroupWaitBits(event_group, DISCONNECT_BIT, pdTRUE, pdFALSE, portMAX_DELAY);
        if (bits & DISCONNECT_BIT) {
            ESP_LOGD(TAG,"Modem disconnected,will restart network...");
            modem_stop_network();
            start_network();
        }
    }

log：
D (38960) ap_to_pppos: Waiting for disconnect event,if disconneted will restart network.

**assert failed: vTaskGenericNotifyGiveFromISR tasks.c:5845 (( ( &( pxTCB->xEventListItem ) )->pvContainer ) == ((void *)0))**

Backtrace: 0x40082241:0x3ffb0a40 0x4008e8e9:0x3ffb0a60 0x40096595:0x3ffb0a80 0x40091edd:0x3ffb0ba0 0x40084fa8:0x3ffb0bc0 0x40085127:0x3ffb0bf0 0x40085771:0x3ffb0c10 0x4015ca43:0x3ffbbd40 0x400d2e7a:0x3ffbbd60 0x400907ed:0x3ffbbd80 0x400926e9:0x3ffbbda0
0x40082241: panic_abort at C:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_system/panic.c:402

0x4008e8e9: esp_system_abort at C:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_system/esp_system.c:128

0x40096595: __assert_func at C:/Espressif/frameworks/esp-idf-v4.4.3/components/newlib/assert.c:85

0x40091edd: vTaskGenericNotifyGiveFromISR at C:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/tasks.c:5845 (discriminator 1)

0x40084fa8: timer_alarm_handler at C:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_timer/src/esp_timer.c:411

0x40085127: timer_alarm_isr at C:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_timer/src/esp_timer_impl_lac.c:186

0x40085771: _xt_medint2 at C:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/port/xtensa/xtensa_vectors.S:1200

0x4015ca43: cpu_ll_waiti at C:/Espressif/frameworks/esp-idf-v4.4.3/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_pm/pm_impl.c:837

0x400d2e7a: esp_vApplicationIdleHook at C:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_system/freertos_hooks.c:63

0x400907ed: prvIdleTask at C:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/tasks.c:3987 (discriminator 1)

0x400926e9: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/port/xtensa/port.c:131

}

idf 版本4.4.2 和4.4.3都试过了，报的同样错误。

请指导一下，感谢。

Alpha-zh, 2023-01-10:
yes,increase the main task's stack size help,thank you!!

event loop task stack size :2304 -> 10240
main task stack size :3584  -> 8000

通过调整这两个参数应用不会崩溃了，手机可以通过wifi连接设备热点，并访问Internet。

在连接过程一直报这个错误，但好像并不影响使用：
E (211218) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211218) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211228) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211228) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211238) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211238) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211248) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211248) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211258) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (211268) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1

closed 2023-03-21 by david-cermak