espressif/esp-protocols #187: cmux_example: Failed to configure multiplexed command mode... exiting

jgus, 2022-12-04:
I'm unable to get the `simple_cmux_client` example (or my own code) to working in CMUX mode.

Command and data modes work fine (both examples and in my own code) but CMUX mode refuses to work.

My issue looks similar to #140, but where that appears to be resolved, I'm seeing this issue in the latest master branch.

I'm using ESP-IDF v5.0, and the latest master branch (457f8335bb876b8af80a65b096e5381f49f29878).

My target device is a LilyGO T-SIM7000G 20200415 (https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/Historical/SIM7000G_20200415/README.MD) with a SIM7000G modem and an ESP-WROVER-E processor. I've set pins appropriately for my device (TX 27, RX 26, RTS & CTS -1, no flow control) and I've verified I can communicate with the modem. Command and data modes work fine and I can successfully communicate over the internet in data mode.

When running the CMUX sample (or trying myself in my own code) the call to `dce->set_mode(esp_modem::modem_mode::CMUX_MODE)` fails.

Anything I can add to help debug this issue, please let me know.

Verbose log follows:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6992
load:0x40078000,len:15456
load:0x40080400,len:3840
0x40080400: _init at ??:?

entry 0x4008064c
I (27) boot: ESP-IDF v5.0 2nd stage bootloader
I (27) boot: compile time 15:51:28
I (27) boot: chip revision: v3.0
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 80MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 16MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00177000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3fabch (260796) map
I (148) esp_image: segment 1: paddr=0004fae4 vaddr=3ffb0000 size=00534h (  1332) load
I (148) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=a8c28h (691240) map
I (254) esp_image: segment 3: paddr=000f8c50 vaddr=3ffb0534 size=02cd8h ( 11480) load
I (257) esp_image: segment 4: paddr=000fb930 vaddr=40080000 size=13ba4h ( 80804) load
I (276) esp_image: segment 5: paddr=0010f4dc vaddr=50000000 size=00010h (    16) load
I (286) boot: Loaded app from partition at offset 0x10000
I (286) boot: Disabling RNG early entropy source...
D (298) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (299) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (303) quad_psram: This chip is ESP32-D0WD
I (309) esp_psram: Found 8MB PSRAM device
I (313) esp_psram: Speed: 80MHz
I (317) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
V (324) mmu: after coalescing, 3 regions are left
V (329) mmu: found laddr is 0x1800000
V (333) esp_psram: 8bit-aligned-range: 0x400000 B, starting from: 0x3f800000
W (340) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (349) cpu_start: Pro cpu up.
I (353) cpu_start: Starting app cpu, entry point is 0x4008174c
0x4008174c: call_start_cpu1 at C:/Users/Josh/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (874) esp_psram: SPI SRAM memory test OK
D (881) clk: RTC_SLOW_CLK calibration value: 3357978
I (886) cpu_start: Pro cpu start user code
I (887) cpu_start: cpu freq: 160000000 Hz
I (887) cpu_start: Application information:
I (891) cpu_start: Project name:     simple_cmux_client
[... 326 lines cut ...]
D (2019) event: created task for loop 0x3ffaffa8
D (2019) event: created event loop 0x3ffaffa8
D (2029) esp_netif_lwip: LwIP stack has been initialized
D (2029) esp_netif_lwip: esp-netif has been successfully initialized
I (2039) uart: queue free spaces: 30
V (2039) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (2049) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (2059) intr_alloc: Connected src 35 to int 13 (cpu 0)
D (2059) esp_netif_objects: esp_netif_add_to_list 0x3ffc8608
D (2069) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (2079) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffc8914
D (2079) esp-netif_lwip-ppp: Phase Dead
I (2089) cmux_example: not set_flow_control, because 2-wire mode active.
V (2099) command_lib: set_cmux
V (2099) command_lib: generic_command_common
V (2099) command_lib: generic_command
D (2109) command_lib: generic_command command AT+CMUX=0

D (2119) command_lib: Response: AT+CMUX=0
OK

E (3239) cmux_example: Failed to configure multiplexed command mode... exiting
V (3239) intr_alloc: esp_intr_free: Disabling int, killing handler
D (3239) event: deleted loop 0x3ffaffa8
```

david-cermak, 2022-12-08:
@jgus The issue you referenced was fixed for SIM800 device which apparently send additional messages, that we misinterpreted as CMUX close.
It's probably another failure you're seeing, maybe related to your device. I think I can get one SIM7000 and test the CMUX mode.

> Anything I can add to help debug this issue, please let me know.

I can only suggest enabling verbose logging and sharing results, but see that the `V` level is included above, but don't see much. So unless you set the log level specifically via runtime API, we can assume that the transition fails early, when we're trying to setup virtual control terminal. In that case I would try to increase a delay between setting the CMUX mode by AT command and the actual CMUX protocol communication:

https://github.com/espressif/esp-protocols/blob/457f8335bb876b8af80a65b096e5381f49f29878/components/esp_modem/src/esp_modem_dce.cpp#L124-L125

Could you please increase the delay to say 500ms?

Another way to try to debug it is to run the console example and use the manual CMUX commands. We've recently introduced commands like `CMUX1`, `CMUX2`, `CMUX3`... `CMUX5`. So you can try to configure CMUX mode via AT command:
```
cmd AT+CMUX=0
```
and then set the manual CMUX transition
```
setmode CMUX1
```

david-cermak, 2023-03-24:
I've tried to test CMUX mode with SIM7000 and couldn't get it to work, it replies with `OK` on `AT+CMUX`. but doesn't seem to switch to the multiplexed mode and doesn't respond to CMUX command any more.

I'll add a brief documentation about the known issues of certain devices, as we've already experienced some problems and limitations with some modems.

closed 2023-11-28 by david-cermak