espressif/esp-protocols #190: Misconfigured mDNS causes ESP32 hang with LoadProhibited error

wuyuanyi135, 2022-12-11:
Hi, I was using the code from the example but accidentally commented out the `mdns_hostname_set` line. 

The code run without error (I `ESP_ERROR_CHECK`ed everything). But once the mDNS request was initiated, the ESP responded once (saw from Wireshark PCAP), then stop responding. And `idf.py monitor` (esp32s3 built-in JTAG) hanged without printing panic messages. Also, the `monitor` is not able to exit by `CTRL+]` until I unplug the USB and then it sometimes printed an incomplete message with `LoadProhibited error`.

This is much like a stackoverflow behavior I have seen before.  I know this is error is due to my misconfiguration, but I think there should be some checking to prevent such a debug puzzle?

```
void start_mdns_service()
{
    //initialize mDNS service
    esp_err_t err = mdns_init();
    if (err) {
        printf("MDNS Init failed: %d\n", err);
        return;
    }

    //set hostname
    // mdns_hostname_set("my-esp32");   // <-------- Accidentally commented out
    //set default instance
    mdns_instance_name_set("Jhon's ESP32 Thing");
}
```

david-cermak, 2022-12-19:
@wuyuanyi135 Thanks for the report. We're adding error codes when initializing mDNS in wrong order, but it was probably missed in `mdns_instance_name_set()` API. will fix.

wuyuanyi135, 2022-12-19:
Thanks for your reply!

closed 2023-02-01 by david-cermak