espressif/esp-protocols #195: How to use Custom Commands ?

franz-ms-muc, 2022-12-13:
Hi, 

is this the right way ? 
it get not issued to the Modem at my side: 

```
```

david-cermak, 2022-12-19:
Yes, this is correct. `command()` is a public method of DCE class and can be used freely.

diplfranzhoepfinger, 2023-01-12:
ok, tested again, they are somtimes sent out, sometimes not ... 
seems this is somehow still buggy ...

david-cermak, 2023-01-13:
This is the same API that's used by all AT commands, so they must work

> they are somtimes sent out, sometimes not ...

Are you sending any other AT commands from some other threads? This uses the same `DTE` and shares the same lock, so might be possibly locked while trying to send commands simultaneously?

diplfranzhoepfinger, 2023-01-13:
Code:
https://gist.github.com/diplfranzhoepfinger/37fba94d67a655ae1b2a5953ca9ea09d
Log:
https://gist.github.com/diplfranzhoepfinger/e40db2debd7ea5443e0a6a0def04891a

while the AT+CGNSSINFO is sent, 
the AT+CGNSSPWR=1 is not, 

also having it 4 times there does not improve it, 
have it 1 time there also not.

david-cermak, 2023-03-24:
This is a low level API, which knows nothing about AT commands. It's just sends the data (`char*`) that you give it.
You have to add a CR and/or a newline characacter:

https://gist.github.com/diplfranzhoepfinger/37fba94d67a655ae1b2a5953ca9ea09d#file-simple_cmux_client_main-cpp-L210

Needs to be:
```
    dce->command("AT+CGNSSPWR=1\r", got_line_fn, 1500);
```

As with your current code snippet, the device's AT parser actually sees a command:
```
AT+CGNSSPWR=1AT+CGNSSPWR=1AT+CGNSSPWR=1AT+CGNSSPWR=1AT\r
```
which is not a valid command, so it simply says:
```
D (22496) command_lib: Response: AT
```
[ERROR](https://gist.github.com/diplfranzhoepfinger/e40db2debd7ea5443e0a6a0def04891a#file-log-L236-L237)

diplfranzhoepfinger, 2023-03-31:
clear thanks. 
we can close this Issue, as we have integrated the AT+CGNSSPWR=1 already, a
and a PR is in place.

closed 2023-11-28 by david-cermak