espressif/esp-protocols #197: using SIM7600. I got IP but can't connect to SNTP

AAl-Hadi, 2022-12-18:
Hello,
I integrated the code with my project that was working on Ethernet
I'm using ESP-IDF v4.4
Here's my modem function:
`
  /* Init and register system/core components */
  ESP_ERROR_CHECK(esp_netif_init());
  // ESP_ERROR_CHECK(esp_event_loop_create_default());
  ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event, NULL));
  ESP_ERROR_CHECK(esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed, NULL));

  event_group = xEventGroupCreate();

  /* Configure the DTE */
  esp_modem_dte_config_t dte_config = ESP_MODEM_DTE_DEFAULT_CONFIG();

  /* setup UART specific configuration based on kconfig options */
  dte_config.uart_config.tx_io_num = CONFIG_EXAMPLE_MODEM_UART_TX_PIN;
  dte_config.uart_config.rx_io_num = CONFIG_EXAMPLE_MODEM_UART_RX_PIN;
  dte_config.uart_config.rts_io_num = CONFIG_EXAMPLE_MODEM_UART_RTS_PIN;
  dte_config.uart_config.cts_io_num = CONFIG_EXAMPLE_MODEM_UART_CTS_PIN;
  dte_config.uart_config.flow_control = ESP_MODEM_FLOW_CONTROL_SW;
  dte_config.uart_config.baud_rate = CONFIG_EXAMPLE_MODEM_UART_BAUDRATE;

  dte_config.uart_config.rx_buffer_size = CONFIG_EXAMPLE_MODEM_UART_RX_BUFFER_SIZE;
  dte_config.uart_config.tx_buffer_size = CONFIG_EXAMPLE_MODEM_UART_TX_BUFFER_SIZE;
  dte_config.uart_config.event_queue_size = CONFIG_EXAMPLE_MODEM_UART_EVENT_QUEUE_SIZE;
  dte_config.task_stack_size = CONFIG_EXAMPLE_MODEM_UART_EVENT_TASK_STACK_SIZE;
  dte_config.task_priority = CONFIG_EXAMPLE_MODEM_UART_EVENT_TASK_PRIORITY;
  dte_config.dte_buffer_size = CONFIG_EXAMPLE_MODEM_UART_RX_BUFFER_SIZE / 2;
  /* Configure the DCE */
  // esp_modem_dce_config_t dce_config = ESP_MODEM_DCE_DEFAULT_CONFIG(CONFIG_EXAMPLE_MODEM_PPP_APN);
  esp_modem_dce_config_t dce_config = ESP_MODEM_DCE_DEFAULT_CONFIG(CONFIG_EXAMPLE_MODEM_PPP_APN);

  /* Configure the PPP netif */
  esp_netif_config_t netif_ppp_config = ESP_NETIF_DEFAULT_PPP();

  /* Run the modem demo app */
  // Init netif object
  esp_netif_t *esp_netif = esp_netif_new(&netif_ppp_config);
  assert(esp_netif);

  ESP_LOGI(TAG, "Initializing esp_modem for the SIM7600 module...");
  esp_modem_dce_t *dce = esp_modem_new_dev(ESP_MODEM_DCE_SIM7600, &dte_config, &dce_config, esp_netif);

  assert(dce);

  ESP_LOGI(TAG, "Waiting...");
  vTaskDelay(pdMS_TO_TICKS(5000));
  ESP_LOGI(TAG, "Continue");

  int rssi, ber;
  esp_err_t err = esp_modem_get_signal_quality(dce, &rssi, &ber);
  while(err != ESP_OK){
    err = esp_modem_get_signal_quality(dce, &rssi, &ber);
    vTaskDelay(pdMS_TO_TICKS(1000));
  }
  if (err != ESP_OK)
  {
    ESP_LOGE(TAG, "esp_modem_get_signal_quality failed with %d", err);
    return;
  }
  ESP_LOGI(TAG, "Signal quality: rssi=%d, ber=%d", rssi, ber);

  ESP_LOGI(TAG, "Setting modem mode");
  err = esp_modem_set_mode(dce, ESP_MODEM_MODE_DATA);
  if (err != ESP_OK)
  {
    ESP_LOGE(TAG, "esp_modem_set_mode(ESP_MODEM_MODE_DATA) failed with %d", err);
    return;
  }
  /* Wait for IP address */
  ESP_LOGI(TAG, "Waiting to IP...");
`
This function works and I do get an IP. But I get no response from SNTP (that was working before with Ethernet)
I tried to change the DNS but that's not supported in this mode
What could be the problem?
Here's my output:
`14:46:04.908 -> [0;32mI (13120) modem_init: Waiting...[0m
14:46:09.911 -> [0;32mI (18120) modem_init: Continue[0m
14:46:30.908 -> [0;32mI (39130) modem_init: Signal quality: rssi=30, ber=99[0m
14:46:30.908 -> [0;32mI (39130) modem_init: Setting modem mode[0m
14:46:30.954 -> [0;32mI (39190) modem_init: Waiting to IP...[0m
14:46:38.075 -> [0;32mI (46310) esp-netif_lwip-ppp: Connected[0m
14:46:38.075 -> [0;32mI (46310) esp-netif_lwip-ppp: Name Server1: 84.235.57.230[0m
14:46:38.075 -> [0;32mI (46310) esp-netif_lwip-ppp: Name Server2: 84.235.6.55[0m
14:46:38.129 -> [0;32mI (46320) modem_init: Modem Connect to PPP Server[0m
14:46:38.129 -> [0;32mI (46320) modem_init: ~~~~~~~~~~~~~~[0m
14:46:38.129 -> [0;32mI (46330) modem_init: IP          : 10.159.39.168[0m
14:46:38.129 -> [0;32mI (46330) modem_init: Netmask     : 255.255.255.255[0m
14:46:38.129 -> [0;32mI (46340) modem_init: Gateway     : 10.64.64.64[0m
14:46:38.129 -> [0;32mI (46340) modem_init: Name Server1: 84.235.57.230[0m
14:46:38.129 -> [0;32mI (46350) modem_init: Name Server2: 84.235.6.55[0m
14:46:38.129 -> [0;32mI (46350) modem_init: ~~~~~~~~~~~~~~[0m
14:46:38.129 -> [0;32mI (46360) modem_init: GOT ip event!!![0m
14:46:38.165 -> [0;32mI (46360) sntp_sync: Waiting for system time to be set (1/15).[0m
14:46:39.132 -> [0;32mI (47370) sntp_sync: Waiting for system time to be set (2/15).[0m
14:46:40.134 -> [0;32mI (48370) sntp_sync: Waiting for system time to be set (3/15).[0m
14:46:41.138 -> [0;32mI (49370) sntp_sync: Waiting for system time to be set (4/15).[0m
14:46:42.142 -> [0;32mI (50370) sntp_sync: Waiting for system time to be set (5/15).[0m
14:46:43.144 -> [0;32mI (51370) sntp_sync: Waiting for system time to be set (6/15).[0m
14:46:44.150 -> [0;32mI (52370) sntp_sync: Waiting for system time to be set (7/15).[0m
14:46:45.149 -> [0;32mI (53370) sntp_sync: Waiting for system time to be set (8/15).[0m
14:46:46.137 -> [0;32mI (54370) sntp_sync: Waiting for system time to be set (9/15).[0m
14:46:47.155 -> [0;32mI (55370) sntp_sync: Waiting for system time to be set (10/15).[0m
14:46:48.140 -> [0;32mI (56370) sntp_sync: Waiting for system time to be set (11/15).[0m
14:46:49.161 -> [0;32mI (57370) sntp_sync: Waiting for system time to be set (12/15).[0m
14:46:50.163 -> [0;32mI (58370) sntp_sync: Waiting for system time to be set (13/15).[0m
14:46:51.166 -> [0;32mI (59370) sntp_sync: Waiting for system time to be set (14/15).[0m
`
I also tried connecting to an AWS http page but got this error:
`14:46:51.868 -> [0;31mE (60110) esp-tls: couldn't get hostname for :greengrass-ats.iot.eu-west-2.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0[0m
14:46:51.915 -> [0;31mE (60110) esp-tls: Failed to open new connection[0m
`

david-cermak, 2022-12-19:
Hi @AAl-Hadi 

Have you switched from Ethernet to PPP or added another PPP interface to the existing application with Ethernet interface?

If the latter, than it's a known issue, since:
* lwip uses one global array of DNS servers
* When Ethernet interface connects (using DHCP) it overrides these global DNS servers

As you found out, we currently don't support setting DNS servers for PPP netifs, but you can
* update DNS info for the Ethernet interface (this feels wrong, but DNS info is a global info, so should work)
* use lwIP API for updating DNS.

Sorry for the inconvenience, we'll have to fix this.

AAl-Hadi, 2022-12-19:
> 

Thank you for your reply. I replaced it. I am using a new HW that doesn't have Ethernet so I don't need it. Maybe there's something wrong with my integration? So, this is what I've done in my project and tell me if I missed something:
1. I copied esp-protocols/components/esp_modem/ from the repo to my project component.
2. I replaced my 'ethernet_init()' function to 'modem_init()' function. The function is posted up. 
I get an IP with this, but I can't connect to any website. Don't know why 
I've added the 'http_request' example from IDF-example as an easy project to see if I find out what's wrong. I got this error (the example project works fine with Ethernet):
```
12:03:44.987 -> [0;32mI (3111) modem_init: Modem powered off[0m
12:03:50.013 -> [0;32mI (8111) modem_init: Modem powered on[0m
12:03:54.996 -> [0;32mI (13111) modem_init: Beginning app_main[0m
12:03:54.996 -> [0;32mI (13111) modem_init: Initializing esp_modem for the SIM7600 module...[0m
12:03:54.996 -> [0;32mI (13111) yo: hello there[0m
12:03:54.996 -> [0;32mI (13121) uart: queue free spaces: 30[0m
12:03:54.996 -> [0;32mI (13121) modem_init: Waiting...[0m
12:03:59.985 -> [0;32mI (18121) modem_init: Continue[0m
12:04:20.994 -> [0;32mI (39131) modem_init: Signal quality: rssi=28, ber=99[0m
12:04:21.041 -> [0;32mI (39131) modem_init: Setting modem mode[0m
12:04:21.087 -> [0;32mI (39191) modem_init: Waiting to IP...[0m
12:04:27.159 -> [0;32mI (45291) esp-netif_lwip-ppp: Connected[0m
12:04:27.159 -> [0;32mI (45291) esp-netif_lwip-ppp: Name Server1: 84.235.57.230[0m
12:04:27.206 -> [0;32mI (45291) esp-netif_lwip-ppp: Name Server2: 84.235.6.55[0m
12:04:27.206 -> [0;32mI (45301) modem_init: Modem Connect to PPP Server[0m
12:04:27.206 -> [0;32mI (45301) modem_init: ~~~~~~~~~~~~~~[0m
12:04:27.206 -> [0;32mI (45311) modem_init: IP          : 10.248.208.75[0m
12:04:27.206 -> [0;32mI (45311) modem_init: Netmask     : 255.255.255.255[0m
12:04:27.206 -> [0;32mI (45321) modem_init: Gateway     : 10.64.64.64[0m
12:04:27.206 -> [0;32mI (45321) modem_init: Name Server1: 84.235.57.230[0m
12:04:27.206 -> [0;32mI (45331) modem_init: Name Server2: 84.235.6.55[0m
12:04:27.206 -> [0;32mI (45331) modem_init: ~~~~~~~~~~~~~~[0m
12:04:27.252 -> [0;32mI (45341) modem_init: GOT ip event!!![0m
12:04:41.003 -> [0;31mE (59111) modem_init: DNS lookup failed err=202 res=0x0[0m
12:04:56.008 -> [0;31mE (74111) modem_init: DNS lookup failed err=202 res=0x0[0m
12:05:11.014 -> [0;31mE (89111) modem_init: DNS lookup failed err=202 res=0x0[0m
12:05:26.001 -> [0;31mE (104111) modem_init: DNS lookup failed err=202 res=0x0[0m
12:05:41.012 -> [0;31mE (119111) modem_init: DNS lookup failed err=202 res=0x0[0m
12:05:55.980 -> [0;31mE (134111) modem_init: DNS lookup failed err=202 res=0x0[0m
12:06:11.010 -> [0;31mE (149111) modem_init: DNS lookup failed err=202 res=0x0[0m
12:06:26.000 -> [0;31mE (164111) modem_init: DNS lookup failed err=202 res=0x0[0m
```

AAl-Hadi, 2022-12-19:
@david-cermak I don't think it's a DNS problem
I commented out the return line to test (I know I shouldn't change the IDF files. It was just to test)
```
    if (_IS_NETIF_ANY_POINT2POINT_TYPE(esp_netif)) {
        // return ESP_ERR_NOT_SUPPORTED;
    }
```
I changes to Google DNS and still got the same errors
```
15:16:14.048 -> [0;32mI (25461) modem_init: Modem Connect to PPP Server[0m
15:16:14.048 -> [0;32mI (25471) modem_init: ~~~~~~~~~~~~~~[0m
15:16:14.048 -> [0;32mI (25471) modem_init: IP          : 10.36.138.72[0m
15:16:14.048 -> [0;32mI (25481) modem_init: Netmask     : 255.255.255.255[0m
15:16:14.048 -> [0;32mI (25481) modem_init: Gateway     : 10.64.64.64[0m
15:16:14.048 -> [0;32mI (25491) modem_init: change dns err = 0x0[0m
15:16:14.095 -> [0;32mI (25491) modem_init: change dns err = 0x0[0m
15:16:14.095 -> [0;32mI (25501) modem_init: Name Server1: 8.8.8.8[0m
15:16:14.095 -> [0;32mI (25501) modem_init: Name Server2: 8.8.4.4[0m
15:16:14.095 -> [0;32mI (25511) modem_init: ~~~~~~~~~~~~~~[0m
15:16:14.095 -> [0;32mI (25511) modem_init: GOT ip event!!![0m
15:16:27.680 -> [0;31mE (39111) modem_init: DNS lookup failed err=202 res=0x0[0m
15:16:42.660 -> [0;31mE (54111) modem_init: DNS lookup failed err=202 res=0x0[0m
15:16:57.697 -> [0;31mE (69111) modem_init: DNS lookup failed err=202 res=0x0[0m
15:17:12.660 -> [0;31mE (84111) modem_init: DNS lookup failed err=202 res=0x0[0m
15:17:27.692 -> [0;31mE (99111) modem_init: DNS lookup failed err=202 res=0x0[0m
15:17:42.687 -> [0;31mE (114111) modem_init: DNS lookup failed err=202 res=0x0[0m
```

david-cermak, 2023-01-02:
You can enable DNS debug in lwip component config. Also it's good to make sure that the DNS request is actually sent (and sent to the correct interface).
You can add this debug code to the modem network module: 

https://github.com/espressif/esp-protocols/blob/c7f20ee98e7a4fb5832044d56ffce93069caf37d/components/esp_modem/src/esp_modem_netif.cpp#L31-L40

```patch
        if (ppp->ppp_dte && ppp->ppp_dte->write((uint8_t *) buffer, len) > 0) {
+            ESP_LOG_BUFFER_HEXDUMP("DTE-transmit", buffer, len, ESP_LOG_INFO);
            return ESP_OK;
        }
```

This is how the DNS query might look like:
```
dns_enqueue: "mqtt.eclipseprojects.io": use DNS entry 0
dns_enqueue: "mqtt.eclipseprojects.io": use DNS pcb 0
dns_send: dns_servers[0] "mqtt.eclipseprojects.io": request
sending DNS request ID 44685 for name "mqtt.eclipseprojects.io" to server 0
I (8738) DTE-transmit: 0x3ffc8fc0   7e 21 45 00 00 45 xx xx  xx xx xx xx xx xx xx xx  |~!E..E....../.da|
I (8748) DTE-transmit: 0x3ffc8fd0   xx xx 08 08 08 08 0c b4  00 35 00 xx xx xx xx xx  |.f.......5.1....|
I (8758) DTE-transmit: 0x3ffc8fe0   01 00 00 01 00 00 00 00  00 00 04 6d 71 74 74 0f  |...........mqtt.|
I (8768) DTE-transmit: 0x3ffc8ff0   65 63 6c 69 70 73 65 70  72 6f 6a 65 63 74 73 02  |eclipseprojects.|
I (8778) DTE-transmit: 0x3ffc9000   69 6f 00 00 01 00 01 79  f9 7e                    |io.....y.~|
dns_recv: "mqtt.eclipseprojects.io": response = 137.135.83.217
I (9038) pppos_example: DNS lookup succeeded. IP=137.135.83.217
```

david-cermak, 2023-01-02:
@AAl-Hadi Does the default (pppos) example work for you, btw? This has to resolve the mqtt broker IP, too, so it might serve as a quick check if everything's okay on the modem/cellular network side.

If you're really using `SIM7600` you can also try the console example and resolve the hostname manually, for example:
```
esp> cmd AT+CDNSGIP="mqtt.eclipseprojects.io"
I (1906026) modem_console: Sending command AT+CDNSGIP="mqtt.eclipseprojects.io" with timeout 1000
I (1906406) modem_console: 
+CDNSGIP: 1,"mqtt.eclipseprojects.io","137.135.83.217"

OK

esp> $
```

AAl-Hadi, 2023-01-10:
Thank you @david-cermak . I think I figured it out.
I took the changes from this comment:
https://github.com/espressif/esp-idf/issues/8648#issuecomment-1084510528
Now it's working. SNTP and HTTP are working. I think the problem is maybe that I didn't integrate the modem libraries correctly?
All I did is copied 'esp_modem' folders and added the sdkconfig parts
Was there something else I need to do? Cause the code in the comment above works but this repo component doesn't work.

closed 2023-01-15 by AAl-Hadi