espressif/esp-protocols #200: How to use other modules for USB modem

tswen, 2022-12-28:
Hello,
When i use the pppos_client example, i'm trying to change vid, pid and if_inx to adapt ML302 module, but no effect.
```c
ESP_LOGI(TAG, "Initializing esp_modem for the BG96 module...");
struct esp_modem_usb_term_config usb_config = ESP_MODEM_DEFAULT_USB_CONFIG(0x1782, 0x4d10, 1); // VID, PID and interface num of BG96 modem
const esp_modem_dte_config_t dte_usb_config = ESP_MODEM_DTE_DEFAULT_USB_CONFIG(usb_config);
ESP_LOGI(TAG, "Waiting for USB device connection...");
esp_modem_dce_t *dce = esp_modem_new_dev_usb(ESP_MODEM_DCE_BG96, &dte_usb_config, &dce_config, esp_netif);
assert(dce);
esp_modem_set_error_cb(dce, usb_terminal_error_handler);
vTaskDelay(pdMS_TO_TICKS(1000)); // Although the DTE should be ready after USB enumeration, sometimes it fails to respond without this delay
```

I noticed esp_modem_new_dev_usb() first parameter is ESP_modem_DCE_BG96, Does it indicate that currently only BG96 modules can be used. Or is there a good suggestion to tell me how to adapt the new module. Thanks!

My ML302 module usb descriptor
```c
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x1782
idProduct 0x4d10
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1

*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 193
bNumInterfaces 8
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xe0
bMaxPower 400mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0xff
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 3
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x81	EP 1 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x1	EP 1 OUT
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 1
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0xff
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 3
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x82	EP 2 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x2	EP 2 OUT
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 2
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0xff
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 3
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x83	EP 3 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x3	EP 3 OUT
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 3
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0xff
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 3
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x84	EP 4 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x4	EP 4 OUT
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 4
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0xff
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 3
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x85	EP 5 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x5	EP 5 OUT
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 5
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0xff
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 3
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x86	EP 6 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x6	EP 6 OUT
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 6
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0xff
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 3
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x87	EP 7 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x7	EP 7 OUT
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 7
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0xff
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 3
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x88	EP 8 IN
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x8	EP 8 OUT
		bmAttributes 0x2	BULK
		wMaxPacketSize 64
		bInterval 0
```

tore-espressif, 2023-03-16:
Hello,
We have recently fixed several enumeration issues in `esp-idf` master. There are also few bugfixes in [esp_modem_usb_dte v1.1.0](https://components.espressif.com/components/espressif/esp_modem_usb_dte). All the fixes are already implemented in esp_modem examples, but currently without explicit support of ML302.
Could you please update your esp-idf and esp_modem and try again?

tore-espressif, 2023-05-04:
You can also refer to this short paragraph about adding new modems https://github.com/espressif/idf-extra-components/blob/master/usb/esp_modem_usb_dte/README.md#adding-a-new-modem

closed 2023-05-04 by tore-espressif