espressif/esp-protocols #201: esp-netif_lwip-ppp: User interrupt

Noushadalik, 2022-12-28:
Hi,
When running on CMUX example code after getting IP we can observe a user interrupt error
```
I (39639) esp-netif_lwip-ppp: Connected
I (39639) esp-netif_lwip-ppp: Name Server1: 10.95.83.1
I (39639) esp-netif_lwip-ppp: Name Server2: 8.8.8.8
I (39649) cmux_example: Modem Connect to PPP Server
I (39649) cmux_example: ~~~~~~~~~~~~~~
I (39659) cmux_example: IP          : 10.150.206.212
I (39659) cmux_example: Netmask     : 255.255.255.255
I (39669) cmux_example: Gateway     : 10.64.64.64
I (39669) cmux_example: Name Server1: 10.95.83.1
I (39679) cmux_example: Name Server2: 8.8.8.8
I (39679) cmux_example: ~~~~~~~~~~~~~~
I (39689) cmux_example: GOT ip event!!!
I (51699) esp-netif_lwip-ppp: User interrupt
I (51699) cmux_example: PPP state changed event 5
I (51699) cmux_example: User interrupted event from netif:0x3ffda698
```
We are subscribing to the PPP status events using the following.
`esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, ppp_event_handler, NULL);`
 We do not wish to use MQTT.
What to do to connect before the SNTP server after getting IP In the CMUX example code?

david-cermak, 2023-01-02:
> When running on CMUX example code after getting IP we can observe a user interrupt error

This is just an event (no error), when we ask the modem to exit PPP mode. Have you changed the example in some way?

> We do not wish to use MQTT.
What to do to connect before the SNTP server after getting IP In the CMUX example code?

Just copy the SNTP init code from the IDF example. There should be no difference to the WiFi/Ethernet example.

Noushadalik, 2023-01-04:
I am working on a simple CMUX example without any changes, I call SNTP init function within the CMUX example code after getting IP then got the Time.  
  ```
ESPEventHandlerSync event_handler(loop);
    event_handler.listen_to(ESPEvent(IP_EVENT, ESPEventID(ESP_EVENT_ANY_ID)));
    auto result = event_handler.wait_event_for(std::chrono::milliseconds(60000));
    if (result.timeout) {
        ESP_LOGE(TAG, "Cannot get IP within specified timeout... exiting");
        return;
    } else if (result.event.id == ESPEventID(IP_EVENT_PPP_GOT_IP)) {
        auto *event = (ip_event_got_ip_t *)result.ev_data;
        ESP_LOGI(TAG, "IP          : " IPSTR, IP2STR(&event->ip_info.ip));
        ESP_LOGI(TAG, "Netmask     : " IPSTR, IP2STR(&event->ip_info.netmask));
        ESP_LOGI(TAG, "Gateway     : " IPSTR, IP2STR(&event->ip_info.gw));
        std::cout << "Got IP address" << std::endl;
         **setclock();**
```
  logs follow
```
   Modem has correctly entered multiplexed command/data mode
Operator name:Vi India Vodafone IN
I (27008) esp-netif_lwip-ppp: Connected
I (27008) esp-netif_lwip-ppp: Name Server1: 10.95.83.1
I (27008) esp-netif_lwip-ppp: Name Server2: 8.8.8.8
I (27018) cmux_example: IP          : 10.150.112.118
I (27018) cmux_example: Netmask     : 255.255.255.255
I (27028) cmux_example: Gateway     : 10.64.64.64
Got IP address
I (27038) SNTP: Initializing SNTP
I (27038) SNTP: Your NTP Server is pool.ntp.org
I (27038) SNTP: Waiting for system time to be set... (1/10)
I (29048) SNTP: Waiting for system time to be set... (2/10)
I (31048) SNTP: Waiting for system time to be set... (3/10)
I (32438) SNTP: Notification of a time synchronization event
I (33048) Time: The current date/time is: Wed Jan  4 06:49:53 2023
I (33048) Time: timeinfo.tm_sec=53
I (33048) Time: timeinfo.tm_min=49
I (33048) Time: timeinfo.tm_hour=6
I (33048) Time: timeinfo.tm_wday=3
I (33058) Time: timeinfo.tm_mday=4
I (33058) Time: timeinfo.tm_mon=0
I (33058) Time: timeinfo.tm_year=123   
```
changed CMUX example code as  a function (gsm_test) then after this function I called SNTP init it is not working

```
extern "C" void app_main(void)
{
    gsm_test();
    setclock();
}
```

logs follow

```
Modem has correctly entered multiplexed command/data mode
Operator name:Vi India Vodafone IN
I (27008) esp-netif_lwip-ppp: Connected
I (27008) esp-netif_lwip-ppp: Name Server1: 10.95.83.1
I (27008) esp-netif_lwip-ppp: Name Server2: 8.8.8.8
I (27018) cmux_example: IP          : 10.151.28.176
I (27018) cmux_example: Netmask     : 255.255.255.255
I (27028) cmux_example: Gateway     : 10.64.64.64
Got IP address
I (27028) system_api: Base MAC address is not set
I (27038) system_api: read default base MAC address from EFUSE
 TOPIC:/topic/esp-modem
 DATA:Hello modem
Data received
Modem IMSI number:404466097514230
I (41688) esp-netif_lwip-ppp: User interrupt
I (41688) esp_modem_netif: PPP state changed event 5
I (41688) SNTP: Initializing SNTP
I (41688) SNTP: Your NTP Server is pool.ntp.org
I (41688) SNTP: Waiting for system time to be set... (1/10)
I (43698) SNTP: Waiting for system time to be set... (2/10)
I (45698) SNTP: Waiting for system time to be set... (3/10)
I (47698) SNTP: Waiting for system time to be set... (4/10)
I (49698) SNTP: Waiting for system time to be set... (5/10)
I (51698) SNTP: Waiting for system time to be set... (6/10)
I (53698) SNTP: Waiting for system time to be set... (7/10)
I (55698) SNTP: Waiting for system time to be set... (8/10)
I (57698) SNTP: Waiting for system time to be set... (9/10)
I (59698) Time: The current date/time is: Thu Jan  1 00:00:59 1970
I (59698) Time: timeinfo.tm_sec=59
I (59698) Time: timeinfo.tm_min=0
I (59698) Time: timeinfo.tm_hour=0
I (59698) Time: timeinfo.tm_wday=4
I (59708) Time: timeinfo.tm_mday=1
I (59708) Time: timeinfo.tm_mon=0
I (59708) Time: timeinfo.tm_year=70
```

david-cermak, 2023-01-04:
@Noushadalik Okay, understand now.

> changed CMUX example code as a function (gsm_test)

The reason is that the objects (`DCE`) are local to that function and generated as smart pointers, so when the function exits it cleans up locals and calls desrtuctors which gracerfully disconnect you from the network.

To fix that, you can make the DCE valid while you intend to use it. The easiest (but ugly) method it to make it a global variable. You can also use plain C API to init and deinit the connection.

Noushadalik, 2023-01-05:
Thank you for your precious support that now works well. I have one more question about GNSS. We are working with the sim7600 module. What changes are desired to the current simple cmux code to make GNSS work properly?

david-cermak, 2023-01-05:
The simple CMUX example already provides some related work (thanks to public contribution). You can start with enabling `CONFIG_EXAMPLE_MODEM_DEVICE_SIM7070_GNSS` and look at the changes, then adapt this for your device.

https://github.com/espressif/esp-protocols/blob/1d9b6f7e901d5af940cb3fc649c0d7ca0f649f75/components/esp_modem/examples/simple_cmux_client/main/simple_cmux_client_main.cpp#L98-L99

Noushadalik, 2023-01-17:
Thank you for your precious support that now works well.

Noushadalik, 2023-01-17:
One more question about OTA.  get an error like this 

```
W (157277) CMUX: Protocol mismatch: Missed trailing SOF, recovering...
I (157277) CMUX: Protocol recovered
W (157277) uart_terminal: HW FIFO Overflow
E (162607) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes
W (162607) HTTP_CLIENT: esp_transport_read returned:-26880 and errno:11
```
Any idea ?

david-cermak, 2023-01-17:
@Noushadalik Could you please try to enable this config option?
* [CONFIG_UART_ISR_IN_IRAM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-uart-isr-in-iram)

OTA performs intensive flash operations, which may cause the UART ISR to be missed from time to time. The protocol should  recover in most cases (extending network timeouts and UART Rx FIFO size and/or UART baud rate usually helps too).

david-cermak, 2023-03-27:
Thanks for the feedback, good to hear that it helps, closing