espressif/esp-protocols #210: SIMCOM A7672 reestarts after connecting to esp32 s3

guianmon99, 2023-01-19:
Hi, I am trying to connect to the module but every time it looks like the module is restarting after it connects. I am using the module console example of the master branch with the following configuration. 

struct esp_modem_usb_term_config usb_config = ESP_MODEM_DEFAULT_USB_CONFIG(0x1e0e, 0x9011, 5); // VID, PID and interface num of BG96 modem
        const esp_modem_dte_config_t dte_config = ESP_MODEM_DTE_DEFAULT_USB_CONFIG(usb_config);
        ESP_LOGI(TAG, "Waiting for USB device connection...");
        auto dte = create_usb_dte(&dte_config);
        dte->set_error_cb([&](terminal_error err) {
            ESP_LOGI(TAG, "error handler %d", err);
            if (err == terminal_error::DEVICE_GONE) {
                exit_signal.set(1);
            }
        });
        std::unique_ptr<DCE> dce = create_SIM7000_dce(&dce_config, dte, esp_netif);
could this be a problem with my IDF version I am on the 4.4.2 but also I have tried on the 4.4.3 please help. 

here are some logs from the console example

D (106384) cdc_acm: Checking list of connected USB devices
D (106434) cdc_acm: Checking list of connected USB devices
D (106484) cdc_acm: Checking list of connected USB devices
D (106534) HUB: Root port reset
D (106534) HUB: Stage done: START
D (106534) HUB: Stage done: GET_SHORT_DEV_DESC
D (106534) HUB: Stage done: CHECK_SHORT_DEV_DESC
D (106534) cdc_acm: Checking list of connected USB devices
D (106594) HUB: Stage done: SECOND_RESET
D (106594) HUB: Stage done: SET_ADDR
D (106594) HUB: Stage done: CHECK_ADDR
D (106594) HUB: Stage done: GET_FULL_DEV_DESC
D (106594) HUB: Stage done: CHECK_FULL_DEV_DESC
D (106594) HUB: Stage done: GET_SHORT_CONFIG_DESC
D (106604) HUB: Stage done: CHECK_SHORT_CONFIG_DESC
D (106604) HUB: Stage done: GET_FULL_CONFIG_DESC
D (106614) HUB: Stage done: CHECK_FULL_CONFIG_DESC
D (106614) HUB: Stage done: SET_CONFIG
D (106624) HUB: Stage done: CHECK_CONFIG
D (106624) HUB: Stage done: GET_SHORT_LANGID_TABLE
D (106634) cdc_acm: Checking list of connected USB devices
D (106664) HUB: Stage done: CLEANUP_FAILED
D (106674) HUB: Recovering root port
D (106674) HUB: Stage done: NONE
D (106684) cdc_acm: Checking list of connected USB devices
D (106734) cdc_acm: Checking list of connected USB devices

david-cermak, 2023-01-23:
Hi @guianmon99 

Does the device restart immediately after init, or is it after connecting as you write? The first issue might be probably related to VID and PID of your device. The second problem is typically a power supply issue.

`esp_modem` with USB is normally supported on IDF `v4.4`.

guianmon99, 2023-01-23:
Hi thank you for responding, The device reestarts just after the init it connects and then it reestarts. I have tried with some other brands of cellular modules, and there is no issue there but I really need to be able to include this simcom modules. Another thing I have noticed is that if I use the code from esp iot solutions the module connects and works just fine as well as with the other modules. however that api doesn't support a notification endpoint like this one so I would like to use cdc_acm_host driver but for some reason it doesn´t work with this modules. I think the code from iot solutions doesn't use the usb_host library maybe there is an issue there?.

guianmon99, 2023-01-23:
Today I  tried again and after changing the timeout for the connection got the following error.

D (10718) cdc_acm: Checking list of connected USB devices
D (10768) cdc_acm: Checking list of connected USB devices
D (10818) cdc_acm: Checking list of connected USB devices
D (10868) cdc_acm: Checking list of connected USB devices
D (10918) cdc_acm: Checking list of connected USB devices
E (10968) ESP_MODEM_THROW: esp_modem_usb.cpp:93 USB Device open failed

abort() was called at PC 0x4200c219 on core 0

after this the esp32 s3 restarts itself.

guianmon99, 2023-02-17:
Hi, got an update on this. I think there is something wrong with the USB enumeration process 

[image]

here is a snapshot I took with a serial analiser I don't think the second reset should be this  long

tore-espressif, 2023-03-26:
Hello @guianmon99 , you are right, we have fixed few enumeration issues in our USB host library recently. They are available on esp-idf `master` and `release/v4.4` branches.

guianmonezm, 2023-03-27:
@tore-espressif  Thank you yes on those branches the problem is solved. 
Excuse me do you have any plans to release a new stable branch with this changes? it is just that we have had problems before with release branches in production.

anyways you may close the issue

closed 2023-03-27 by tore-espressif