espressif/esp-protocols #213: C-APIs for CMUX manual mode are not working.

mallaprashant, 2023-01-30:
I am currently working on SIM7600 and I am trying to use CMUX manual mode C-APIs but not able to enter CMUX manual mode.

mallaprashant, 2023-01-30:
@david-cermak  when I use these ESP_MODEM_MODE_CMUX_MANUAL,  ESP_MODEM_MODE_CMUX_MANUAL_EXIT,  ESP_MODEM_MODE_CMUX_MANUAL_SWAP,  ESP_MODEM_MODE_CMUX_MANUAL_DATA, ESP_MODEM_MODE_CMUX_MANUAL_COMMAND, I am getting error on esp_modem_set_mode() function but I can use cmux , command and data mode

david-cermak, 2023-01-30:
Hi @mallaprashant 

Could you please describe which transitions don't work?
Please note that not all transitions withing `CMUX_MANUAL` are allowed (e.g. not possible to combine with auto mode), and not all devices support all transitions (e.g. A7670 fails to exit `CMUX_MANUAL` mode).

I've just tested basic transitions and they seem to work.
Used the default pppos client, but instead of switching to `DATA_MODE` here:

https://github.com/espressif/esp-protocols/blob/0e215b118f6302fe0d3109a373b248a1ccb6ff00/components/esp_modem/examples/pppos_client/main/pppos_client_main.c#L261-L265

Switched to `CMUX_MANUAL` with:

```cpp
    err = esp_modem_set_mode(dce, ESP_MODEM_MODE_CMUX_MANUAL);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_set_mode(ESP_MODEM_MODE_CMUX_MANUAL) failed with %d", err);
        return;
    }
    err = esp_modem_set_mode(dce, ESP_MODEM_MODE_CMUX_MANUAL_SWAP);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_set_mode(ESP_MODEM_MODE_CMUX_MANUAL_SWAP) failed with %d", err);
        return;
    }
    err = esp_modem_set_mode(dce, ESP_MODEM_MODE_CMUX_MANUAL_DATA);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_set_mode(ESP_MODEM_MODE_CMUX_MANUAL_DATA) failed with %d", err);
        return;
    }
```

Then waited for the mqtt to publish/subscribe and instead of going back to the `COMMAND_MODE` here:

https://github.com/espressif/esp-protocols/blob/0e215b118f6302fe0d3109a373b248a1ccb6ff00/components/esp_modem/examples/pppos_client/main/pppos_client_main.c#L289-L293

Used manual exit (plus sending some AT command to benefit from CMUX)

```cpp
    err = esp_modem_get_imsi(dce, imsi);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_get_imsi failed with %d", err);
        return;
    }
    ESP_LOGI(TAG, "IMSI=%s", imsi);
    err = esp_modem_set_mode(dce, ESP_MODEM_MODE_CMUX_MANUAL_COMMAND);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_modem_set_mode(ESP_MODEM_MODE_CMUX_MANUAL_COMMAND) failed with %d", err);
        return;
    }
```

And everything works as expected.

I'd suggest using the [`modem_console`](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console) example to exercise with the manual mode, the actual state and your device interaction.

mallaprashant, 2023-01-31:
@david-cermak 
Thanks for the reply.
I am working with SIM7600EI.
After creating DCE  and sending some AT commands . I tried to transit to CMUX_MANUAL mode. I am getting this error (ESP_ERR_NOT_SUPPORTED). But the transitions between COMMAND to DATA  and COMMAD to CMUX works fine.
[image]

david-cermak, 2023-01-31:
Oh, I think I understand now. you're using the latest version of [`esp-modem`](https://components.espressif.com/components/espressif/esp_modem) from component registry (`0.1.25` as of now) is that correct?
The manual CMUX mode support was added after that and we haven't bumped the version number yet, I'm sorry.

As a workaround, you can use the latest version from this repository and add it as a local dependency using `EXTRA_COMPONENT_DIR`.
(I'll also increase the version number at the earliest opportunity)

closed 2023-01-31 by david-cermak

mallaprashant, 2023-02-01:
@david-cermak thank you for the update.
 I tested new version of esp_modem and it worked.

lucle-bonboncar, 2024-05-30:
I am devloping with EC800 module, this is not support CMUX, so when I switch between DATA and AT mode, device have lost connection with the server, MQTT client is disconnected, any advise on that, how to avoid disconnected with server when transition from DATA mode to AT mode?