espressif/esp-protocols #220: PdpContext is not declared

rpirsc13, 2023-02-08:
Hi,I am having the following issue that I am unable to compile the example pppos_client

ESP-IDF:  v5.0
esp_modem : v0.1.26

```
In file included from xyz/managed_components/espressif__esp_modem/include/esp_modem_api.h:10,
                 from xyz/main/pppos_client.cpp:18:
xyz/managed_components/espressif__esp_modem/include/generate/esp_modem_command_declare.inc:127:78: error: 'PdpContext' has not been declared
  127 | ESP_MODEM_DECLARE_DCE_COMMAND(set_pdp_context, command_result, 1, STRUCT_OUT(PdpContext, p1)) \
      |                                                                              ^~~~~~~~~~
```

david-cermak, 2023-02-08:
@rpirsc13 Could you please share more details on how to reproduce the problem?

My guess is that you've just renamed the example's source `pppos_client.c` -> `pppos_client.cpp`, correct?
Note that `esp-modem` provides both C-API and C++ API and you have to choose which to use.
Keeping both APIs up to date and to reduce duplication, they are declared as macros and based on your language, the correct version is active.

By that renaming you have chosen C++ and you're trying to include the plain C API. This is unfortunately not possible.
Please use C++ API instead and replace `esp_modem_api.h` include with
```cpp
#include "cxx_include/esp_modem_api.hpp"
```

rpirsc13, 2023-02-08:
@david-cermak Your guess is correct.

I have used my test implementation from  pppos_example.c.

I was not aware that there are separate headers for cpp. Thank you for pointing that out. Everything works as expected now.

closed 2023-02-08 by rpirsc13

michaelboeding, 2023-08-18:
Im also having this issue. I have included them.

```c++
#include "cxx_include/esp_modem_api.hpp"
#include "cxx_include/esp_modem_buffer.hpp"
#include "cxx_include/esp_modem_cmux.hpp"
#include "cxx_include/esp_modem_command_library_utils.hpp"
#include "cxx_include/esp_modem_command_library.hpp"
#include "cxx_include/esp_modem_dce_factory.hpp"
#include "cxx_include/esp_modem_dce_module.hpp"
#include "cxx_include/esp_modem_dce.hpp"
#include "cxx_include/esp_modem_dte.hpp"
#include "cxx_include/esp_modem_exception.hpp"
#include "cxx_include/esp_modem_netif.hpp"
#include "cxx_include/esp_modem_primitives.hpp"
#include "cxx_include/esp_modem_terminal.hpp"
#include "cxx_include/esp_modem_types.hpp"

```

Can't seem to find these functions still

```c++

    
    esp_modem_dce_t *dce = esp_modem_new(&dte_config, &dce_config, esp_netif);

    

    // Attach the DCE to the network interface
    err = esp_modem_attach(dce, esp_netif);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Could not attach DCE to NetIF");
        // Handle error
    }

    // Finally, start the PPP mode
    err = esp_modem_start_ppp(dce);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Could not start PPP mode");
        // Handle error
    }
    // End of the missing par

```

david-cermak, 2023-08-22:
@michaelboeding Have you tried the suggestion from https://github.com/espressif/esp-protocols/issues/220#issuecomment-1422978061 ?

Are you compiling the snippet you shared above in C++? This library does not support using C-API from C++.
If you want to use C-API, please compile with C compiler (in IDF build system, your source file extension should be `*.c`)
If you want to use C++ compiler, please use C++ API and include `cxx_include/esp_modem_api.hpp`.