espressif/esp-protocols #224: Add Certificate Bundle Support

drguthrie, 2023-02-15:
Hello,

HTTP Client has support for the certificate bundle but WebsSocket Client does not. Is there any interest in adding support for the certificate bundle into WebSocket Client? I got it to work with only a few changes by following the code in the HTTP Client initialization method.

If there is interest I would be happy to make the changes and submit a pull request.

david-cermak, 2023-02-16:
Hello @drguthrie 

We welcome all contributions and the certificate bundle is a very useful addition to the websocket client.
Please check the [contribution guidelines](https://github.com/espressif/esp-protocols/blob/master/CONTRIBUTING.md) and post a PR with your changes. Thanks for the help!

david-cermak, 2024-03-18:
It seems to me that the certificate bundle is already supported:

https://github.com/espressif/esp-protocols/blob/cffe26818b432c428b8c1160bd8358b985d9d27c/components/esp_websocket_client/include/esp_websocket_client.h#L120

Implemented in d56b5d90

closed 2024-03-18 by david-cermak