espressif/esp-protocols #226: Crash on IDF v4.4.4 when using http_client

mbastida123, 2023-02-20:
Hi, we are developing a LTE solution based on SIM7080 and ESP32-S3.

When using v4.4.2 everything works fine:

> I (54503) esp_modem_netif: PPP state changed event 5
> Modem has correctly entered multiplexed command/data mode
> Operator name:Orange
> I (69893) esp-netif_lwip-ppp: Connected
> I (69893) esp-netif_lwip-ppp: Name Server1: 85.62.229.135
> I (69893) esp-netif_lwip-ppp: Name Server2: 85.62.229.136
> I (69893) dom HTTPS Modem OTA: Modem Connect to PPP Server
> I (69903) dom HTTPS Modem OTA: ~~~~~~~~~~~~~~
> I (69913) dom HTTPS Modem OTA: IP          : 10.2.13.178
> I (69913) dom HTTPS Modem OTA: Netmask     : 255.255.255.255
> I (69923) dom HTTPS Modem OTA: Gateway     : 10.64.64.64
> I (69933) dom HTTPS Modem OTA: Name Server1: 85.62.229.135
> I (69933) dom HTTPS Modem OTA: Name Server2: 85.62.229.136
> I (69943) dom HTTPS Modem OTA: ~~~~~~~~~~~~~~
> I (69943) dom HTTPS Modem OTA: GOT ip event!!!
> I (69983) dom HTTPS Modem OTA: MODEM started OK
> I (74983) dom_https_ota_update_4G: Running firmware version: 1.1-test
> I (74983) system_api: Base MAC address is not set
> I (74983) system_api: read default base MAC address from EFUSE
> I (74983) dom_https_ota_update_4G: MAC address of device: 68:b6:b3:45:d2:40
> I (74993) dom_https_ota_update_4G: URL: https://mockdb.dom.net/ota?mac=68:b6:b3:45:d2:40&version=1.1-test
> I (75023) dom_https_ota_update_4G: Starting OTA 4G
> I (76193) esp-x509-crt-bundle: Certificate validated
> I (77283) esp_https_ota: Starting OTA...
> I (77283) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
> E (77283) esp_https_ota: Complete headers were not received
> E (77293) dom_https_ota_update_4G: esp_https_ota_read_img_desc failed
> Return of function OTA_Start_Modem: -3
> I (78573) esp-x509-crt-bundle: Certificate validated
> dom HTTPS GET -> Code: 200, Buffer: {
>   "args": {}, 
>   "headers": {
>     "Content-Length": "0",
>     "Host": "httpbin.org",
>     "User-Agent": "ESP32 HTTP Client/1.0",
>     "X-Amzn-Trace-Id": "Root=1-63f33633-49a9eccd01bfd9142677c6fb"
>   }, 
>   "origin": "86.104.187.21",
>   "url": "https://httpbin.org/get"
> }
> I (88243) dom HTTPS Modem OTA: DNS lookup success
> I (91213) esp-x509-crt-bundle: Certificate validated
> dom HTTPS POST -> Code: 200, Buffer: {
>   "args": {}, 
>   "data": "{\"clave\":\"valor\"}",
>   "files": {},
>   "form": {},
>   "headers": {
>     "Content-Length": "17",
>     "Content-Type": "application/json",
>     "Host": "httpbin.org", 
>     "User-Agent": "ESP32 HTTP Client/1.0",
>     "X-Amzn-Trace-Id": "Root=1-63f33640-0fd6d8f706d5afe44804d398"
>   },
>   "json": {
>     "clave": "valor"
>   },
>   "origin": "86.104.187.21",
>   "url": "https://httpbin.org/post"
> }
> I (98243) dom HTTPS Modem OTA: DNS lookup success
> I (106173) esp-x509-crt-bundle: Certificate validated
> dom HTTPS DELETE -> Code: 200, Buffer: {
>   "args": {
>     "clave": "valor"
>   },
[... 319 lines cut ...]
> D (106222) HTTP_CLIENT: content_length = 375
> D (106232) HTTP_CLIENT: data_process=0, content_length=375
> D (106232) HTTP_CLIENT: http_on_body 375
> D (106242) HTTP_CLIENT: http_on_message_complete, parser=3fc9dd48
> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  0 register dump:
> PC      : 0x400556d5  PS      : 0x00060230  A0      : 0x8208d974  A1      : 0x3fcf56e0
> A2      : 0x8037daa8  A3      : 0x8037daa4  A4      : 0x000000ff  A5      : 0x0000ff00  
> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fcf56a0  
> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3fcf5924  A13     : 0x00000000
> A14     : 0x3fcf05b4  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
> EXCVADDR: 0x8037daa8  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff
> 
> 
> Backtrace: 0x400556d2:0x3fcf56e0 |<-CORRUPTED
> 
> 
> 
> 
> ELF file SHA256: 87df058d3e9ad9aa
> 
> Rebooting...

What I have found is that it doesn't always fail on the same point. Sometimes it fails on the TLS checkup too.

david-cermak, 2023-02-20:
I'm sorry I cannot tell much about the issue from the log posted above, the backtrace is corrupted. The last debug message from `http_on_message_complete()` looks correct including the `parser`'s context pointer (0x3fc9dd48), then (after some time?) we can see Load-Prohibited  error trying to access `0x8037daa8`, which is not an address. You can check what's on the PC address with `addr2line`, but that might not be related at all.

> What I have found is that it doesn't always fail on the same point. Sometimes it fails on the TLS checkup too.

Could you please also post the other crashes? Maybe some would be more readable?

You can also try to find the offending commit with `git bisect`, which tells me you would need about 10 runs to find out.
```
$ git bisect good v4.4.2 
$ git bisect bad v4.4.4 
Bisecting: 373 revisions left to test after this (roughly 9 steps)
```

mbastida123, 2023-02-21:
> I'm sorry I cannot tell much about the issue from the log posted above, the backtrace is corrupted. The last debug message from `http_on_message_complete()` looks correct including the `parser`'s [...]

Hi @david-cermak and thanks for the quick response.

The panic was not the core error. This panic was due to the esp_http_client_fetch_headers returning 0 and my code not being prepared for this situation (the malloc for the recieve buffer was not done and so the panic was thrown). Once I discover more I will follow up on the issue (if there is one).

mbastida123, 2023-02-21:
> I'm sorry I cannot tell much about the issue from the log posted above, the backtrace is corrupted. The last debug message from `http_on_message_complete()` looks correct including the `parser`'s [...]

Upon further inspection the problem is here:

```
int content_length =  esp_http_client_fetch_headers(client);
printf("Headers Length: %d", content_length);
```

> Headers Length: -1

It returns -1 (only on the case of POST resquest)

Before that I simply do:

```
esp_http_client_config_t config = {
                .url = urlPost,
                .crt_bundle_attach = esp_crt_bundle_attach
            };
            esp_http_client_handle_t client = esp_http_client_init(&config);

            esp_http_client_set_url(client, urlPost);
            esp_http_client_set_method(client, HTTP_METHOD_POST);
            esp_http_client_set_header(client, "Content-Type", "application/json");
            esp_http_client_set_post_field(client, post_data, strlen(post_data));

            esp_err_t err;

            if ((err = esp_http_client_open(client, strlen(post_data))) != ESP_OK) {
                ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
                free(bufferResponse);
                respR.statusCode=-1;
                respR.buffResp = NULL;

                esp_http_client_close(client);
                esp_http_client_cleanup(client);
                /*esp_http_client_close(client);
                esp_http_client_cleanup(client);
                free(bufferResponse);*/

                xSemaphoreGive( xSemaphoreModem );
                return respR;
            }
```

I have tried git bisect but it only shows the diferences on the submodule file. I can see that lwip has changed...

The URL I'm doing the requests is httpbin.org. So I doubt that is the problem. 
Moreover, the code for the GET and DELETE requests is equal to the above and those don't return -1 when fetching headers.

david-cermak, 2023-02-22:
Looking at the relevant commits between `v4.4.2` and `v4.4.4`, it looks like https://github.com/espressif/esp-idf/commit/62884dae3a might be related? Do you expect any re-directions?
But the general suggestion is IMO to always for error codes, as fetching headers doesn't have to always succeed.

mbastida123, 2023-02-24:
> Looking at the relevant commits between `v4.4.2` and `v4.4.4`, it looks like [espressif/esp-idf@62884dae3a](https://github.com/espressif/esp-idf/commit/62884dae3a) might be related? Do you expect any re-directions? But the general suggestion is IMO to always for error codes, as fetching headers doesn't have to always succeed.

Yes I have seen that. Now I'm using the http_event_handler approach and it seems to work fine.

closed 2023-02-24 by mbastida123