espressif/esp-protocols #235: Understanding versioning, releasing, changelog of components

llange, 2023-02-27:
(This is more a general question than an issue)

Sorry if this has already been discussed (I did not find it), but I'm a little confused how those components are versionned, released, and documented.

Let's take mdns as an example : If I go https://components.espressif.com/components/espressif/mdns, I see that the current version is `1.0.8`.

But on this page, I have no information about the Changelog between the version that I'm using (`1.0.7`) and the newer one (`1.0.8`).

Going to [the source repository](https://github.com/espressif/esp-protocols/tree/master/components/mdns), I find that the current version is displayed as `1.0.7` in the [README.md](https://github.com/espressif/esp-protocols/blob/master/components/mdns/README.md) file (it seems like a caching issue with the badge, but still it's disturbing).
(But the correct version is set in [idf_component.yml](https://github.com/espressif/esp-protocols/blob/master/components/mdns/idf_component.yml))

No Changelog here neither.

So next try is to do a diff between the previous tag (1.0.7) and the current tag (1.0.8).

And it's here that I discover that there are [no tags set at all in the whole repository](https://github.com/espressif/esp-protocols/tags)... which definitely confuses me (and of course, the idea of having different components in the same repository but with a different version number each does add to the confusion)

How can we match a released version of a component (one that has been deployed to the component registry like https://components.espressif.com/components/espressif/mdns ) and the state of the source files in the source control (git in GitHub) ?

How can we track the changes done from one version to another (diff of the files, etc...) ?

How can we know which new feature / bug fix / .... is present in one version ?

Thanks !

david-cermak, 2023-03-10:
@llange Thanks for raising this important question, we indeed neglected the versioning and updating a changelog. We'll address this problem in multiple steps:

1) Reintroduce missing change logs for all components (https://github.com/espressif/esp-protocols/pull/247)
2) Using conventional commits in this repository (https://github.com/espressif/esp-protocols/pull/236), so we're able to generate changelog and tag versions automatically
3) Update the changelogs and apply tags on any component version bump

llange, 2023-03-10:
@david-cermak thank you for taking the time to address this, very appreciated.
and thanks for all the work on those very useful components.

The Changelogs looks very detailed and I'd suggest to link them in both the README.md file and the component library ( e.g. https://components.espressif.com/components/espressif/mdns ) so that everybody can benefit from them.

Regarding the tags, what would be the plan ? A kind of naming convention ?

david-cermak, 2023-03-13:
> Regarding the tags, what would be the plan ? A kind of naming convention ?

yes, will probably use the component name as a prefix, e.g. `mdns-v1.0.0`.
cz based toolset is not perfect for development in monorepo, but still allows for defining/scoping a custom tag format. Would probably have to use one config per each component.

Note: Reopening, as the point `3)` is still not addressed

david-cermak, 2023-10-30:
All components are now follow the release process, described in 

https://github.com/espressif/esp-protocols/blob/master/CONTRIBUTING.md#release-process

closed 2023-10-30 by david-cermak