espressif/esp-protocols #238: components/esp_websocket_client/examples

cinargursoy, 2023-03-01:
Hello,

I'm trying to use `esp_websocket_client` component in my project.  But I get errors below:
```
/Users/cinargursoy/esp/esp-protocols/components/esp_websocket_client/esp_websocket_client.c: In function 'set_websocket_transport_optional_settings':
/Users/cinargursoy/esp/esp-protocols/components/esp_websocket_client/esp_websocket_client.c:448:14: error: 'esp_transport_ws_config_t' has no member named 'auth'
  448 |             .auth = client->config->auth,
      |              ^~~~
/Users/cinargursoy/esp/esp-protocols/components/esp_websocket_client/esp_websocket_client.c:449:41: warning: initialized field overwritten [-Woverride-init]
  449 |             .propagate_control_frames = true
      |                                         ^~~~
/Users/cinargursoy/esp/esp-protocols/components/esp_websocket_client/esp_websocket_client.c:449:41: note: (near initialization for 'config.propagate_control_frames')
/Users/cinargursoy/esp/esp-protocols/components/esp_websocket_client/esp_websocket_client.c: In function 'esp_websocket_client_task':
/Users/cinargursoy/esp/esp-protocols/components/esp_websocket_client/esp_websocket_client.c:849:70: error: implicit declaration of function 'esp_transport_ws_get_upgrade_request_status' [-Werror=implicit-function-declaration]
  849 |                 client->error_handle.esp_ws_handshake_status_code  = esp_transport_ws_get_upgrade_request_status(client->transport);
      |                                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[840/849] Building C object esp-idf/wifi_provisio...les/__idf_wifi_provisioning.dir/src/manager.c.objninja: build stopped: subcommand failed.
HINT: Please check that the function name is correct. Also it is possible that you've forgot to import esp_transport_ws_get_upgrade_request_status library(s) in header file or add the necessary REQURIES component. Try to add missing libraries to your project header file or check idf_component_register(REQUIRES ...) section in your component CmakeList.txt file. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the function has been removed, renamed or replaced by an alternative function - refer to the migration guide for more information.
```

I'm also getting the same error while trying to build `components/esp_websocket_client/examples`.
My current,
IDF version:  _release/v5.0_  -> `886e98a2c1`, but I also tried on `885e501d99 `
esp-protocols version: _master_ -> `3bf0511`

gabsuren, 2023-03-01:
Hello, thank you for your question. 
Please use latest IDF version: release/v5.0 -> `564b94fe5b`

gabsuren, 2023-03-01:
I am sorry. I checked again and it seems it is not available in public `esp-idf:release-v5.0` repository yet. It will be available soon. 
Alternatively you can use `esp-idf-master`, the changes are there and should be back-ported in `esp-idf:release-v5.0` soon

gabsuren, 2023-03-01:
And another option is (in case if you want to use `esp-idf:release-v5.0` ) you can cherry-pick those two commits `9def9b6a03`
`fecf4768f0`
manually

cinargursoy, 2023-03-02:
Hi Suren,

Thanks for quick reply! It worked with,
IDF version: _master_ -> `94c87a9106`
esp-protocols version: _master_ -> `3bf0511`

I haven't tried cherry-picking yet, but it seems like  it will work. I'll cherry-pick and reply the results.
What would be the predicted date for this commits to be available in `esp-idf:release-v5.0`?

gabsuren, 2023-03-03:
The changes probably will be available in `esp-idf:release-v5.0` within next week. 
I will notify when they will be merged

gabsuren, 2023-03-06:
@cinargursoy the changes already available in public `esp-idf:release-v5.0`. 
Please update the branch and check it again.
In case of any question/issue please let us know

cinargursoy, 2023-03-13:
Hi Suren,

I've checked and it works. Thank you!

closed 2023-03-13 by cinargursoy