espressif/esp-protocols #250: Unable to send data in PPP mode.

yanmartins, 2023-03-14:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I'm working in a project using ESP32-S3 and Telit module for use mobile data.

I can make the module send data over the network with the default baud rate module (115200). But when I change the baud rate to any other value, I can't communicate over the network.

This is the code snippet where I change the baudrate of the ESP UART and the Telit module UART:

```c
#define DEFAULT_BAUDRATE 115200
#define MAX_BAUDRATE        921600

uart_set_baudrate(UART_MODEM, DEFAULT_BAUDRATE);
if(modem_handle->dce->set_baud(MAX_BAUDRATE) == command_result::OK) uart_set_baudrate(UART_MODEM, MAX_BAUDRATE);
```

As you can see bellow, even with a higher baud rate, I got the IP event. But I am unable to communicate in PPP mode.

```
W (11233) srv_modem_manager: New baudrate: 921600
I (19693) srv_mobile_network: IP event! 6
I (19693) srv_mobile_network: Modem Connected to PPP Server
I (19693) srv_mobile_network: ~~~~~~~~~~~~~~
I (19693) esp-netif_lwip-ppp: Connected
I (19693) srv_mobile_network: IP          : 10.160.68.106
I (19703) srv_mobile_network: Netmask     : 255.255.255.255
I (19713) srv_mobile_network: Gateway     : 10.64.64.64
I (19713) srv_mobile_network: Name Server1: 172.31.128.53
I (19723) srv_mobile_network: Name Server2: 172.31.0.53
I (19733) srv_mobile_network: ~~~~~~~~~~~~~~
```

Do I need to do some more configuration? I found little information about it.

Thanks in advance,
Yan

david-cermak, 2023-03-15:
This is weird, you received an IP with the new baudrate, so the PPP link works correctly.
I've quickly tested this scenario with ESP32+SIM7600 and it works as expected, after applying your snippet (just after init), got an IP and was able to ping/http to the internet.

Could you please enable verbose logging and share the result? Would be also helpful to see what (data) we're trying to send/receive, could you also add these logs into the netif module?
```patch
diff --git a/components/esp_modem/src/esp_modem_netif.cpp b/components/esp_modem/src/esp_modem_netif.cpp
index 75e271d..f328994 100644
--- a/components/esp_modem/src/esp_modem_netif.cpp
+++ b/components/esp_modem/src/esp_modem_netif.cpp
@@ -32,6 +32,7 @@ esp_err_t Netif::esp_modem_dte_transmit(void *h, void *buffer, size_t len)
 {
     auto *ppp = static_cast<Netif *>(h);
     if (ppp->signal.is_any(PPP_STARTED)) {
+        ESP_LOG_BUFFER_HEXDUMP("DTE transmit", (uint8_t *) buffer, len, ESP_LOG_VERBOSE);
         if (ppp->ppp_dte && ppp->ppp_dte->write((uint8_t *) buffer, len) > 0) {
             return ESP_OK;
         }
@@ -66,6 +67,7 @@ esp_err_t Netif::esp_modem_post_attach(esp_netif_t *esp_netif, void *args)
 void Netif::receive(uint8_t *data, size_t len)
 {
     if (signal.is_any(PPP_STARTED)) {
+        ESP_LOG_BUFFER_HEXDUMP("DTE recv", (uint8_t *) data, len, ESP_LOG_VERBOSE);
         esp_netif_receive(driver.base.netif, data, len, nullptr);
     }
 }
```

david-cermak, 2023-03-27:
@yanmartins Any update with this? Have you succeeded in running the PPP mode in higher baudrates? Note that this would also depend on wiring, how do you connect the ESP32 with the device? Connectors, some mezzanine board, custom board, loose wires? I think that running this on 921600 would probably require routing carefully, even if you design a custom board.

david-cermak, 2023-11-28:
Tested on `921600` and works as expected. Also added an iperf test in https://github.com/espressif/esp-protocols/pull/354

Please reopen with more details

barbiani, 2025-10-31:
This could be related to the flow control mechanism? Can the modem accept a full frame at those speeds?