espressif/esp-protocols #280: implicit swap of Terminals gone ?

diplfranzhoepfinger, 2023-04-22:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### What component are you using? If you choose Other, provide details in More Information.

esp_modem

### component version

1.0.0

### IDF version.

ESP-IDF Release v5.0

### More Information.

Hi, 

before when i made 

```
dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_MODE)
dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_DATA)
```

then the Terminals were automatically swapped after this. 
so the Primary Terminal was AT Commands, and the secondary was Data. 

do i now need to write: 

```
dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_MODE)
dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_DATA)
dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_SWAP)
```
or is this a misunderstanding ?

diplfranzhoepfinger, 2023-04-22:
seems not to help. in both cases i land in the Data-Mode Terminal ...

diplfranzhoepfinger, 2023-04-22:
maybe the Gist helps: 

[https://gist.github.com/diplfranzhoepfinger/8411a48f3493d197cb3f5813d37aa4da](https://gist.github.com/diplfranzhoepfinger/8411a48f3493d197cb3f5813d37aa4da)

franz-ms-muc, 2023-04-22:
do i need this : 

enter some ms here ? 

[image]

david-cermak, 2023-04-26:
@franz-ms-muc You have to swap the terminals first and then switch modes:
```cpp
dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_MODE)
dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_SWAP)
dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_DATA)
```

or, from the console example:
```bash
esp> set_mode CMUX1
I (172679) modem_console: Switching to CMUX1 name...
I (172809) modem_console: OK
esp> cmd at
I (174619) modem_console: Sending command at with timeout 1000
I (174629) modem_console: 
OK

esp> set_mode CMUX3
I (188589) modem_console: Switching to CMUX3 name...
I (188589) modem_console: OK
esp> set_mode CMUX4
I (192719) modem_console: Switching to CMUX4 name...
I (192769) modem_console: OK
esp> cmd at
I (197379) modem_console: Sending command at with timeout 1000
I (197389) modem_console: 
OK

esp> I (198829) esp-netif_lwip-ppp: Connected
esp> 
esp> ping 8.8.8.8
esp> I (205329) modem_console_ping: 64 bytes from 8.8.8.8 icmp_seq=1 ttl=53 time=317 ms
```

diplfranzhoepfinger, 2023-04-27:
Still same Problem: 

[https://gist.github.com/diplfranzhoepfinger/c8ad5b9db6e1d602ab5a6fb287463ce6](https://gist.github.com/diplfranzhoepfinger/c8ad5b9db6e1d602ab5a6fb287463ce6)

this was all working prior to 1.0.0

same behavoiur if i go for CMUX\_MODE without the "Manual" part like in the Simple CMUX Example.

diplfranzhoepfinger, 2023-04-27:
what does it mean: 

```
W (34819) CMUX: Protocol mismatch: Missed trailing SOF, recovering...
I (34829) CMUX: Protocol recovered
```

diplfranzhoepfinger, 2023-04-27:
this log is made with the older esp-modem Version: 

[https://gist.github.com/diplfranzhoepfinger/d442af49f701b3277d7bbc4213cc6f78](https://gist.github.com/diplfranzhoepfinger/d442af49f701b3277d7bbc4213cc6f78)

works good there. 

this was the time where DUAL\_MODE was not present in esp\_modem\_types.hpp

diplfranzhoepfinger, 2023-04-27:
https://github.com/espressif/esp-protocols/commit/44bae24c78d15f8b0a028e3016d5043c1198fe01

could it be that this commit damaged the CMUX ?

david-cermak, 2023-04-28:
Are you still seeing the issue when running some default example?
I've checked the latest version with `CONFIG_ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD` set to both `true` and `false` and worked as expected in both configs.

Could you please share some snippet of code where it doesn't work for you?

diplfranzhoepfinger, 2023-04-28:
i will

diplfranzhoepfinger, 2023-05-19:
> Are you still seeing the issue when running some default example? 

I made a Test: 

here : 

https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client

using latest ESP-Modem and latest ESP-IDF.

on Commit 38895e69857a5c15773fd9ee056b40ca98fa216b i get this Result at first Start: 

https://gist.github.com/diplfranzhoepfinger/9a7b6efd77f8f780c43b0e7df5ef03d2

and after a Reset of the ESP32 ONLY: 

https://gist.github.com/diplfranzhoepfinger/64f387372bd1ec94596b519b4c799dd4

so it seems to work at least, but does hang after 1st Start. 
2-wire Mode.

diplfranzhoepfinger, 2023-05-19:
i will enable advanced logging, to see what is going on here:

diplfranzhoepfinger, 2023-05-19:
see here: 
https://gist.github.com/diplfranzhoepfinger/47d598841c7364469147326379d76e8f
First Start CMUX not working.

diplfranzhoepfinger, 2023-05-19:
i give it more Time to Boot the Modem: 
https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/commit/e340a4c1c20dff22c69d694ee7582589c5f864a8

diplfranzhoepfinger, 2023-05-19:
that solved it !

diplfranzhoepfinger, 2023-05-19:
https://gist.github.com/diplfranzhoepfinger/067537a3630a9ae337373e60c69d4b68

diplfranzhoepfinger, 2023-05-19:
same with Normal Logging Level: 

https://gist.github.com/diplfranzhoepfinger/64fee371bf31594f8331ca4ce8630f39

so the Pause seems to help. !

diplfranzhoepfinger, 2023-06-15:
and it is crashing this way:

diplfranzhoepfinger, 2023-06-15:
> @franz-ms-muc You have to swap the terminals first and then switch modes:
> 
> ```c++
> dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_MODE);
> dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_SWAP);
> dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_DATA);
> ```

i have incorporated this here: 

[https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/ManualCMUX](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/ManualCMUX)

and get a chrash: 

[https://gist.github.com/diplfranzhoepfinger/37fc98d4b35bc615fe3dbd121764d3cc](https://gist.github.com/diplfranzhoepfinger/37fc98d4b35bc615fe3dbd121764d3cc)

Unfortunately the Modem is also hung after this, so it cannot be switched off after this Operation. 

(which might be a Problem inside the SIM A7672 Firmware)

diplfranzhoepfinger, 2023-06-15:
https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/NormalCMUX

diplfranzhoepfinger, 2023-06-15:
INFO Log

https://gist.github.com/diplfranzhoepfinger/9214050627573aa7d4378ee7c7404530

diplfranzhoepfinger, 2023-06-15:
Verbose Log for "normal CMUX" ... 

[https://gist.github.com/diplfranzhoepfinger/bd58eccf675a2878bda3aaa87dbc27ea](https://gist.github.com/diplfranzhoepfinger/bd58eccf675a2878bda3aaa87dbc27ea)

no Crash

diplfranzhoepfinger, 2023-06-15:
found out this: 

> @franz-ms-muc You have to swap the terminals first and then switch modes:
> 
> ```c++
> dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_MODE);
> dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_SWAP);
> dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_DATA);
> ```
> 

seems to be wrong. 

> 
> ```c++
> dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_MODE);
> dce->set_mode(esp_modem::modem_mode::CMUX_MANUAL_DATA);
> ```
> 

but this seems to work. 

log:
https://gist.github.com/diplfranzhoepfinger/32af76599f07182c7f33d8509372b76e

code:
https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/ManualCMUX_no_swap

david-cermak, 2023-10-10:
Manual CMUX is still working, I'm currently adding the sequence to one of the test. It should be:

https://github.com/espressif/esp-protocols/blob/dee2df69ef4a31bfe179fae885db7ef3d10e7e3e/components/esp_modem/test/target_ota/main/simple_cmux_client_main.cpp#L154-L156

Feel free to reopen in case of any trouble

closed 2023-10-10 by david-cermak