espressif/esp-protocols #281: I need an sim card insert and remove event when using bg96 and ppos lin=b which isused in esp-idf v 4.3

radhikapatel15, 2023-04-26:
### Describe the solution you'd like.

I want the sim insert and remove events and also an event when modem fails for esp-idf v4.3

david-cermak, 2023-04-28:
I think you get a *lost connection* event immediately after removing a SIM card.
So, as a workaround you can listen to already supported events and just check (with AT commands) if the card has or hasn't been removed/inserted.
The correct solution would probably be to listen to URC messages in the modem library, but that would need some work (and probably some customization) to make it universally working for most devices.

radhikapatel15, 2023-04-28:
Thanks for your reply, But what u are saying will conflict when the server connection is also lost .
And for the workaround can share any sample code for it ? using bg96 and ppos lib for esp-idf v4.3?

diplfranzhoepfinger, 2023-07-03:
ah, cool. i will try this !

david-cermak, 2023-11-28:
Should be handled by an URC message, will be added in `v1.2.0`

david-cermak, 2024-11-07:
URC handling has been merged in https://github.com/espressif/esp-protocols/pull/620

will be published as v1.2.0 in https://github.com/espressif/esp-protocols/pull/688

closed 2024-11-07 by david-cermak