espressif/esp-protocols #285: Crashing when receiving NO CARRIER from modem

peergum, 2023-04-27:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When modem antenna is unplugged(to reproduce error, likely happens with a very weak signal too), I get a NO CARRIER error, and my app crashes. Below is a log dump:
```D (139774) esp_netif_handlers: esp_netif action stopped with netif0x3ffbe4a0 from event_id=0
D (139774) esp-netif_lwip-ppp: esp_netif_stop_ppp: Stopped PPP connection: 0x3ffbe700
ppp_close[1]: kill_link -> lcp_close
ppp phase changed[1]: phase=11
ppp phase changed[1]: phase=6
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 0 0
ppp_recv_config[1]
sent [LCP TermReq id=0x4 "User request"]
pppos_write[1]: len=20
D (140804) esp-modem: debug_data: 0x3ffc10c8   0d 0a 4e 4f 20 43 41 52  52 49 45 52 0d 0a        |..NO CARRIER..|
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008b3f8  PS      : 0x00060e33  A0      : 0x800e748c  A1      : 0x3ffc0d90  
0x4008b3f8: xEventGroupSetBits at /Users/phil/esp/esp-idf-5.0/components/freertos/FreeRTOS-Kernel/event_groups.c:628

A2      : 0x3ffc0690  A3      : 0x00000008  A4      : 0x3ffc069c  A5      : 0x00000000  
A6      : 0x3ffc06ac  A7      : 0x0000cdcd  A8      : 0x8008b3ea  A9      : 0x3ffc0d70  
A10     : 0x00060820  A11     : 0x3ffb2be0  A12     : 0x00060e20  A13     : 0x00060e23  
A14     : 0x00000002  A15     : 0x0000cdcd  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00060824  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffec  

Backtrace: 0x4008b3f5:0x3ffc0d90 0x400e7489:0x3ffc0db0 0x400e7a2b:0x3ffc0dd0 0x400e7bfa:0x3ffc0e10 0x4008b59d:0x3ffc0e30
0x4008b3f5: xEventGroupSetBits at /Users/phil/esp/esp-idf-5.0/components/freertos/FreeRTOS-Kernel/event_groups.c:626

0x400e7489: esp_modem::SignalGroup::set(unsigned long) at /Users/phil/dev/hydroficient/components/espressif__esp_modem/src/esp_modem_primitives_freertos.cpp:45

0x400e7a2b: esp_modem::UartTerminal::task() at /Users/phil/dev/hydroficient/components/espressif__esp_modem/src/esp_modem_uart.cpp:121

0x400e7bfa: esp_modem::UartTerminal::s_task(void*) at /Users/phil/dev/hydroficient/components/espressif__esp_modem/src/esp_modem_uart.cpp:75

0x4008b59d: vPortTaskWrapper at /Users/phil/esp/esp-idf-5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```

david-cermak, 2023-04-28:
@peergum Could you please share the initial portion of your code? Based on the backtrace you shared, the crash's happening in the uart communication loop:

https://github.com/espressif/esp-protocols/blob/31187b7d1732556ca5f0e260c91f290b72670fa0/components/esp_modem/src/esp_modem_uart.cpp#L121

and seeing `LoadProhibited`, it seems like the content of this `signal` object (which is basically FreeRTOS's EventGroups) is getting either corrupted or used after `delete`.
What kind of interface are you using? Is it C++ interface, are you sure that the DTE is still valid, e.g. after exiting a scoped block?

peergum, 2023-05-02:
I'm using C++ indeed. I'll update here ASAP with more details.

david-cermak, 2023-10-10:
@peergum Any update from your side? Could you please share your init code?

peergum, 2023-10-10:
Hey DavidSorry it may still happen but I’ve not seen it happening in a while. I think you can make it happen when sending a hang up to the modem at any time.Phil

david-cermak, 2023-10-31:
> when sending a hang up to the modem at any time.

tested this scenario with a few modems, but worked as expected. Let me close the issue and ask you to reopen it if it happens again.