espressif/esp-protocols #294: websocket client build failed with esp-idf version 5.1

hoangdovan, 2023-05-14:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### What component are you using? If you choose Other, provide details in More Information.

esp_websocket_client

### IDF version.

v5.1-dev-2356-gedd815af2e-dirty

### More Information.

Hello, I have just try to compile esp_websocket_client component with esp-idf v5.1. Then there are some error happen:

managed_components/espressif__esp_websocket_client/esp_websocket_client.c:448:14: error: 'esp_transport_ws_config_t' has no member named 'auth'
  448 |             .auth = client->config->auth,
      |              ^~~~
managed_components/espressif__esp_websocket_client/esp_websocket_client.c:449:41: warning: initialized field overwritten [-Woverride-init]
  449 |             .propagate_control_frames = true
      |                                         ^~~~
/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:449:41: note: (near initialization for 'config.propagate_control_frames')
/managed_components/espressif__esp_websocket_client/esp_websocket_client.c: In function 'esp_websocket_client_task':
/managed_components/espressif__esp_websocket_client/esp_websocket_client.c:849:70: error: implicit declaration of function 'esp_transport_ws_get_upgrade_request_status' [-Werror=implicit-function-declaration]
  849 |                 client->error_handle.esp_ws_handshake_status_code  = esp_transport_ws_get_upgrade_request_status(client->transport);
      |                                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

I think esp_websocket_client is not compatible with esp-idf version 5.1 yet?

david-cermak, 2023-05-16:
@hoangdovan Which version of the component are you using? The current latest version is `v1.0.0` (https://components.espressif.com/components/espressif/esp_websocket_client) and it should work with both `v5.0` and `v5.1` IDF releases.

hoangdovan, 2023-05-18:
@david-cermak Hello David, I use latest version v1.0.0, and esp-idf version v5.1-dev-2356-gedd815af2e-dirty (master branch). And cannot compile because some following problems:
1. esp_transport_ws_config_t does not have "auth" property member
2. esp_transport_ws_get_upgrade_request_status() function does not exist

I doubt that with esp-idf version 5.1 some breaking change cause this problem happen.
I will check again about this matter.
Thank you!

david-cermak, 2023-05-18:
FYI: just referencing the recent CI job building our websocket example with IDF `v5.1`: https://github.com/espressif/esp-protocols/actions/runs/5004207678/jobs/8966478990#step:4:55

(note that this is the latest version of the component, not `v1.0.0`, but I see only cosmetic/docs changes in between)

hoangdovan, 2023-05-28:
@david-cermak I already confirmed again with esp-idf version 5.2, and I can compile without any error. Thank you very much!

closed 2023-05-28 by hoangdovan