espressif/esp-protocols #304: How to wait for *ATREADY: 1 without polling?

0xFEEDC0DE64, 2023-05-31:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My modem does not answer any requests after reset for the first 30 seconds or so.

And I added a workaround to my application like this:

```
    if (dte_config.uart_config.flow_control == ESP_MODEM_FLOW_CONTROL_HW)
    {
        again:
        if (const auto result = dce->set_flow_control(2, 2); result != esp_modem::command_result::OK)
        {
            ESP_LOGE(TAG, "Failed to set the set_flow_control mode %i", std::to_underlying(result));
#if defined(CONFIG_ESP_TASK_WDT_PANIC) || defined(CONFIG_ESP_TASK_WDT)
        if (const auto result = esp_task_wdt_reset(); result != ESP_OK)
            ESP_LOGE(TAG, "esp_task_wdt_reset() failed with %s", esp_err_to_name(result));
#endif
            espcpputils::delay(1s);
            goto again;
        }
        else
            ESP_LOGI(TAG, "set_flow_control OK");
    }
    else
    {
        ESP_LOGI(TAG, "not set_flow_control, because 2-wire mode active.");
    }
```

which will try again and again, until the modem finally responds. I added log statements on the communication to see what is going on there, and it seems the modem tells us it is ready now with this `*ATREADY: 1` message?

[image]

Now my question:
How can I initialize the modem without polling indefinately and halting the rest of the application?

david-cermak, 2023-06-01:
You can send an empty command and wait for this `ATREADY...` as it's *passphrase*.
Unfortunately the `dce->at()` method is hardcoded to check for `OK\r` (I think I'll provide an override to check for arbitrary output)
so it's necessary to create a new method, for example:
```cpp
class MyModule: public esp_modem::GenericModule {
    using GenericModule::GenericModule;
public:
    esp_modem::command_result wait_for_atready(uint32_t timeout)
    {
        return esp_modem::dce_commands::generic_command(dte.get(), "", "ATREADY", "ERROR", timeout);
    }
};
```

That would of course block the execution of current task, but won't busy wait, you're still able to perform other tasks, or you can use `std::await` or something similar.

Alternatively, you can set `on_read()` callback of DTE (added in `1.0.0`) and continue with the modem operations after you receive the `ATREADY...` from the callback. You can find a similar example of using this callback for URC handling in the modem-console application:

https://github.com/espressif/esp-protocols/blob/fbc79a846b1052bc5c97f4b6c7e19762148c56f6/components/esp_modem/examples/modem_console/main/my_module_dce.hpp#L67-L81

david-cermak, 2023-10-31:
Closing this as "answered", feel free to reopen with more details or questions.

closed 2023-10-31 by david-cermak